vvip-68/GyverPanelWiFi #149: Самопроизвольное уменьшение яркости

antidotez, 2021-11-03:
Имеется матрица 15х10 на светодиодной гирлянде, плата wemos d1 mini, прошивка 1.11. Проблема в том, что яркость убавляется самопроизвольно до нуля, отследить почему так происходит у меня не получилось. В приложении ползунок яркости уменьшается сам, притом это даже при отключённой ленте, то есть сама лента не при чем. Почему так и как исправить такое поведение?

vaicekas, 2021-11-03:
А кнопка подключена? Настроена в скече правельно? Так как долгое нажатие кнопки регулирует яркость.

vvip-68, 2021-11-04:
Совершенно справедливые вопросы.

vvip-68, 2021-11-04:
Смотрите [тут](https://github.com/vvip-68/GyverPanelWiFi/wiki/FAQ) в самом конце страницы

closed 2021-11-04 by vvip-68

antidotez, 2021-11-04:
Тип кнопки был указан правильно, кнопки физически не было. Проблема похоже в распиновка платы, изначально пин ленты д2 должен быть, а физически оказался на д4. Походу и было причиной это, поменял пин кнопки с д4 на д6 - сейчас всё ок. Спасибо за советы!

vvip-68, 2021-11-05:
Если у вас пин ленты (был) указан как #define LED_PIN 2 и при этом оказался физически на D4 - значит вы используете библиотеку ядра ESP8266 версии 3.x.x. Следующей проблемой у вас будет постоянно горящий зеленым самый первый светодиод в матрице.