vvip-68/GyverPanelWiFi #151: Инициализация eeprom

tih66, 2021-11-10:
Добрый день!
Не совсем понял в описании a_def_soft.h
"При запуске микроконтроллера выполняется инициализация всех параметров значениями, сохраненными в EEPROM (функция loadSettings() в eeprom.ino)
// При этом если флаг EEPROM_OK в строке 8 этого файла был изменен и не совпадает со считанными яз ячейки #0 значением -
// выполняется инициализация параметров значениями по умолчанию (функция loadSettings() в eeprom.ino, ветвь if (isInitialized) -> false )

#define EEPROM_OK     0x5D       // Флаг, показывающий, что EEPROM инициализирована корректными данными 

// ******************* ОПРЕДЕЛЕНИЯ ПАРАМЕТРОВ подключения к сети *********************

// Внимание!!! Если вы меняете эти значения ПОСЛЕ того, как прошивка уже хотя бы раз была загружена в плату и выполнялась,
// чтобы изменения вступили в силу нужно также изменить значение константы EEPROM_OK в первой строке в файле eeprom.ino "
Если была загружена более ранняя прошивка, то чтобы все прошло на чисто, необходимо поменять #define EEPROM_OK     0x5D? Если да, то на что поменять?

vvip-68, 2021-11-10:
Да, лучше поменять. На любое другое шестнадцатеричное число.

tih66, 2021-11-10:
Спасибо!

closed 2021-11-10 by tih66

Simafor47, 2021-11-27:
сейчас видимо по другому стало во вкладке eeprom

// Внимание!!! Если вы меняете эти значения ПОСЛЕ того, как прошивка уже хотя бы раз была загружена в плату и выполнялась,
// чтобы изменения вступили в силу нужно также изменить значение константы EEPROM_OK в первой строке в файле eeprom.ino

вот первая строка
void loadSettings() {
где конкретно что менять?в скобках?

vvip-68, 2021-11-27:
Ну да, по-другому. А где такой текст остался - надо его поправить.
Сейчас это в a_def_soft.h в строке 7

vvip-68, 2021-11-27:
Нашел где. Исправил.

Simafor47, 2021-11-27:
вместо 0х5D можно например 1x5D?

vvip-68, 2021-11-27:
Конечно нет. Прочитайте что-нибудь про 16-ричную систему счисления и формат записи 16-ричных чисел.
0x - это префикс, сообщающий компилятору, что далее идет число, записанное в 16-ричном формате.

Simafor47, 2021-11-27:
из того что я понял возможно 0x7F подойдёт?))

vvip-68, 2021-11-27:
Если он отличается от предыдущего значения - подойдет.