vvip-68/GyverPanelWiFi #153: Предложение по бегущей строке

CrazyGhostRider, 2021-11-10:
Здравствуйте
Предлагаю сделать более независимую от года прошивку.
Сейчас, например для {R01.01.2021#1} нужно всегда указывать год. И при включении гирлянды на следующий год нужно будет лезть в настройки и менять на {R01.01.2022#1}.

Но можно сделать параметр, например {R01.01.XXXX#1} или {X01.01#1} и автоматически брать текущий год и добавлять +1

CrazyGhostRider, 2021-12-02:
Здравствуйте
Прошивка от 2.12.2021 16:40

0	#123456789ABCD
1	ДО НОВОГО ГОДА ОСТАЛОСЬ {S**.12.****}{R01.01.2022#2}{B#000000}
2	С НОВЫМ 2021 ГОДОМ!!! {S01.01.****#16.01.****}{B#000000}
——//——
D	ДО НОВОГО ГОДА ОСТАЛОСЬ {S**.12.****}{R01.01.2022#2}{B#000000}

Строка 1 сама прописывает "-" в начале строки и отключается. Для проверки скопировал эту строчку в строку D — счетчик заработал.

Даже так "ДО НОВОГО ГОДА ОСТАЛОСЬ {R01.01.2022}{B#000000}" она сама отключается.

vvip-68, 2021-12-02:
А где тексты всех остальных строк?
Скорее всего - какая-то из строк содержит макрос с  {R}, в котором в ветке "иначе" (после наступления события) показывать строку 1. В этом случае, строка, указанная в "иначе" - принудительно отключается, чтобы не отображалась когда не нужно.

vvip-68, 2021-12-02:
>Но можно сделать параметр, например {R01.01.XXXX#1} или {X01.01#1} и автоматически брать текущий год и добавлять +1

Кто бы еще взялся и всё это переделал...
Кроме того - в этом случае вторая часть макроса #R - часть "иначе" выполняться вообще никогда не будет, посколько "иначе" никогда не насттупит. Как только момент прошел - часть года увеличивается на 1 - и снова начинается новый отсчет времени - "До нового года осталось 365 дней". Как-то идея не выглядит слишком удачной...

CrazyGhostRider, 2021-12-02:
Спасибо. Нашел ошибку в строке
ДО НОВОГО ГОДА ОСТАЛОСЬ {P01.01.#1#120#60}{B#0}

CrazyGhostRider, 2021-12-02:
> > Но можно сделать параметр, например {R01.01.XXXX#1} или {X01.01#1} и автоматически брать текущий год и добавлять +1
> 
> Кто бы еще взялся и всё это переделал... Кроме того - в этом случае вторая часть макроса #R - часть "иначе" выполняться вообще никогда не будет, поскольку "иначе" никогда не наступит. Как только момент прошел - часть года увеличивается на 1 - и снова начинается новый отсчет времени - "До нового года осталось 365 дней". Как-то идея не выглядит слишком удачной...

Но при таком сочетании
1 ДО НОВОГО ГОДА ОСТАЛОСЬ {S**.12.**}{R01.01.ХХХХ}
2 С НОВЫМ 2021 ГОДОМ!!! {S01.01.**#16.01.**}

Можно игнорировать, что начнется опять отсчёт.

vvip-68, 2021-12-02:
Согласен. Вопрос кто это переделает. Мне сейчас некогда этим заниматься.
К тому же отладка, что все работает как задумывалось - довольно трудоемкий и затратный по времени процесс...

Я и issue то не закрыл, в надежде когда-нибудь к этому вернуться. Чтобы глаза мозолил.

RRiVan, 2021-12-05:
Здравствуйте! Что-то в крайней версии apk  у меня проблема с текстом. То что ввожу на одной строке , тоже самое отображается на других. И просмотр не работает. В прошлогодней версии все работало.

vvip-68, 2021-12-05:
Ну у меня и в этой все работает. Вот затрудняюсь дать ответ что у вас происходит?
Версия apk соответствует версии прошивки, загруженной в контроллер?

vvip-68, 2021-12-05:
> И, если Галка текст поверх эффектов снята, все равно эффекты не отключаются

Потому что эта галка отключает не эффект при воспроизведении строки, а отключает текст при вооспроизведении эффекта. Если галки нет - строка не будет показываться пока этот эффект "крутится"

RRiVan, 2021-12-05:
> спасибо! Вероятно я балбес!!! Прошивку не обновлял.

tushkanius, 2021-12-13:
Предложение по бегущей строке.
Прошу добавить в бегущую строку вывод из ленты новостей или сообщения поступающие из твиттера.

xxjack0023, 2021-12-13:
Всем привет. Спасибо за проделанную работу. GyverPanelWiFi (vvip-68) пользуюсь уже третий год, всё нравится. В этом году залил на esp32, всё работает ( даже mqtt), но раньше по кругу в эффектами крутились текстовые строки (вроде бы)))) ), сейчас запускается строка только из редактора строк кнопкой “просмотр”. Везде строка по верх эффекта отключена, на изменение интервала не реагирует, строка “0” имеет текст ##. За ранее спасибо.

xxjack0023, 2021-12-13:
[image]

CrazyGhostRider, 2021-12-13:
>  Везде строка по верх эффекта отключена
Если у всех эффектов отключить "Использовать бегущую строку поверх эффекта", то бегущей строки не будет.
Бегущая строка ждет свой таймер, потом проверяет проигрываемый эффект на вкл. опцию. Если всё ОК — показывает строчку.

п.с. можно добавить в строчку {B # 000000}, чтобы строчка была на черном фоне.

xxjack0023, 2021-12-13:
> Если у всех эффектов отключить «Использовать бегущую строку эффекта», то бегущей строки не будет. >

В том то и дело, что в том году  ставил интервал и с этим интервалом выводился текст

> п.с. можно добавить в строчку {B # 000000} (без пробелов), чтобы строчка была на черном фоне.

Если делать так то все строки выводятся сразу все (по очереди) пока идет время эффекта, а между строками моргает сам эффект

CrazyGhostRider, 2021-12-13:
> В том то и дело, что в том году ставил интервал и с этим интервалом выводился текст

В том году была версия прошивки v1.07 и я не знаю, как она работала. А сейчас, и приложение обновилось, и прошивка стала v1.12.

> Если делать так то все строки выводятся сразу все (по очереди) пока идет время эффекта, а между строками моргает сам эффект

Сразу все не должны. Таймер между строчками, случаем, не очень маленький?
Как я понял, строчки работают, как я описал. Поэтому у себя поставил 30 сек между эффектами и 200сек между строчками на черном фоне.

playeeerua, 2021-12-15:
А не подскажете где вы брали тонкий белый пластик, рассеиватель? Обхожу рекламщиков, искал в эпицентре (аналог Леруа), то нету. Когда лампу делал, то рассеиватель использовал с матрицы ноута😂

simonovich, 2021-12-16:
> А не подскажете где вы брали тонкий белый пластик, рассеиватель?
Точно так же бегал по городу. Снять с ноута или телика - хорошее решение, но эти девайсы не валяются просто так и нужной диагонали.
В итоге сделал по другому. На строительном рынке или в гипере просим отпилить кусок оргстекла "для светодиодов" (оно же "акрил"). Это важно, т.к. оргстекла очень много всяких видов и толщины, и не все умеют пропускать и правильно рассеивать свет. Удобно проверять фонариком со своего мобильного телефона. Врятле получится найти такой пластик, чтобы он еще и затемнял в нужном спектре (хотя бывает). Обычный белый вполне подойдёт. Но его придётся доработать. 
Для этого едем в автолавку или на авторынок и ищем рулон тонировочной плёнки для стекла. Еще можно поискать аналогичную для оклейки стекла на окнах, но она реже встречается. Плёнки тоже много всяких видов по степени пропускания света, а также по цвету. Увы, тоже только на глаз выбираем. Если надо - могу посмотреть, что у меня. Эту пленку клеим на ваш кусок пластика - и получаем готовый рассеиватель.
В самом крайнем случае можно попробовать обтянуть чем-нибудь другим - темный мусорный пакет, какой-нить лак, презик... )) Но эффект будет значительно хуже.

vvip-68, 2021-12-16:
Я делал так - брал оргстекло 1-2мм в магазине типа Леруа или Вираж, прозрачное. Со стороны, обращенной к диодам наклеивал белую самоклеющуюся пленку из того же леруа. С лицевой стороны - автомобильную тонировочную пленку - самую светлую, которая 35% что ли пропускания. Вот такой пирог светит нормально. Естественно чем тоньше оргстекло, тем меньше рассеивание в толщине стекла - четче "пиксели"

vvip-68, 2021-12-16:
А вообще - какого лешего всё это обсуждение здесь - в Issue? Есть же специальное место для дискуссий. 
Я специально этот Issue не закрываю, чтобы помнить о предложенной фиче и когда-нибудь сделать ее. А тут развели обсуждение вообще к теме не относящееся...

closed 2022-11-25 by vvip-68