vvip-68/GyverPanelWiFi #154: Настройка е131

Vasap62, 2021-11-13:
Подскажите как настроить синхронизацию контроллеров. Желательно для чайников))) Никак не могу настроить. Заранее спасибо!

vvip-68, 2021-11-13:
Я вряд ли смогу что-то подробнее рассказать, чем это в wiki расписано.
Если есть конкретные вопросы - готов выслушать.

Vasap62, 2021-11-13:
Я имею ввиду конкретно на WiFi панели. Вы там добавили функцию синхронизации эффектов, а я не могу Че-то настроить её

vvip-68, 2021-11-13:
Вики читали? Что сделали для настройки?

Vasap62, 2021-11-13:
Вики пока не читал. Выставил с настройках в приложении один контроллер ведущим, другой ведомым. Один контроллер сделал точкой доступа (ведущий), другой попытался к нему подключить введя ssid и пароль ведущего контроллера, ничего не вышло((( или они должны быть подключены к одной общей сети где точкой доступа выступает роутер например? Я не очень разбираюсь в программировании наверное что-то делаю не так

vvip-68, 2021-11-13:
Давайте, вы сначала почитаете вики, что-то попробуете, потом будете задавать конкретные вопросы?
Синхронизация работает только при подключении к локальной сети. В режиме точки доступа - эта технология не работает.

closed 2021-11-13 by vvip-68

Vasap62, 2021-11-13:
Понял, спасибо большое!