vvip-68/GyverPanelWiFi #155: помогите с настройкой mqtt!!!

Simafor47, 2021-11-18:
Зарегистрировался на сайте mqtt.by получил пароль/префикс и т.д.
Что то ещё нужно создавать на этом сайте?топики например? просто не могу подключиться.. Описание читал не однократно,не врубаюсь никак((

vvip-68, 2021-11-19:
Кроме как зарегистрироваться больше ничего на сервере не требуется.
В статье wiki исчерпывающе описана настройка подключения - что прописывать в скетче, как настраивать подключение в приложении.
Единственный момент - во вторник 16 ноября они на сервере изменили структуру префикса подключения. 
Если раньше в префиксе требовалось указать просто нечто уникальное (как в статье - *af7cd12a*), теперь в качестве префикса требуется указывать имя пользователя.

Например, если на mqtt.by вы зарегистрировались как *Simafor47*, то в настройках префикса в скетче и в приложении требуется писать *user/Simafor47/af7cd12a*.  Впрочем, если у вас только одно устройство - то вот этой части *af7cd12a* (уникального хвоста для идентификации разных устройств) можно и не писать теперь.

Все это (структура префикса) написано в личном кабинете на mqtt.by. Достаточно просто прочитать описание.

vvip-68, 2021-11-19:
Поставьте себе на телефон приложение вроде [MyMQTT](https://play.google.com/store/apps/details?id=at.tripwire.mqtt.client&hl=ru&gl=US), настройте подключение там. В Мониторе порта видно подключился ли скетч к серверу и какие сообщения отправляет, в этом приложении видно, что сообщения получаются, то есть что канал связи работает. Если **MyMQTT** видит сообщения от устройства, значит приложение с правильными настройками тоже должно быть способно подключиться к устройству.

Simafor47, 2021-11-19:
Большое спасибо за ответ! Дело было в бракованой плате. Там помимо подключения ещё и один из пинов RX/TX не работал. Регулярно после перезапуска не стартовала,прочерки на экране и таймаут запросов в мониторе порта. В общем поменял плату и сразу всё заработало как надо,правда ещё пока без гирлянды(жду посылку) но функционал работает!

closed 2021-11-19 by Simafor47