vvip-68/GyverPanelWiFi #156: Здравствуйте, нужна помощь с подключением к роутеру

ferakond, 2021-11-20:
Контроллер подключается к роутеру, но с приложения не получается подключится

vvip-68, 2021-11-20:
Настройте свой телефон (или роутер) так, чтобы пакеты, отправляемые с телефона доходили через роутер до устройства. Найдите и установите на телефон утилиту проверки связи типа Ping, Когда эта утилита будет получать ответы от контроллера - программа управления сможет подключиться к контроллеру.

Некоторые телефоны требуют отключения передачи "Мобильные данные", иначе они отправляют пакеты опроса контроллера не роутеру в локальной сети, а куда-то в интернет по каналу мобильной связи. Отключение этого канала (или корректная настройка в параметрах телефона) позволяет запросам приложения к контроллеру уходить в локальную сеть роутера и передаваться контроллеру, вместо отправки "в никуда".

closed 2021-11-20 by vvip-68

ferakond, 2021-11-20:
Спасибо, попробую