vvip-68/GyverPanelWiFi #158: Не сохраняются настройки Wi-Fi

timof123, 2021-11-20:
В настройках a_def_soft.h указаны данные Wi-Fi, к которому должно подключаться устройство. Когда подключается первый раз (данные загружаются не из eeprom), то все окей, оно подключается. Но после перезагрузки оно забывает эти настройки и создает свою точку доступа:

> Инициализация файловой системы... 
> 
> Выполняется разметка файловой системы... 
> Файловая система недоступна.
> Нет конфигурации сети: SSID не задан
> Нет конфигурации сети: пароль не задан
> Загрузка строк выполнена.
> Использованы адреса EEPROM 1000 - 1037
> Свободно ячеек 3058
> 
> Host: 'WiFiPanel' >> 10x15
> 
> Создание точки доступа PanelAP
> Точка доступа создана. Сеть: 'PanelAP'. Пароль: '12341234'.
> IP адрес: 192.168.4.1
> 
> UDP-сервер на порту 2390
> 
> Режим работы: АВТОНОМНЫЙ, синхронизация E1.31 отключена
> 
> Режим: The Matrix

Пробовал и через приложение настраивать. Все равно забывает.
Как сохранить настройки в eeprom? Чтобы не забывались....

vvip-68, 2021-11-20:
Вся мугамба кроется в первых сроках лога, который вы привели:

>Инициализация файловой системы...
>
>Выполняется разметка файловой системы...
Файловая система недоступна.
Нет конфигурации сети: SSID не задан
Нет конфигурации сети: пароль не задан

Имя сети и пароль сохраняются в файлах в файловой системе микроконтроллера.
У вас, как пишет контроллер - не удается инициализировать файловую систему.
Вероятно в настройках компиляции вы выбрали вариант разметки - без файловой системы, поэтому сохранять настройки контроллеру негде.

[image]

Скорее всего у вас выбран пункт с 'none'

timof123, 2021-11-20:
О дааа, там стояло none почему-то. Спасибо огромное!!!!

closed 2021-11-20 by timof123