vvip-68/GyverPanelWiFi #159: Белый цвет.

kost111233, 2021-11-23:
Извините что сам не попробовав, а сразу сюда пишу.  Нет возможности в ближайшие дни попробовать. Проблема в том, что сначала купил готовые 5 лент по 50 светодиодов, а сейчас докупил еще 100 штук просто светодиодов. У них отличается белый цвет, в лентах более розовый оттенок, другие цвета вроде как одинаковыми кажутся. Вот думаю что если я в скетче изменю везде где белый цвет на другой оттенок белого, получится так их привести к одинаковому цвету? 
[image]

kost111233, 2021-11-23:
Покупал у одного и того же китайца, заранее спрашивал будут ли одинаковыми, ответил что одинаковые. Сейчас спор открыл и пробую хотя бы 200 руб вернуть) думаю это как бы не неисправность, но еслиб китаец так не ответил, мог купить дешевле.

vvip-68, 2021-11-23:
Думаю что нельзя. Это физические характеристики диодов и скетчем b] не исправишь. Очевидно что диоды (даже купленные у одного продавца) могут быть из разных партий произведенных заводом. Почти наверняка они будут отличаться от другой партии.

closed 2021-11-23 by vvip-68

kost111233, 2021-11-23:
Знал что такое может быть, надеялся меня не коснется) спасибо за помощь, попробую потом поменять цвет в эффектах снег, дождь и зебра, вдруг все таки получится сделать схожий оттенок