vvip-68/GyverPanelWiFi #161: Тупик

playeeerua, 2021-11-26:
Поменял имя вай фая, без пробела теперь, припаяно без кнопки (на фото есть). Решил на чистой плате попробовать, где ничего не спаянно, то в мониторе порта одни вопросы вылезли, на основной плате аналогично вопросы
[image]
 
[image]

247760, 2021-11-26:
@tarasifua 
Для того, чтобы понять работает ли прошивка, не обязательно подсоединять к гирлянде. Когда всё правильно настроено, записано, com порт показывает всё что происходит, достаточно подключения по одному шнуру питания к ноутбуку.
Я вижу у Вас порт 3, у меня был случай на не рабочей плате, в какой пот не вставь на ноуте везде порт 3 был. Огоньки моргаю у Вас? Скорость порта по моему 115200. Иногда после подключения в поте кракозябры, после нажатия на ресет платки всё стартует.

playeeerua, 2021-11-26:
@247760 
Перед прошивкой с помощью проги флешер для данной платы вроде форматирую плату, оно показывает какой порт и мак адрес устройства (в прошлый раз не показывало, оказалось что кабель только для зарядки предназначен без передачи, поменяли)
Проверяю на двух платах, одна чистая, без пайки, подключена только в комп, вторая спаянная и подключена в розетку (помню прошлый проект гайверламп так и прошивался через розетку+кабель в комп. Возможно не работает из-за этого сообщения, но почитал на форумах это не влияет
[image]

247760, 2021-11-26:
@playeeerua Моя не рабочая платка, прекрасно шилась программатором прошивкой с расширением .bin, в таком виде: прогер + питание по шнуру от ноута, даже шёл обмен через ком порт, светодиоды моргали. Но стоило всё подключить к гирлянде, как ничего не работало. Рабочая платка, и правильная прошивка, работают сходу и шьётся одним кабелем прямо с Ардуино ИДЕ, даже не прописывая свою сеть оно стартанёт по умолчанию, вот выдержка из описания:
_Если не удается подключиться к сети (неверный пароль или имя сети) - автоматически создается точка подключения с именем PanelAP, пароль 12341234, IP 192.168.4.1. Подключившись к точке доступа из приложения можно настроить параметры сети. Если после задания параметров сети WiFi соединение установлено - в приложении на смартфоне виден IP адрес подключения к сети WiFi._
Боюсь попутать, но вроде на пине 2 или 3, не должно быть 3,3 вольта, а на моей не исправной на обоих напряжение было.

playeeerua, 2021-11-26:
@247760  У меня проблема что не создает точку доступа, в IP 4.1 тоже глухо в 115200 ничего нету, а в 74880 вопросы печатает (я в коммите просто писал, автор написал чекнуть 74880), в модеме 192.168.1.1 его не видно (дали айпи 1.90)

playeeerua, 2021-11-26:
[image]

vvip-68, 2021-11-27:
Вот те вопросы на картинки вверху на скорости 74800 - это сразу после нажатия кнопки ресет на плате?
Если да - у вас не исправна плата. Сразу после старта контроллера он выдает на этой скорости информацию о себе.
Если это не сразу после ресета - то это работает прошивка и выдает диагностическую информацию (что происходит) на скорости 115200. Из за несовпадения скоростей - печатаются вопросы.

Если без подключенной ленты прошивка загружается и работает - есть сообщения в мониторе порта на скорости 115200, а после того как подключили ленту - контроллер не стартует - у вас или неисправна лента или подсоединена не тем концом.

247760, 2021-11-27:
@playeeerua Лента проверяется быстро простым скетчем, записать его на исправную плату, хоть на любую Ардуину, только укажите в строке сколько у Вас светодиодов: 
```
#define STRIP_PIN 2     // пин ленты
#define NUMLEDS 100     // кол-во светодиодов
#include <microLED.h>   // подключаем библу
microLED < NUMLEDS, STRIP_PIN, -1, LED_WS2812, ORDER_GRB > strip;
void setup() {
  strip.setBrightness(60);  // яркость
}
void loop() {
  static byte counter = 0;
  for (int i = 0; i < NUMLEDS; i++) {
    strip.set(i, mWheel8(counter + i * 255 / NUMLEDS));   // counter смещает цвет
  }
  counter += 3;   // counter имеет тип byte и при достижении 255 сбросится в 0
  delay(30);      // 30 кадров в секунду
}
```
Если огни пробегут всю лену - она исправна и можно заниматься основной прошивкой, если несть не пропай или сгоревший светодиод, то перед ним огни остановятся, если есть КЗ на участке, то хвост перед ним будет тускло гореть\моргать. Как по мне, самый просто и эффективный способ закрыть вопрос по исправности ленты.
Так же проверьте платку свою, на разных портах ПК, если везде будет COM3 и без бегущих данных - она скорее всего не исправна, отсутствие моргания на платке светодиодов - говорит о не исправности. При исправной платке и корректно записанной прошивке, обмен в порте есть всегда, иногда надо нажать ресет (при этом моргнёт светодиод). 
Описал как было у меня, поэтому проверяйте, рабочий модуль и правильная прошивка, в купе с исправной лентой работают сразу.

vvip-68, 2021-11-27:
@247760 Во первых - этот скетч под ардуину, а у него ESP8266 на которую скетч даже не загрузится
Во вторых - у него контроллер вообще не запускается, если он в мониторе порта ничего внятного не видит.

247760, 2021-11-27:
@vvip-68 в частности про Ардуину в посте и упомянул (лишнее убрал), а снизу  сразу добавил, что всё надо проверять. 
На половину моих вопросов и замечаний ответов никаких нет, поэтому и приходится "толочь в ступе воду", выуживая информацию по кругу. 
Предположение о не рабочей платке, находит всё больше подтверждений. 
Ещё один вариант проверки ленты (при сборе её из отдельных светодиодов на круглой платке), на правильность направления пайки светодиодов, это вручную проверить, чтобы все стрелочки на светодиодах сзади, шли в одном направлении т.е. если вытянуть ленту в одну линию, то светодиоды должны идти строго справа налево либо слева на право, но этот способ для ЗИГЗАГ подключения.

playeeerua, 2021-11-27:
@vvip-68 вопросы на скорости 74800, после того, как прошил устройство, когда переключаю на 115200, то там или пусто, или одна строка с иероглифами. На ресет еще не нажимал, потом попробую, отпишу. Я прошиваю 2 платы, одна которая подключена к леньа, вторая чистая без него и оба на скорости 115200 без реакции, на 74800 с вопросами

vvip-68, 2021-11-27:
@247760  - я к тому, что прежде чем ленту проверять стоило бы убедиться, что плата вообще рабочая. Или нерабочая. Пока я из описания ничего внятного не вижу. Если вопросы в мониторе порта - значит что-то пытается выводить.  А выводить она может только на двух скоростях - 74800 сразу после прошивки / нажатия на кнопку ресет - вывод диагностики с самого микроконтроллера, либо когда уже работает - на скорости 115200 информацию о том, чем сейчас занят контроллер. Но ни того ни другого вывода автор увидеть не может. Значит, есть вероятность, что плата вообще не рабочая.

247760, 2021-11-27:
@vvip-68 солидарен с тем, что ничего не понятно, плюс опыта у пользователя видимо нет в таких вещах.
И хочется помочь, но Бог его знает как, так как обратная связь хромает.
Как вариант, рекомендую @playeeerua посмотреть ролики с гирляндой на ютубе. может кто то на видео запечатлел процесс прошивки, и тогда дело пойдёт веселее.

playeeerua, 2021-11-27:
@247760 
Есть 2 платы nodemcu. Одна уже спаянная во всеми (как на фото)  и подключена к светодиодам. И есть всторая, чистая, новая, прям с Китая, нетронутая, ничего к нему не паял. Делал я так, потому что по видео все было, сначала спаяй и подключи, потом проверяй. При прошивке светодиод на плате мигает. Остановился на том, что оно как-бы прошилось, но точки доступа или подключения нет. Еще раз форматировал и скомпилировал, После нажатия на кнопку ресет, выдает, то что на скрине, знаков вопроса которые бесконечно прописываются - пропали. Да, опыта у меня практически ноль, был опыт с гайвер лампой, но это время прошло и все забылось) Возможно я что-то упускаю, для меня это джунгли и некоторые слова я просто возможно не понимаю, поэтому не отвечаю на них)
[image]
[image]

247760, 2021-11-27:
@playeeerua что на скорости 115200 после нажатия ресет?

vvip-68, 2021-11-27:
@playeeerua - ну на скорости 74880 хотя бы видно, что процессор жив и что-то выводит вразумительное.
да, @247760 прав: теперь - что в порт выдается на 115200 после нажатия на reset?

playeeerua, 2021-11-27:
@247760 @vvip-68 
[image]

vvip-68, 2021-11-27:
Ну, тогда всё. У меня нет ответа. Потому что сразу же после запуска процессора практически первой строкой скетча - идет вывод номера версии программы в монитор порта. То есть фактически, получается, что выполнение скрипта не доходит даже до первых строк в setup(). Я такой ситуации никогда не встречал. У меня нет ответа что делать в такой ситуации.

Попробуйте в контроллер загрузить следующий скетч:
```

void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println("Hello world!");
  delay(1000);
}
```

playeeerua, 2021-11-27:
@vvip-68 работает, выдает текст.
Значит дело получается не в плате? Возможно я где-то стер что-то в коде? Хотя я просто поменял размер матрицы, угол и откуда идет лента.

vvip-68, 2021-11-27:
Значит, дело не в плате.
В Serial может ничего не выводить, если выбран 
#define DEVICE_VARIANT 1 в строке 52 файла a_def_hard.h - это когда делается устройство со всеми подключенными модулями и пин, который обычно используется Serial для коммуникации с компом занят на какое-то устройство.
Тогда вывод в порт отключается и ничего выводиться не будет.

У вас устройство с подключенными TM1637, DFPlayer, SD-картой, Кнопкой, реле питания и что там еще в "полном фарше"?

playeeerua, 2021-11-27:
Стоит 0. У меня только плюс минус и D2 с резистором (на фото в начале есть), ничего не контачит, проверял
Я вроде смотрел код, там еще было есп8266 (поставил 1) и еср8266+алл (переставил на 0)

vvip-68, 2021-11-27:
Ну, высод еще может быть отключен тогда, в строке 141
```
#define DEBUG_SERIAL 0
```
Для того, чтобы сообщения шли в монитор порта тут должна быть 1

playeeerua, 2021-11-27:
стоит 1

vvip-68, 2021-11-27:
В основном файле в строке 122 есть 
```
Serial.begin(115200);
```
Это инициализация порта на вывод на указанной скорости.
Начиная со строки 125 до 142 - вставляйте перед/после каждого логического блока кода
Serial.println("#1");
затем
Serial.println("#2");
 и смотрите будет ли что выводиться и на чем закончится.
В строке 143 - вывод текущей версии, но как мы видим - до него дело не доходит - в монитор порта ничего не печатается

playeeerua, 2021-11-27:
[image]
я правиль понял, вот так надо?

vvip-68, 2021-11-27:
Ну, типа того. Можно еще перед randomSeed воткнуть

vvip-68, 2021-11-27:
Если ничего выводиться не будет - поставьте для эксперимента в строке 109 в самом начале `/*`
и в строке 140 в самом начале `*/`

Simafor47, 2021-11-27:
Прошу прощения что влез! У меня проблема с постоянной перезагрузкой платы. Уже удалял полностью и переустанавливал всё по инструкции,но пока без изменений... Не могу понять в чём может быть проблема. Имеетя 2 платы 8266 и на обоих одинакого..
монитор прилагаю.

20:33:57.916 -> 
20:33:57.916 -> WiFiPanel v.1.12.2021.1127
20:33:57.916 -> 
20:33:57.916 -> 
20:33:57.916 -> Инициализация файловой системы... 
20:33:57.916 -> FS: Использовано 16384 из 2072576 байт
20:33:57.916 -> Загрузка строк выполнена.
20:33:57.963 -> Использованы адреса EEPROM 1000 - 1037
20:33:57.963 -> Свободно ячеек 3058
20:33:57.963 -> 
20:33:57.963 -> Host: 'WiFiPanel' >> 26x13
20:33:57.963 -> 
20:34:00.221 -> 
20:34:00.221 -> Инициализация MP3 плеера.
20:34:06.786 -> Звуков будильника найдено: 0
20:34:06.786 -> Звуков рассвета найдено: 0
20:34:06.786 -> Звуков сообщений найдено: 0
20:34:06.786 -> MP3 плеер недоступен.
20:34:07.086 -> Создание точки доступа PanelAP
20:34:08.944 -> Точка доступа создана. Сеть: 'PanelAP'. Пароль: '12341234'.
20:34:08.991 -> IP адрес: 192.168.4.1
20:34:08.991 -> 
20:34:08.991 -> UDP-сервер на порту 2390
20:34:08.991 -> 
20:34:08.991 -> Режим работы: АВТОНОМНЫЙ, синхронизация E1.31 отключена
20:34:08.991 -> 
20:34:09.091 -> Режим: Спектрум
20:34:09.091 -> 
20:34:09.091 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
20:34:09.144 -> 
20:34:09.144 -> Exception (28):
20:34:09.144 -> epc1=0x40237f3b epc2=0x00000000 epc3=0x00000000 excvaddr=0x0aff1424 depc=0x00000000
20:34:09.144 -> 
20:34:09.144 -> >>>stack>>>
20:34:09.144 -> 
20:34:09.144 -> ctx: cont
20:34:09.144 -> sp: 3ffffd00 end: 3fffffc0 offset: 0190
20:34:09.144 -> 3ffffe90:  ffffffa3 0000001c 3ffffee4 3fff151c  
20:34:09.144 -> 3ffffea0:  3fff13d8 00000000 3fff13d8 00000000  
20:34:09.144 -> 3ffffeb0:  3fff13fc 3fff12c8 00000000 4023043d  
20:34:09.144 -> 3ffffec0:  3fff7400 001c001f 80c6a7f0 00000000  
20:34:09.144 -> 3ffffed0:  00000000 00000003 40100940 00002d01  
20:34:09.144 -> 3ffffee0:  00000000 3fff7400 000c000f 8023bde4  
20:34:09.144 -> 3ffffef0:  0034096f 3fff13c8 3ffe8fdb 4022a64c  
20:34:09.144 -> 3fffff00:  00747400 0000001a 3fffff48 4023bdc5  
20:34:09.144 -> 3fffff10:  3fff1fba 3fff13c8 3fff1280 4023bde4  
20:34:09.191 -> 3fffff20:  3fff1fba 3fff13c8 3fff1280 4022ae9a  
20:34:09.191 -> 3fffff30:  3fff35e4 00004000 40205920 4020581c  
20:34:09.191 -> 3fffff40:  00000005 00000020 52415400 000e0054  
20:34:09.191 -> 3fffff50:  80002d6c 203e3e00 00340000 80efeffe  
20:34:09.191 -> 3fffff60:  3fff4b00 001a001f 80efeffe 3fff4b00  
20:34:09.191 -> 3fffff70:  0019001f 80efeffe feefeffe feefeffe  
20:34:09.191 -> 3fffff80:  00000152 00000000 3ffef650 3fff2598  
20:34:09.191 -> 3fffff90:  3fffdad0 00000000 3fff2558 40230689  
20:34:09.191 -> 3fffffa0:  3fffdad0 00000000 3fff2558 4023d324  
20:34:09.191 -> 3fffffb0:  feefeffe feefeffe 3ffe85f0 40101595  
20:34:09.245 -> <<<stack<<<
20:34:09.245 -> 
20:34:09.245 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
20:34:09.245 -> 
20:34:09.245 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
20:34:09.245 -> 
20:34:09.245 -> load 0x4010f000, len 3584, room 16 
20:34:09.245 -> tail 0
20:34:09.245 -> chksum 0xb0
20:34:09.245 -> csum 0xb0
20:34:09.245 -> v2843a5ac
[... 142 lines cut ...]
20:34:33.000 -> FS: Использовано 16384 из 2072576 байт
20:34:33.000 -> Загрузка строк выполнена.
20:34:33.047 -> Использованы адреса EEPROM 1000 - 1037
20:34:33.047 -> Свободно ячеек 3058
20:34:33.047 -> 
20:34:33.047 -> Host: 'WiFiPanel' >> 26x13
20:34:33.047 -> 
20:34:35.306 -> 
20:34:35.306 -> Инициализация MP3 плеера.
20:34:41.868 -> Звуков будильника найдено: 0
20:34:41.868 -> Звуков рассвета найдено: 0
20:34:41.868 -> Звуков сообщений найдено: 0
20:34:41.921 -> MP3 плеер недоступен.
20:34:42.168 -> Создание точки доступа PanelAP
20:34:44.073 -> Точка доступа создана. Сеть: 'PanelAP'. Пароль: '12341234'.
20:34:44.073 -> IP адрес: 192.168.4.1
20:34:44.073 -> 
20:34:44.073 -> UDP-сервер на порту 2390
20:34:44.073 -> 
20:34:44.073 -> Режим работы: АВТОНОМНЫЙ, синхронизация E1.31 отключена
20:34:44.073 -> 
20:34:44.173 -> Режим: Рассвет

и так по кругу. 
п.с. плата отдельно подключена к компьютеру

vvip-68, 2021-11-27:
@Simafor47 Ну как минимум, судя по логам видно, что у вас скорее всего плата скомпилирована с настройками БЕЗ выделения места под файловую систему. Возможно поэтому, но скорее всего нет. Чтобы попытаться выяснить точное место проблемы вам нужно установить утилиту расшифровки вот того стека ошибок и загрузить стек из лога в эту утилиту. Тогда может быть станет яснее...

https://github.com/me-no-dev/EspExceptionDecoder

Simafor47, 2021-11-27:
утилиту установил по инструкции,но что то не понял:
"Откройте эскиз и постройте его.
Загрузите скетч и следите за последовательным портом на предмет исключений
Когда вы получите исключение, откройте меню Инструменты> ESP Exception Decoder . Это откроет новое окно."

у меня открывается проводник,но что открывать и что искать?

vvip-68, 2021-11-27:
Вы скорее всего не выполнили компиляцию программы перед тем как в окне декодера поместить скетч.
Окно с загруженным и дешифрованным стеком должно выглядеть примерно так:

[image]

Simafor47, 2021-11-27:
[image]

vvip-68, 2021-11-28:
Ну, судя по картинке - какая-то проблема с DFPlayer. И выделением памяти (или чтением неинициализированного объекта). К сожалению - ничего конкретного сказать не могу.  У меня в 357 строке a_main.ino - вообще delay(0) в функции подключения к WiFi и никакой связи с плеером нет. 

А с отключенным в настройках DFPlayer происходит циклическая перезагрузка?
Многие жалуются на неработоспособность плеера. Их сейчас развелось великое множество, выполненных на разных микросхемах и ведут они себя несколько по разному. Библиотека DFRobotDFPlayerMini показала себя не очень надежной. В следующих версиях планируется переход на другую версию (другого разработчика) библиотеки для работы с плеером. Может быть тогда ошибка исчезнет...

A файловую систему все же включите в настройках компиляции, иначе параметры подключения к сети не будут сохраняться и вместо работы в локальной сети придется все время подключаться к точке доступа.

Simafor47, 2021-11-28:
вот с подключенной сетью и так по кругу..
08:34:11.531 -> WiFiPanel v.1.12.2021.1127
08:34:11.531 -> 
08:34:11.531 -> 
08:34:11.531 -> Инициализация файловой системы... 
08:34:11.531 -> FS: Использовано 16384 из 2072576 байт
08:34:11.531 -> Загрузка строк выполнена.
08:34:11.531 -> Использованы адреса EEPROM 1000 - 1037
08:34:11.531 -> Свободно ячеек 3058
08:34:11.585 -> 
08:34:11.585 -> Host: 'WiFiPanel' >> 26x13
08:34:11.585 -> 
08:34:13.837 -> 
08:34:13.837 -> Инициализация MP3 плеера.
08:34:20.405 -> Звуков будильника найдено: 0
08:34:20.405 -> Звуков рассвета найдено: 0
08:34:20.405 -> Звуков сообщений найдено: 0
08:34:20.405 -> MP3 плеер недоступен.
08:34:20.606 -> 
08:34:20.606 -> Подключение к XXX -> 192.168.2.160
08:34:20.653 -> .............
08:34:24.716 -> WiFi подключен. IP адрес: 192.168.2.160
08:34:24.716 -> 
08:34:24.716 -> UDP-сервер на порту 2390
08:34:24.716 -> 
08:34:24.716 -> Режим работы: АВТОНОМНЫЙ, синхронизация E1.31 отключена
08:34:24.716 -> 
08:34:24.716 -> 
08:34:24.716 -> Подключаемся к MQTT-серверу 'mqtt.by:1883'; ClientID -> 'WiFiPanel-d981' ...
08:34:25.264 -> Подключение к MQTT-серверу выполнено.
08:34:25.264 -> MQTT >> OK >> [r] af7cd12a/pwr	 >> online
08:34:25.264 -> MQTT >> OK >> [ ] af7cd12a/sta	 >> START
08:34:25.317 -> MQTT >> OK >> [r] af7cd12a/stt/LE	 >> Часы,Лампа,Снегопад,Кубик,Радуга,Пейнтбол,Огонь,The Matrix,Шарики,Звездопад,Конфетти,Цветной шум,Облака,Лава,Плазма,Радужные переливы,Павлин,Зебра,Шумящий лес,Морской прибой,Смена цвета,Светлячки,Водоворот,Циклон,Мерцание,Северное сияние,Тени,Лабиринт,Змейка,Тетрис,Арканоид,Палитра,Спектрум,Синусы,Вышиванка,Дождь,Камин,Стрелки,Анимация,Погода,Жизнь,Узоры,Рассвет
08:34:25.364 -> MQTT >> OK >> [r] af7cd12a/stt/S1	 >> One Step Over,In the Death Car,Труба зовет,Маяк,Mister Sandman,Шкатулка,Banana Phone,Carol of the Bells
08:34:25.417 -> MQTT >> OK >> [r] af7cd12a/stt/S2	 >> Птицы,Гроза,Прибой,Дождь,Ручей,Мантра,La Petite Fille De La Mer
08:34:25.464 -> MQTT >> OK >> [r] af7cd12a/stt/S3	 >> Piece Of Сake,Swiftly,Pristine,Goes Without Saying,Inflicted,Eventually,Point Blank,Spring Board,To The Point,Serious Strike,Jingle Bells,Happy New Year,Christmas Bells,Door Knock,Alarm Signal,Viber Message,Viber Call,Text Message,Old Clock 1,Old Clock 2,Old Clock 3
08:34:25.464 -> MQTT >> OK >> [r] af7cd12a/stt/LT	 >> ##~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ 
08:34:25.464 -> MQTT >> OK >> [r] af7cd12a/stt/W	 >> 26
08:34:25.464 -> MQTT >> OK >> [r] af7cd12a/stt/H	 >> 13
08:34:25.517 -> MQTT >> OK >> [r] af7cd12a/stt/DM	 >> true
08:34:25.517 -> MQTT >> OK >> [r] af7cd12a/stt/PS	 >> true
08:34:25.517 -> MQTT >> OK >> [r] af7cd12a/stt/PD	 >> 60
08:34:25.517 -> MQTT >> OK >> [r] af7cd12a/stt/IT	 >> 0
08:34:25.517 -> MQTT >> OK >> [r] af7cd12a/stt/AL	 >> false
08:34:25.517 -> MQTT >> OK >> [r] af7cd12a/stt/RM	 >> true
08:34:25.517 -> MQTT >> OK >> [r] af7cd12a/stt/PW	 >> 10000
08:34:25.517 -> MQTT >> OK >> [r] af7cd12a/stt/BR	 >> 32
08:34:25.517 -> MQTT >> OK >> [r] af7cd12a/stt/WU	 >> 1
08:34:25.517 -> MQTT >> OK >> [r] af7cd12a/stt/WT	 >> 15
08:34:25.517 -> MQTT >> OK >> [r] af7cd12a/stt/WR	 >> 10867
08:34:25.517 -> MQTT >> OK >> [r] af7cd12a/stt/WS	 >> 1502026
08:34:25.517 -> MQTT >> OK >> [r] af7cd12a/stt/WC	 >> true
08:34:25.517 -> MQTT >> OK >> [r] af7cd12a/stt/WN	 >> false
08:34:25.564 -> MQTT >> OK >> [r] af7cd12a/stt/WZ	 >> true
08:34:25.564 -> MQTT >> OK >> [r] af7cd12a/stt/EF	 >> 21
08:34:25.564 -> MQTT >> OK >> [r] af7cd12a/stt/EN	 >> Светлячки
08:34:25.564 -> MQTT >> OK >> [r] af7cd12a/stt/UE	 >> true
08:34:25.564 -> MQTT >> OK >> [r] af7cd12a/stt/UT	 >> true
08:34:25.564 -> MQTT >> OK >> [r] af7cd12a/stt/UC	 >> true
08:34:25.564 -> MQTT >> OK >> [r] af7cd12a/stt/SE	 >> 175
08:34:25.564 -> MQTT >> OK >> [r] af7cd12a/stt/SS	 >> 50
08:34:25.564 -> MQTT >> OK >> [r] af7cd12a/stt/SQ	 >> X
08:34:25.564 -> MQTT >> OK >> [r] af7cd12a/stt/BE	 >> 192
08:34:25.564 -> MQTT >> OK >> [r] af7cd12a/stt/CE	 >> true
08:34:25.564 -> MQTT >> OK >> [r] af7cd12a/stt/CC	 >> 0
08:34:25.618 -> MQTT >> OK >> [r] af7cd12a/stt/CO	 >> 0
08:34:25.618 -> MQTT >> OK >> [r] af7cd12a/stt/CK	 >> 0
08:34:25.618 -> MQTT >> OK >> [r] af7cd12a/stt/NC	 >> 0
08:34:25.618 -> MQTT >> OK >> [r] af7cd12a/stt/SC	 >> 5
08:34:25.618 -> MQTT >> OK >> [r] af7cd12a/stt/C1	 >> 255,255,255
[... 106 lines cut ...]
08:34:26.166 -> 3fffff10:  3ffe8f56 3fff1fba 3fff1fc8 3fff2598  
08:34:26.219 -> 3fffff20:  00003980 00000000 3fff1fc8 4023d242  
08:34:26.219 -> 3fffff30:  00427c72 00004000 3fff6524 4023d27c  
08:34:26.219 -> 3fffff40:  3fffdad0 00000000 3fff1fc8 40232293  
08:34:26.219 -> 3fffff50:  3fffdad0 00000000 3fff1fc8 40233d48  
08:34:26.219 -> 3fffff60:  3fffdad0 00000000 3fff2b4c 402411f5  
08:34:26.219 -> 3fffff70:  0019001f 80efeffe 3fff2344 4023f746  
08:34:26.219 -> 3fffff80:  00000152 00000000 3ffef650 3fff2598  
08:34:26.219 -> 3fffff90:  3fffdad0 00000000 3fff2558 40230689  
08:34:26.219 -> 3fffffa0:  3fffdad0 00000000 3fff2558 4023d324  
08:34:26.219 -> 3fffffb0:  feefeffe feefeffe 3ffe85f0 40101595  
08:34:26.219 -> <<<stack<<<
08:34:26.219 -> 
08:34:26.219 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
08:34:26.266 -> 
08:34:26.266 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
08:34:26.266 -> 
08:34:26.266 -> load 0x4010f000, len 3584, room 16 
08:34:26.266 -> tail 0
08:34:26.266 -> chksum 0xb0
08:34:26.266 -> csum 0xb0
08:34:26.266 -> v2843a5ac
08:34:26.266 -> ~ld

плеер не доступен потому что плата сейчас голая к компьютеру подключена

Simafor47, 2021-11-28:
с памятью вроде тоже норм..
[image]

Simafor47, 2021-11-28:
это кстати после того как вчера новые файлы скачал и полностью поменял везде(ардуино иде,скетч,библиотеки и т.д.)

Simafor47, 2021-11-28:
выключил в скетче плеер(USE_MP3 0) и стал работать нормально. Вот что ему не нравится?)

vvip-68, 2021-11-28:
Если ESP8266 -  возможно не хватает памяти.  Поставьте в 895 строке файла a_def_soft.h 
```
#define    DEBUG_MEM 1
```
Контроллер при смене эффектов будет выводить сколько памяти доступно.
Я MQTT только на ESP32 использую - там памяти много.
Хотя, вроде пишет 20200 байт доступно - этого должно быть достаточно...

Я сейчас переписываю скетч под использование библиотеки плеера от Makuna - может с ней будет лучше работать...

vvip-68, 2021-11-28:
> это кстати после того как вчера новые файлы скачал и полностью поменял везде(ардуино иде,скетч,библиотеки и т.д.)

А до этого какая версия стояла?
В последних файлах, вроде, ничего серьезного не менялся - поменялся выбор скорости в эффекте "Узоры" в смешанном варианте движения, Исправлен символ доллара в шрифтах, исправлено позиционирование вывода календаря и эффект "Кубик".  Там нет серьезных изменений, способных повлиять на падение скетча...

vvip-68, 2021-11-28:
@Simafor47, я переделал в проекте работу с плеером на библиотеку от Makuna, изменения выложил. Попробуй как будет работать эта версия скетча...

Simafor47, 2021-11-28:
библиотеку DFRobotDFPlayerMini удалять?

Simafor47, 2021-11-28:
12:39:56.496 -> 
12:39:56.496 -> WiFiPanel v.1.12.2021.1128
12:39:56.496 -> 
12:39:56.496 -> 
12:39:56.496 -> Инициализация файловой системы... 
12:39:56.496 -> FS: Использовано 16384 из 2072576 байт
12:39:56.496 -> Инициализация EEPROM...
12:39:56.550 -> Сохранение строк выполнено.
12:39:56.550 -> Ипользованы адреса EEPROM 1000 - 1037
12:39:56.550 -> Свободно ячеек 3058
12:39:56.550 -> Настройки сохранены в EEPROM
12:39:56.550 -> 
12:39:56.550 -> 
12:39:56.550 -> Host: 'WiFiPanel' >> 26x13
12:39:56.550 -> 
12:39:56.650 -> DFPlayerError:Ошибка стека!
12:39:56.897 -> 
12:39:56.897 -> Инициализация MP3 плеера.
12:39:57.452 -> Звуков будильника найдено: 8
12:39:57.452 -> Звуков рассвета найдено: 8
12:39:57.452 -> Звуков сообщений найдено: 8
12:39:57.652 -> 
12:39:57.652 -> Подключение к XXX -> 192.168.2.160
12:39:57.752 -> .............
12:40:01.762 -> WiFi подключен. IP адрес: 192.168.2.160
12:40:01.762 -> 
12:40:01.762 -> UDP-сервер на порту 2390
12:40:01.762 -> 
12:40:01.762 -> Режим работы: АВТОНОМНЫЙ, синхронизация E1.31 отключена
12:40:01.762 -> 
далее подключение к MQTT,затем исключения..
[image]
.

vvip-68, 2021-11-28:
Как ни странно, но и в прежней версии библиотеки и в новой версии у вас падает на выполнении ОДНОЙ И ТОЙ ЖЕ команды, а именно:
```
_serial->available()
```
_serial - это объект, создаваемый при инициализации SoftwareSerial из одноименной библиотеки.
Вы исправляли в коде что-нибудь связанное с инициализацией SoftwareSerial и самого MP3 плеера?
Впечатление, что у вас или SoftwareSerial Не инициализирован или не передается объекту плеера...

Какая версия ядра ESP8266 у вас стоит?

playeeerua, 2021-11-28:
Такс, оба варианта сделал. ВЫо втором варианте теперь плата моргает каждые 10 секунд и вот что выдает в мониторе порта 
[image]

vvip-68, 2021-11-28:
@playeeerua - вашу ветку перехватили и забили другими вопросами. Я уже не помню что за варианты.
Что я вижу тут: постоянно циклически перегружающийся микроконтроллер.
Причина перезагрузки - cause:2 - аппаратный сброс по пину rst микроконтроллера. Он у вас, похоже, постоянно замкнут на землю. Проверяйте пайку и сборку. Нет ли какой сопли.

У вас плата вообще к чему-нибудь подключена или вот голая плата и только USB шнур к компу?

vvip-68, 2021-11-28:
@playeeerua - перечитал ветку. Простой скетч выполняется. Если залить прошивку гирлянды - то не запускается и даже до первых строк скетча не доходит - все время перезагрузка с диагностикой - сброс по кнопке reset. Как то это все странно и необъяснимо... На самой первой фотке - возле кнопки ресет - какая-то большая грязная сопля вокруг элементов...

Simafor47, 2021-11-28:
> Как ни странно, но и в прежней версии библиотеки и в новой версии у вас падает на выполнение ОДНОЙ И ТОЙ ЖЕ команды, а именно:
> 
> ```
> _serial->available()
> ```
> 
> _serial - это объект, создаваемый при инициализации SoftwareSerial из одноименной библиотеки. Вы исправляли в коде что-нибудь связанное с инициализацией SoftwareSerial и самого MP3 плеера? Впечатление, что у вас или SoftwareSerial Не инициализирован или не передается объекту плеера ...
> 
> Какая версия ядра ESP8266 у вас стоит?

версия 2.7.4 как у вас в гайде. Скетч нигде,кроме как настройки матрицы не правил. Пока работает с отключенным плеером. Заказал плеер и платы... Ещё заметил странности с запуском.От компьютера 9 из 10 стабильный запуск,а от БП с выходом 4.8В 1 из 20,остальные с прочерками на экране... Монитор не вывести,т.к. далеко от компьютера

playeeerua, 2021-11-28:
Есть вторая плата, которая не паялась, чистая без соплей, то она реагирует так-же как запаянная. Возможно действительно что-то в коде намудрил, непонятно. Думаю еще прошить в гайверлампу, посмотреть как эта плата отреагирует, но там вроде зеленый провод на д4 запаян, проблемой не станет?

vvip-68, 2021-11-28:
@Simafor47 
>Ещё заметил странности с запуском.От компьютера 9 из 10 стабильный запуск,а от БП с выходом 4.8В 1 из 20,остальные с прочерками на экране...

Так может у вас блок питания отвратительного качества? Шумный или не выдает нужного тока?
Конденсаторы фильтрующие электролит и керамика на питание контроллера и плеера стоят?
Диод на питание контроллера по минусу не стоит часом?

vvip-68, 2021-11-28:
> Думаю еще прошить в гайверлампу, посмотреть как эта плата отреагирует, но там вроде зеленый провод на д4 запаян, проблемой не станет?

D4 - это GPIO2.  GPIO15, GPIO0 и GPIO2 - уровни напряжения на них при старте - определяют режим запуска микроконтроллера. Если при запуске на них установлены неверные уровни (например что-то неправильно подключено) - контроллер вообще может не запуститься. Но если при старте правильные уровни - запустится.

vvip-68, 2021-11-28:
[image]

Вот пины обозначенные синими и зелеными звездочками - если при старте на них неверный уровень - контроллер не запустится (см комментарий соответствующего цвета)

Simafor47, 2021-11-28:
> @Simafor47
>
>
> Так может у вас блок питания отвратительного качества? Шумный или не выдает нужного тока? Конденсаторы фильтрующие электролит и керамика на питание контроллера и плеера стоят? Диод на [...]

БП обычный который в перфорированом корпусе с подводящим резистором, 30А. Настроен на 4.8В. Саму плату пытался запитать от юсб адаптера зарядного, тоже самое. Ему главное стартануть😂 далее работает исправно. Кондеры стоят обычные электролиты на 6.3В 470uf и там и там. Диода и керамики нет. На прошлой прошивке(недельной давности) работало всё четко, правда в тестовом виде( на полу)

playeeerua, 2021-11-29:
Заказал еще одну плату, приедет через дня 2-3. Если и она будет вымахиваться, значит дело в коде, где-то я что-то лишнее добавил/убрал. С 2-3 компов делали и 1 ноут

vvip-68, 2021-11-29:
@playeeerua ну так тогда простой вопрос - а просто скачанный с хранилища и не исправленный скетч, залитый на плату к которой ничего не подключено - как себя ведет? Что вы исправляли в скетче?

vvip-68, 2021-11-29:
@Simafor47 - а вернуть прошивку недельной давности - нельзя для проверки?
За последнюю неделю-две было исправлено - начертание символа '$' в массиве шрифтов, добавлена команда установки яркости при нажатии в приложение на кнопку "Предыдущий эффект" увеличена скорость эффекта "Узор" со смещением в разные стороны, подправлена позиция вывода гола при отображении календаря и в эффекте Кубик одна переменная изменена с беззнакового типа на знаковый. Ну вот вчера переделан DFPlayer на другую библиотеку.

Ни одно из этих исправлений в моем понимании не могло серьезно повлиять на запуск приложения и его падение.

Simafor47, 2021-11-29:
К сожалению старой прошивки не осталось. Когда обновляю, скачиваю новый пакет zip, а старый удаляю. Есть откуда скачать для теста?

vvip-68, 2021-11-29:
> К сожалению старой прошивки не осталось. Когда обновляю, скачиваю новый пакет zip, а старый удаляю. Есть откуда скачать для теста?

А вдруг там в новой что-то такое изменилось, что даже не запустится? И вы остаетесь совсем без ничего. Особенно, если не умеете пользоваться Git`ом. Опрометчиво удалять старые версии пока полностью не проверена наовая.

Гит позволяет "скачать" любую версию, на любой момент разработки проекта. Есть консольные команды - я их тоже не знаю, поскольку консолью не пользовался. У меня стоит утилита SmartGit, которая позволяет помимо всего прочего следить что в каких файлах изменялось и загрузить состояние хранилища на любой момент времени. С ним тоже нужно конечно поразбираться как и что в нем, но если научиться - функции свои он выполняет.

vvip-68, 2021-11-29:
@Simafor47 
Впрочем, всё это легко гуглится
https://coderoad.ru/7832770/Как-получить-определенную-фиксацию-коммит-из-проекта-GitHub

Simafor47, 2021-11-29:
Прошивка от 2 ноября работает исправно с плеером....

playeeerua, 2021-11-29:
@vvip-68 он просит ж 2 файла dev_что-то там, а я понятия не имею, как их создавать)

vvip-68, 2021-11-29:
@playeeerua - попробуйте хотя бы просто прочитать комментарий в тех строчках, в которых он просит эти два файла

[13 more replies]

closed 2021-11-30 by vvip-68