vvip-68/GyverPanelWiFi #165: WiFi

playeeerua, 2021-11-30:
Это снова я :D
Вроде прошилось, спасибо за терпение, теперь с вай-фай. Его почему-то нет. В устройствах его нет (смотрел через 192.168.1.1) Пытался зайти через 192.168.4.1 глухо :( Как точка доступа он тоже не появился. Вот что выдает монитор порта
[image]

playeeerua, 2021-11-30:
Ну хоть не стоим на месте, маленькими шажками продвигаемся))

vvip-68, 2021-12-01:
Тема закрыт - проблем решена?
В логе пишет, что подключилась к локалке, адрес контроллера - 192.168.1.123
Вот этот адрес и нужно использовать в программе для подключения к контроллеру.
Естественно адрес должен быть в той же подсетке, которой управляет роутер.
Например если контроллер подключился с адресом 192.168.**1**.123, а роутер управляет сеткой 192.168.**0**.xx - программа на телефоне не сможет подключиться к контроллеру.

Судя по вашему нику - вы с Украины. Ваше правительство заблокировало российские сервисы типа Яндекс - а именно оттуда контроллер получает погоду. Вы не сможете получать погоду с Яндекса. Именно для таких случаев имеется второй источник погоды - OpenWeatherMap - он включается в приложении на телефоне.

playeeerua, 2021-12-01:
по дефолту стояло 0.90, я поменял на 1.123, потому что роутер управляет сеткой 1.1 в подключенных устройствах в роутере он не появляется. Я смотрел 123 никем не занято в роутере было
В программе на телефоне я вписывал его айпи, но не подключается. 
Та мне бы сначала сделать так, что-бы заработало все, а потом с погодой заморачиваться, но нам эта функция вообще не нужна, ее можна убрать

vvip-68, 2021-12-01:
Сначала тогда добейтесь, чтобы телефон вообще видел контроллер в сети.
Поставьте приложение [Ping](https://play.google.com/store/apps/details?id=ua.com.streamsoft.pingtools) или подобное и добейтесь, чтобы пакеты с телефона до контроллера проходили.

После того как пинг будет проходить - сделайте скриншот монитора порта что происходит в момент попытки подключения телефона к контроллеру.

ferrrum, 2021-12-01:
Роутер RT-GM-2
Подключается через ровно через раз. И не только ваша прошивка. 
Делал отдельный скетч для WiFi. Всё равно с первого раза не хочет подключатся.

При раздаче WiFi с телефона 100% подключение.

Проблему решил программным ресетом. 
if (!wifi_connected && !stop_waiting){ESP.restart();}
После перезагрузки всегда подключается.
Но тут если не будет знакомой WiFi сети. Будет постоянная перезагрузка.

Помогите с записью в eeprom подсчёта количества перезагрузок.

vvip-68, 2021-12-01:
@ferrum - а может роутер поменять на нормальный?
А что не я сно с записью в EEPROM? Там вест файл eeprom.ino - сплошные записи и чтения

ferrrum, 2021-12-01:
Попробуй раздай интернет с телефона. Может проблема как у меня в роутере.

playeeerua, 2021-12-01:
завтра найду телефон на андроиде, отпишу или получилось решить проблему

playeeerua, 2021-12-02:
При попітке подключения телефона, вот что відает монитор порта (Не подключает)
[image]

vvip-68, 2021-12-02:
А если отключить запрос погоды - может он мешает подключению?
Поставьте где-нибудь в setup() после LoadSettings()
set_useWeather(0);

playeeerua, 2021-12-02:
Не совсем понимаю куда вставить. А там нет как например с мп3 use_mp3=1 не 0 поменять и отключаю?)

playeeerua, 2021-12-02:
[image]
Оно?)

Witska, 2021-12-02:
Оно

vvip-68, 2021-12-02:
Вот. Ничего никуда не добавляйте, а здесь поставьте 0

playeeerua, 2021-12-02:
Ееееесть, подключило и не пропадает. Проблема номер 9999. Теперь с матрицей.

playeeerua, 2021-12-02:
На стол не смотрите, это подвал :D
Перепроверили спайку, вроде ок все, нигде не перепутано
[image]

playeeerua, 2021-12-02:
Я хочу проверить и поставить другую матрицу что-бы понять в ней ли дело. Нашел схему 16х16, вижу что зеленую надо в дин паять, а те 3 провода , подсоеденитьне трогать и просто еще один в дин допаять?
[image]

playeeerua, 2021-12-02:
[image]

playeeerua, 2021-12-02:
[image]

vvip-68, 2021-12-02:
Зеленый слева - DIN - к выходу управляющего сигнала с контроллера. Белый - к минусу питания контроллера (и блока питания). Красный - к плюсу питания

vvip-68, 2021-12-02:
А куда вообще все это подключено? На фото выше виден угол микроконтроллера, и пин на который идет сигнал - D2 никуда не подключен.

playeeerua, 2021-12-02:
У меня ж их 3 теперь :D
К 3-му подключен (с резистором 220 Ом походу, уже не помню)

vvip-68, 2021-12-02:
И питание бы вам к КАЖДОМУ куску ленты параллельно подводить. А то тоненькие проводнички ленты и тоненькие проводочки между кусками ленты - не додут нормально работать. Конец ленты будет еле светиться из за падения напряжения.

playeeerua, 2021-12-02:
А можно как-то полностью проверить или светит вся лента?

playeeerua, 2021-12-02:
что ж, буду паять. Скоро увидимся :D

vvip-68, 2021-12-02:
> А можно как-то полностью проверить или светит вся лента?

Залейте тестовый скетч. который поочередно зажигает каждый светодиод ленты. Где-то в примерах FastLed были подобные простенькие скетчи.

simonovich, 2021-12-03:
А я не могу прописать устройство в свою wifi сеть, т.к. мобильное приложение не предоставляет вкладки для настройки подключения.
См. https://disk.yandex.ru/i/_B1taETPtS-lXw

vvip-68, 2021-12-04:
Потому что никто не любит читать инструкции, зато любят мучиться. Прям мазохисты какие-то.
Для настройки матрицы (в т.ч. настройки подключения к сети) рядом лежит приложения PanelSetup

simonovich, 2021-12-04:
Во первых, я читал инструкцию:
https://github.com/vvip-68/GyverPanelWiFi/wiki/%D0%9D%D0%B0%D1%81%D1%82%D1%80%D0%BE%D0%B9%D0%BA%D0%B0-%D0%BF%D0%B0%D1%80%D0%B0%D0%BC%D0%B5%D1%82%D1%80%D0%BE%D0%B2-%D0%B2-%D0%BF%D1%80%D0%B8%D0%BB%D0%BE%D0%B6%D0%B5%D0%BD%D0%B8%D0%B8
Во-вторых, у меня свежая версия софта, в том числе все 3 apk установлены.
И прочеканы.
Когда захожу в panel setup, то вижу только panel setup. А зачем оно мне? Тем более, если настройки матрицы подгружены в отдельном файле с профилем устройства.

vvip-68, 2021-12-04:
Скриншот этого приложения настройки можно?
На розовую кнопку с ключиком вверху экрана в синей полоске нажимать пробовали?

simonovich, 2021-12-04:
Новости с полей.
Нашёл кнопку подключения в Panel Setup, я реально её просмотрел. Вписал сеть и свободный ip. Созранить - перезагрузка.
После этого приложенька Wifi Panel зашла по новому адресу. А вот Panel Setup по новому адресу не заходит. По старому, понятное дело, тоже.

В мониторе порта 
```
Подключение к MyWiFi -> 192.168.5.9
.............
```

Примерно через полчаса этот лог вешает Arduino IDE и я его убиваю.

В общем, это уже что-то другое. Я буду с этим работать, если совсем плохо станет - напишу.
Спасибо.

vvip-68, 2021-12-04:
Там у каждого приложения - своя настройка адреса к какому IP подключаться. Там исправлено на новый адрес?

simonovich, 2021-12-04:
Исправлено.

closed 2021-12-04 by vvip-68