vvip-68/GyverPanelWiFi #166: Предложение по эффектам.

kost111233, 2021-12-01:
Удалите эту запись, если уже подобное где-то есть. Предлагаю делиться интересными и красочными анимациями сделанными в программе jinx.

247760, 2021-12-04:
Всё равно топик закроется, напишу тут. Небольшая ошибка в ссылке, ведёт не туда, думаю и другие ведущие на этот сайт не работают.
[image]

closed 2021-12-05 by kost111233

7918514, 2021-12-05:
Я Джинксом граблю видео с ютюба. Нахожу какую-нибудь анимацию компьютерную и в Джинксе включаю захват экрана. Некоторые получаются  даже ничего...
https://youtu.be/pwcmANSvtu4
https://youtu.be/bx-0YlFprqc
https://youtu.be/msnYBMdaB08
ну и так далее, справа в подборке ещё много вылазит.

247760, 2021-12-05:
@7918514 О, спасибо! Лень было что то добавлять, но теперь сделаю, получается действительно круто.
Правда не знаю как наглядно помочь другим, небольшим мануалом.

В двух словах, начало такое, а дальше как и с другими вариантами в JINX:
Предварительная базовая настройка под Вашу гирлянду. И погнали.
1. Выбрать режим **Capture Screen**
2. Сразу ниже: **Edit** 
3. Появится окошко в котором нужно нажать : **Select screen region**
4. Появится новое окошко: **Capture Screen** с кнопкой **Close**
5. Это окошко тянется за углы, это и есть площадь видеозахвата, т.е. как только вы растянете его на размер окна ролика с Ютуба или видеоплеера, жмёте **Close**, окошко пропадёт. 
[image]
6. Вы увидите в программе (окно с размерами Вашей гирлянды) то что она захватывает. Саму JINX утяните вниз экрана, чтобы она не попадала в видеозахват.
[image]
Вот ещё видео, с отображением видеозахвата на панели в режиме реального времени
[image]

Тапками не кидайте, как смог побыстрому.

vvip-68, 2021-12-05:
@7918514 А с учетом, что в последней версии можно с джинкса без сохранения в файл напрямую на матрицу транслировать, а видео - по 6 часов непрерывных эффектов - вообще круть получается!

sepulka123, 2021-12-11:
Как это "... с джинкса без сохранения в файл напрямую на матрицу транслировать..." ?  Можно подробнее, пожалуйста, где почитать (и, хорошо бы, по русски).

vvip-68, 2021-12-12:
[Внешние эффекты Jinx!](https://github.com/vvip-68/GyverPanelWiFi/wiki/Синхронизация-устройств#Внешние-эффекты-jinx)

sepulka123, 2021-12-12:
Огромное СПАСИБО !
Попробую во всем этом разобраться.

HuskySmile, 2023-11-09:
Добрый вечер. Пытаюсь найти информацию как транслировать видео напрямую на матрицу, но ссылка оставленная в этой теме 

> Настройка внешнего источника синхронизации на примере программы управления световым оборудованием "Jinx!" описана в этой [статье](https://github.com/vvip-68/GyverPanelWiFi/wiki/%D0%9D%D0%B0%D1%81%D1%82%D1%80%D0%BE%D0%B9%D0%BA%D0%B0-%D0%9F%D0%9E-%D1%83%D0%BF%D1%80%D0%B0%D0%B2%D0%BB%D0%B5%D0%BD%D0%B8%D1%8F-%D1%8D%D1%84%D1%84%D0%B5%D0%BA%D1%82%D0%B0%D0%BC%D0%B8-%22Jinx!%22-%D0%BA%D0%B0%D0%BA-%D0%B8%D1%81%D1%82%D0%BE%D1%87%D0%BD%D0%B8%D0%BA%D0%B0-%D1%81%D0%B8%D0%BD%D1%85%D1%80%D0%BE%D0%BD%D0%B8%D0%B7%D0%B0%D1%86%D0%B8%D0%B8-%D1%83%D1%81%D1%82%D1%80%D0%BE%D0%B9%D1%81%D1%82%D0%B2-%D0%B2-%D0%B3%D1%80%D1%83%D0%BF%D0%BF%D0%B5).

ведет на главную страницу. Может я тупо найти не могу в ней эту тему. 
Подскажите пожалуйста в каком посте есть инструкция по настройке Jinx для трансляции видео на матрице в режиме реального времени

7918514, 2023-12-11:
@HuskySmile, двумя постами выше рабочая ссылка