vvip-68/GyverPanelWiFi #167: Зависает плата Esp32 на определенных эффектах

promo148, 2021-12-01:
Плата зависает (в монитор порта ничего не выводит, от WiFi отваливается) на эффектах Змейка и Арканоид

Эффект Палитра вызывает ребут платы. В мониторе порта:
[image]

Пробовал на нескольких платах Esp32 (разное количество пинов), разное питание (от usb-компа, powerbank, от БП для питания ленты на 4,85 вольта).
Может есть какие идеи почему так происходит?
Или подскажите как вырезать данные эффекты. 

Версия ядра ESP32 - 1.0.6
Версия FastLED - 3.4.0
WiFiPanel v.1.12.2021.1127

[image]
[image]

vvip-68, 2021-12-01:
Нет, не сталкивался с таким. Все указанные режимы у меня стабильно работают....
Вырезать - просто удалить их из кода и все ссылки, что на них завязаны. 
Но проще в приложении поставить галочки на этих эффектах - "Не использовать".

Если галочки поставить не удается - можно в скетче где-нибудь в setup() после вызова LoadSettings() написать
putEffectUsage(MC_PALETTE, false);
putEffectUsage(MC_SNAKE, false);
putEffectUsage(MC_ARKANOID, false);

kost111233, 2021-12-02:
Попробуйте библиотеку FastLED не 3.4.0

vvip-68, 2021-12-02:
А каким MQTT брокером пользуетесь?
Я вот смотрю по скриншотам - если дефолтным mqtt.by у вас топик MQTT совершенно неправильно настроен. Это, кстати, тоже может вызывать проблемы и отвалы.

promo148, 2021-12-02:
> Попробуйте библиотеку FastLED не 3.4.0

Попробую вечером все версии

promo148, 2021-12-02:
> А каким MQTT брокером пользуетесь? Я вот смотрю по скриншотам - если дефолтным mqtt.by у вас топик MQTT совершенно неправильно настроен. Это, кстати, тоже может вызывать проблемы и отвалы.

Упс. А что не так?
Я настроил не для mqtt.by, я локально у себя на сервере с Home Assistant, поднял стандартный Mosquito mqtt. Соответственно взял интеграцию под HA 
https://github.com/tarasifua/vvip-68-matrix-home-assistant-integration

vvip-68, 2021-12-02:
Ну для не mqtt.by - наверное нормально 
На mqtt.by 16 ноября они поменяли формат топика.
У меня пока нет времени исправить в инструкции, но, в принципе в личном кабинете там все написано.

closed 2021-12-04 by vvip-68