vvip-68/GyverPanelWiFi #168: Какой версией FastLED прошить, чтобы предотвратить хаотичные мерцания по всей матрице?

AndreyLyamkin, 2021-12-02:
Как можно устранить хаотичные вспышки на гирлянде?
Пробовал FastLED понижать до 3.3.3, 3.3.2... 
Может есть патч какой-то для ESP8266 (NodeMCU 1.0(ESP-12E Module))&

vvip-68, 2021-12-02:
Хаотичные вспышки на гирлянде можно устранить только настройкой напряжения питания системы. Никакие изменения версии FаstLED не помогут в принципе.
Если схема собрана без поддерживающего диода по минусу питания контроллера - напряжение питания системы не должно быть выше 4.8 вольта. Если напряжение питание в диапазоне 5.0-5.8 вольт - нужно ставить поддерживающий диод на минус питания контроллера.
Естественно источник питания не должен должен иметь шумов на выходе, на питании контроллера очень желательно ставить электролитический конденсатор на 1000-2000 мкф и параллельно ему керамический конденсатор на 100 нф.

closed 2021-12-02 by vvip-68

AndreyLyamkin, 2021-12-02:
Спасибо за совет!!!
попробовал понизить даже до 4.0 вольт... Ничего не изменилось. 
Но! Я в прошлом году собирал на этом же железе, на этих же светодиодах.... И не было мерцаний.

vvip-68, 2021-12-02:
Ядро контроллера какой версии стоит?

AndreyLyamkin, 2021-12-02:
Это удар в спину ))))
Как проверить? и как обновить???

vvip-68, 2021-12-02:
В Arduino IDE - Инструменты - Платы - Менеджер плат - ESP8266
Версия должна быть 2.7.4

AndreyLyamkin, 2021-12-02:
Была 3.0.2
Ставлю 2.7.4 и шьюсь!

247760, 2021-12-02:
Кусочек из самого начала GyverPanelWiFi_v1.12.ino
`
// -------------------------------------------------------------------------------------------------------

// На версии ядра ESP8266 v3.x.x и FastLED 3.4 работать не будет! 
//  -> Постоянно горит самый первый светодиод в цепочке синим или зелёным
//  -> некоторые эффекты сильно мерцают.
//  -> пин ленты с D2 переназначается ядром на D4
// На версии ядра ESP32 1.0.5 или 2.x.x и FastLED 3.4 работать не будет! 
//  -> 1.0.5 - При USE_E131 == 1 - при переключении с MASTER на STANDALONE или SLAVE - crash  и перезагрузка. 
//             Для исправления - в файле C:\Users\<user>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\AsyncUDP\src\AsyncUDP.cpp
//             найти деструктор AsyncUDP::~AsyncUDP() и перенести вызов close(); из начала процедуры в конец  
//  -> 2.0.0 - Нет вывода изображения на ленту ни на одном из пинов

// Версия ядра ESP8266 - 2.7.4    
// Версия ядра ESP32   - 1.0.6
// Версия FastLED      - 3.4.0

// -------------------------------------------------------------------------------------------------------
`

AndreyLyamkin, 2021-12-02:
я уже откатился на прошивку 1.0.8....

247760, 2021-12-02:
@AndreyLyamkin раз осилили 1,08 в своё время, то потратьте 5 минут и настройте крайнюю версию, всё самое важное что надо прочитать (https://github.com/vvip-68/GyverPanelWiFi/issues/168#issuecomment-984744642) , сделать и перенести старые данные.

vvip-68, 2021-12-02:
@AndreyLyamkin - вы не только не читаете комментарии в скетче, но даже не читаете то, что вам отвечают...
Так что - откатиться на 1.0.8 - самое верное решение. Но оно скорее всего тоже не заработает по той же самой причине.
> // На версии ядра ESP8266 v3.x.x и FastLED 3.4 работать не будет!  
 //  -> Постоянно горит самый первый светодиод в цепочке синим или зелёным 
 //  -> некоторые эффекты сильно мерцают. 
 //  -> пин ленты с D2 переназначается ядром на D4

**пин ленты с D2 переназначается ядром на D4**

AndreyLyamkin, 2021-12-02:
Да как же! Я же  в самом начале написал, что FastLED поставил != 3.4
Я пробовал все вниз: 3.3.3, 3.3.2
И все вроде как работает! Только эти хаотичные вспышки... И питанием не помогает убрать..

Сейчас прошился на крайний релиз, все как по Вашим подсказкам. 
Но теперь другая беда - к Wi-Fi подключается, а вот дальше никуда, ни адрес NTP не получает, ни погоду никак не получит...
ДА!!! Пробовал и на рабочем роутере, и через Wi-Fi поднятого на телефоне...

vvip-68, 2021-12-02:
Вам же написали - вспышки - из за ядра 3.0.2. При назначении "LED_PIN 2" это ядро садит выход на D4.
Рабочее ядро - 2.7.4. При этом, если ничего не менять в настройках - при том же "LED_PIN 2" ядро направид выход сигнала на D2.

Про WiFi - смотрите правильно ли настроена сеть, находится ли устройство в той  же сети, что и роутер. Смотрите в мониторе порта на какой адрес подключается и подключается ли вовсе. Ну и далее по алгоритму...

AndreyLyamkin, 2021-12-06:
Вы меня простите. Я Вас обоих понимаю, сам работаю программистом и по совместительству сисадмином. И повторять одно и тоже этим тупым пользователям... Которые не читают инструкции... В общем, без обид!
Теперь по существу:

Чтобы я не делал, как бы не варьировал версии ядра и библиотеки FastLED, мой контроллер не работает с последней прошивкой! А именно: в мониторе порта тупо пишет, что подключился к сети, но в логах роутера (Mikrotik) пусто!, так же не пингуется плата вообще никак. Ну, и соответственно, ни время, ни погоду получить из интернета не может.
Это первое...  Было бы интересно понять, почему так происходит, тем более я не один [такой](https://github.com/vvip-68/GyverPanelWiFi/issues/165). 

Я восстанавливаю 1.0.8 работала всю с декабря по май без проблем.... А вот в этом году почему-то хаотичные вспышки.
Блок питания у меня вполне норм, 5V30А... такой железный "в дырочку"... И питание я понижал до 4 вольт, и кондеры ставил 5.3V1000mF... И диод, чтобы подтянуть напругу.... Ничто это не помогло убрать хаотичные вспышки...

Но я победил этот недуг:
1. В библиотеке FastLED ищем файл "\FastLED\platforms\esp\8266\clockless_esp8266.h"
2. В нем ищем строку "//template <int DATA_PIN, int T1, int T2, int T3, EOrder RGB_ORDER = RGB, int XTRA0 = 0, bool FLIP = false, int WAIT_TIME = 50>"
3. Меняем ее на "//template <int DATA_PIN, int T1, int T2, int T3, EOrder RGB_ORDER = RGB, int XTRA0 = 0, bool FLIP = false, int WAIT_TIME = 5>"   т.е. в конце 50 меняем на 5. Возможно частые отключения от Wi-Fi, но у меня за час "НЕ БЫЛО НИ ЕДИНОГО РАЗРЫВА" )))

Далее... Во всех файлах скетча ищем:
Строку #include <FastLED.h> и перед ней, выше пишем: #define FASTLED_ALLOW_INTERRUPTS 0 (это в одном файле только)
Далее ищем строку "FastLED.show()" и после каждого вхождения с новой строки пишем:  yield();

После прошивки того-же контроллера все артефакты как ветром сдуло!
Может кому-то еще поможет...

KAKTO, 2021-12-09:
Была такая проблема с хаотичными вспышками на гирлянде  500 Leds 25 метров - поставил диод (1N4007) - проблема ушла. Думаю наводка по проводам (провода не скручивал). Но ближе к этой проблеме новая матрица 16х16, которую  купил. Собрал, подключил - хаотичные вспышки. Подключил старую матрицу - работает. Поставил диод, подключил новую матрицу - работает. 
Правда собрано на Wemos
FastLED version 3.003.003
LED_PIN 3
PIN_BTN D4