vvip-68/GyverPanelWiFi #169: Коды городов для погоды от Яндекс

vvip-68, 2021-12-02:
Яндекс поменял вывод поисковой страницы и теперь способ узнать код города, описанный в инструкции не работает - теперь в поисковой строке не код города, а координаты.

Предлагается здесь составить таблицу известных кодов городов.
Вдруг пригодится тому, кто живет в вашем городе, но не успел узнать код города, а теперь уже и не понятно как...

> Красноярск - 62

247760, 2021-12-02:
@vvip-68 специально проверил с 3-х браузеров, все показали одинаковый правильный код города. Надо подождать кто ещё отпишется, есть ли такая проблема.

vvip-68, 2021-12-02:
@247760 - мне уже отписывались. В тот день не работало и мы так и не смогли определить код города вопрошающего.
Тогда текст в адресной строке браузера выглядел примерно так:
`https://yandex.ru/search/?text=&lat=55.980572&lon=92.882607`

Сейчас проверил - снова выглядит как
`https://yandex.ru/search/?text=&lr=62`

Наверное опять назад вернули. Ну что ж - коли так - можно закрывать тему

closed 2021-12-02 by vvip-68

7918514, 2021-12-05:
Я дико извиняюсь, что влез в ваш разговор, но при нажатии на кнопку "Найти" (как это написано в wiki) в яндексе при пустом поле ничего не происходит. Нужно ввести любой символ или пробел, тогда всё получится. Испытал на двух браузерах.

> Москва - 213

CrazyGhostRider, 2021-12-21:
[Регионы в Яндексе.xlsx](https://github.com/vvip-68/GyverPanelWiFi/files/7757300/default.xlsx)
Извините, что поднял тему. Но нашел, наверное, более точный список кодов.

Ещё можно добавить в вики ссылку https://yandex.ru/time/sync.json?geo=213
Чтобы быстро проверить, а правильно ли нашёл код своего города.