vvip-68/GyverPanelWiFi #170: Иконки погоды

ferakond, 2021-12-04:
Здравствуйте, а как можно настроить отображение иконок погоды ? Или это от региона погоды как то зависит? Погоду показывает через макросы и на часах

vvip-68, 2021-12-04:
Иконки погоды лежат в файле weather.h
Формат тот же, что у анимации - RGB565, левый верхний угол картинки слева вверху массива.

В файле weather.ino - процедура получения номера картинки в массиве по коду погоды, полученному от погодного сервера. getWeatherFrame() - для Яндекса, getWeatherFrame2() - для OpenWeatherMap

closed 2021-12-04 by vvip-68

ferakond, 2021-12-05:
А есть какой-то подробный мануал? Мне кажется я что-то пропустил)

vvip-68, 2021-12-05:
Специального мануала по замене картинок погоды нет. Это уже такие программистские дебри, что мало кому интересно. Как выбирается картинка из массива картинок погоды - описано выше. Как создаются картинки в коде в формате RGB565 - в принципе можно почитать статью wiki, посвященную утилите [JinxFramer](https://github.com/vvip-68/GyverPanelWiFi/wiki/JinxFramer---редактор-анимации,-роликов,-картинок#Экспорт-кадра-в-код-скетча) в разделе "Экспорт кадра в код скетча", нарисовать картинки, экспортировать их и заменить в weather.h картинки своими. 

Картинки погоды - не анимация, просто картинка. Более-менее приличная анимация -
а) требует достаточно много памяти контролера (там же много кадров нужно на одну картинку)
б) а кто ее рисовать будет? - я не художник.

Если вы художник и нарисуете красивые картинки - поделитесь, будет красиво у всех.

ferakond, 2021-12-05:
Я прошу прощения, я наверное не так вопрос сформулировал) Я имел ввиду на роликах где использовали вашу прошивку в строке погоды после текста на матрице отрисована иконка погоды)
А на счёт картинок я попробую)

vvip-68, 2021-12-05:
А это точно моя прошивка? Может кто переделывал? У меня такой возможности нету.
Ролик этот посмотреть можно?

ferakond, 2021-12-05:
https://youtu.be/1qtvohFDlQc на восьмой минуте
Там правда просто иконка погоды и температура

Может м вправду в другой прошивке видел

vvip-68, 2021-12-05:
Ну вот та иконка что на восьмой минуте - это оно и есть, всё о чем описано выше.

ferakond, 2021-12-05:
А подскажите пожалуйста как этот эффект включить.
То что во вкладке  погода просто по очереди показывает иконки погоды

vvip-68, 2021-12-05:
Очевидно что у вас не получены сведения о погоде с погодного сервера и ему просто нечего показывать. Аот он по очереди все и показывает. Почему нет сведений о погоде? Ну, это вас спрашивать нужно.
- Или эта возможность выключена в настройках прошивки - стоит USE_WEATHER 0
- Или выключено получение в настройках (в программе) - тип погодного сервера "Ничего"
- Или нет соединения с интернетом (или сервер заблокирован, например как Яндекс в Укараине)
- Или еще что-то...

Как только контроллер получит сведения о погоде - он на этом эффекте будет показывать текущую погоду, а не перебирать картинки по циклу...

ferakond, 2021-12-05:
Ну на счёт не настроен я писал, что в бегущей строке если макросом то показывает и на часах температуру 
В программе выставлено все. Ладно буду в этой области ковырять) Спасибо за помощь и за мотивацию на развитие))

vvip-68, 2021-12-06:
Ну по другому не бывает. Есть погода - показывает ее. Нет погоды - "крутит" иконки по циклу.
Я говорю об оригинальной прошивке. Если кто-то что-то там переписывал - тогда я не знаю.