vvip-68/GyverPanelWiFi #171: Низкий FPS трансляции из Jinx

timof123, 2021-12-04:
Гирлянда 10*15, nodemcu. По инструкции настроил и подключил программу jinx к гирлянде. Эффекты транслируются, но на гирлянде низковат fps. Можно как-то сделать выше fps?

vvip-68, 2021-12-04:
Ну смотрите - fps - сколько раз в секунду обновляется отображение на матрице.
Если матрица (каждый новый фрейм) отображает 5 фреймов в секунду - будет 5fps.
Хотте чтобы за секунду отображалось 40 фреймов (40 новых кадров) - увеличьте скорость эффекта.
Попробуйте на эффекте "Узоры" например, сдвинув ползунок скорости то упора вправо.
Вы получите fps около 40-50. Но эффект вы смотреть не сможете - будет сплошное мельтешение, не различимое глазом.
Уно вам надо? Если да - выкручивайте скорость эффекта.

То же самое с эффектами Jinx. Насколько я помню - Jinx имеет фиксированный FPS 25.
Значит при захвате кадра и сохранении его в файл будет формироваться 25 кадров в секунду.
НЕ знаю есть ли возможность повысить FPS у Jinx! - гуглите.

Далее скорость эффекта в скетче определяет как часто будет браться очередной кадр из файла с SD карточки и передаваться на матрицу. Это определяет FPS вывода картинки на матрицу. Увеличивайте скорость эффекта (ползунок настройки скорости вправо) - больше FPS 0 большее количество кадров за секунду будет выведено на матрицу.

Естественно есть аппаратные ограничения - скорость процессора микроконтроллера (сколько времени занимает один цикл обработки), с какой скоростью происходит считывание кадра с SD-карты, скорость вывода пакета на матрицу для диодов WS2812 (частота - 800Кгц

Можно попробовать 
а) Поднять частоту NodeMCU с 80 до 160 МГц - настройка в меню "Инструменты" Ардуино ИДЕ
б) Взять другой микроконтроллер, например ESP32
в) Взять SD карту с бОльшей скоростью считывания (но это может быть ограничено тем , с какой скоростью вообще может читать SD-Shield
г) Взять ленту с бОльшей скоростью передачи сигнала - например WS2813 или APA102 - у них скорость 2000КГц

closed 2021-12-04 by vvip-68

vvip-68, 2021-12-04:
Вот ситуация для матрицы 16x16
[image]