vvip-68/GyverPanelWiFi #172: подключение к mqtt.by из apk

limittouch, 2021-12-04:
гирлянда смотрит в мир через роутер и успешно подключается к брокеру mqtt.by без проброса портов
```
17:58:23.420 -> Подключаемся к MQTT-серверу 'mqtt.by:1883'; ClientID -> 'WiFiPanel-be12' ...
17:58:23.607 -> Подключение к MQTT-серверу выполнено.
```
приложение GyverPanelWiFi.apk с теми же настройками по MQTT-каналу не подключается

заменить в инструкции подключения к mqtt.by формат префикса на 
user/<аккаунт>/<уникальный идентификатор>

4 часа убил, пока issues/167 не прочитал :)

vvip-68, 2021-12-04:
Ну нет у меня пока времени поправить инструкцию... 
Это только 16 ноября у них поменялось...
Поправлю. Потом.

vvip-68, 2021-12-04:
А то - сделайте доброе дело, обновите текст инструкции, картинок наделайте. 
Пришлите мне md-файл и картинки - я в вики заменю :)

limittouch, 2021-12-04:
скриншотов больно много, а у меня все иначе настроено, не потяну

зато есть рац-предложение:
ничего не менять, а добавить красным в начале главы **MQTT-сервер по-умолчанию** :

**ВНИМАНИЕ! В связи с изменением у брокера mqtt.by формата префикса, в приведенных ниже примерах везде вместо строки "af7cd12a" требуется использовать "user/<имя пользователя>/af7cd12a". Так же требуется задать имя пользователя и пароль.
Подробности настройки подключения см. на странице брокера "Профиль"** 

дешево и сердито
плюс, понятно, что у разных брокеров разный формат префикса может быть

vvip-68, 2021-12-05:
Спасибо за предложение, строку добавлю.
А как в md-файлах делать текст красным?

247760, 2021-12-05:
@vvip-68 
<h2>

```diff
- red
+ green
! orange
# gray
```

</h2>

vvip-68, 2021-12-05:
Спасибо. Костыли, конечно, но...

simonovich, 2021-12-06:
Там мальчики  играются со своим серваком (mqtt.by). Рабочий вход по паролю иногда доступен только на порту 1885. А он с недавнего времени платный! :)
А на бесплатном тестовом 1883, то жесткие требования по паролю / топику, то наоборот пускает по-любому и всех подряд... Также какие-то странные подвисания в ночное время.
Пробуйте.

closed 2021-12-11 by vvip-68