vvip-68/GyverPanelWiFi #173: Отображение к верх ногами

brumel37, 2021-12-04:
добрый день не могу никак настроить гирлянду у меня вемос как бы я не менял тип подключения у меня всегда матрица отображает все кверх ногами. могу предположит что это из за библиотеки литлфс пришлось ее ставить со стороны так которая здесь почему то неподхватыветься.

247760, 2021-12-04:
@brumel37 хоть всё вверх ногами, но отображаются слова "корректно" ?
Эти настройки правильно выставили?
https://github.com/vvip-68/GyverPanelWiFi/blob/100874877d09c1c183670a38ff117b6e71ef47a2/firmware/GyverPanelWiFi_v1.12/a_def_hard.h#L103-L115

vvip-68, 2021-12-04:
LittleFS - библиотека для создания файловой системы в микроконтроллере для хранения настроек подключения к сети и картинок, нарисованных пользователем. И никакого отношения к выводу изображения на матрицу не имеют.

Ориентация изображения на матрице определяется ТОЛЬКО значениями MATRIX_TYPE, CONNECTION_ANGLE и STRIP DIRECTION и больше ни чем. Значения этих констант определяют вывод на матрицу (ориентацию) при ПЕРВОЙ загрузке скетча в контроллер. Или после полного стирания EEPROM - настройка в меню Инструменты - Erase All

В остальных случаях что бы вы тут не меняли - настройки берутся из сохраненных в постоянной памяти EEPROM.

Нужно установить приложение PanelSetup - в папке Android проекта и настроить ориентацию матрицу там.

closed 2021-12-05 by vvip-68