vvip-68/GyverPanelWiFi #174: обрезка списка эффектов SD-карты в apk-панели по WiFi

limittouch, 2021-12-05:
при большом количестве эффектов на SD-карте и **подключении через WiFi**, часть списка эффектов не отображается в приложении GyverPanelWiFi
диагностика по порту показывает правильный полный список

при **подключении через MQTT** список в приложении GyverPanelWiFi отображается полностью

подозреваю, что список с гирлянды передается одной строкой и режется где-то в районе 1000 символов

всего 70 эффектов со смысловой нагрузкой
UDP XXX>> $18 UP:240|EF:44|EN:[SD-Карта]|UE:1|UT:1|UC:1|SE:175|SS:X|BE:X|SQ:L>0>Случайный выбор,Последовательно,Banana.out,Big smile.out,Blue rain.out,Blue tears.out,Brace wheel.out,Brain.out,Brain night.out,Brains green.out,Butterfly.out,Cat.out,Cats eyes.out,Circles.out,Cloudy waffles.out,Connects.out,Corona discharge.out,Crosses.out,Cubes.out,Disco.out,Dog 1.out,Dog 2.out,Explosion.out,Eyes.out,Father Frost.out,Fireworks & glitter 1.out,Fireworks & glitter 2.out,Fireworks 1.out,Fireworks 1-2.out,Fireworks 2.out,Fireworks 2-2.out,Flowers.out,Flowers 2.out,Ghost.out,Holi.out,Kaleidoscope.out,Lines swing.out,Lines swing & circles.out,Lvl equalizer.out,Lvl stereo.out,Matrix.out,Matrix - 4-side.out,Matrix-BG.out,Melting film.out,Octahedrons.out,Paints - collision.out,Paints - explosion.out,Peacock tail.out,Plasma.out,Rainbow.out,RNA.out,Russian flag.out,Saur-brace.out,Saur-circles.out,Saur-squares.out,Shark.out,Silk.out,Sinuses & square.out,Snakes swim.out,Snowflake.out,Spiral happy.out,Spiral hypnoInv.out,Spotlights.out,Square drops.out,Square drops & balls.out,Square spiral.out,Star.out,Stars and squares.out,Stars to left.out,Stars up.out,Surf.out,Waterfall multi.out;

имя последнего поместившегося эффекта "Snowfla**ack45**" :)

имхо:
даже, если обрезать (не передавать) расширения файлов, полезная нагрузка увеличится

vvip-68, 2021-12-05:
Вы совершенно правы. Список не влазит из за размера буфера на отправку сообщения.
Размер буфера выбран из соображений размера выделяемой под буфер памяти микроконтроллера, размера пакета UDP и ограничений компонентов конструктора приложений AppInventor на котором написано приложение.

Передавать список эффектов (файлов на карте, звуков и т.п.) в несколько порций у меня не получилось - почему-то приложение на телефоне видит только то, что ему отправляет контроллер в ответ на запрос приложения. Так , чтобы скетч что-то отправил приложению по своей инициативе  без запроса - у меня не получилось. Приложение не видит таких сообщений. Увы. 

Отправка же списков по одному в алгоритме 
1. приложение спросило сколько будет файлов
2. Скетч ответил
3. приложение в цикле спрашивает "дай файл N"
4. скетч отправляет имя файла N
5. Повторять пока не кончатся

а) выполняется довольно долго - обмен может идти до 3-5 минут и в это время приложение либо не работает, либо все равно оперирует не полным списком
б) в случае потери пакетов (а такое в UDP бывает) - последовательность нарушается
в) добавление в алгоритм подтверждение получения - еще больше удлиняет время обмена

Поэтому был выбран именно такой (реализованный) алгоритм получения списков.
С рекомендацией не давать слишком длинные имена файлам и сокращать их до разумно-понятных, чтобы общий
размер (длина строки) не превышала размер буфера. Ситуация осложняется тем, что строка обмена имеет кодировку UTF-8 - а это ДВА байта на один символ.

Предложение не передавать расширения файлов, кажется разумным. Подумаю над его реализацией.

Вариа

closed 2021-12-05 by vvip-68

limittouch, 2021-12-10:
> Предложение не передавать расширения файлов, кажется разумным. Подумаю над его реализацией.

дополнительное предложение:
формировать, передавать или отображать список эффектов с карты в алфавитном порядке
искать эффекты по дате добавления сомнительное удовольствие

vvip-68, 2021-12-10:
Это технологически невозможно. 
Эффекты используются не по имени, а по индексу.
Приложение передает в контроллер команду "Включить эффект номер N" - где N - индекс в этом списке.
Хотите алфавитную сортировку - вам вручную нужно отсортировать список названий эффектов, который передается в приложение и упорядочить в этом же порядке список констант идентификаторов эффектов MC_XXX

Но и это не во всем будет работать, т.к некоторые эффекты (например "Рассвет" и, кажется, "SD-карта") - должны обязательно быть в конце списка эффектов.

vvip-68, 2021-12-10:
Можно попробовать...
Пришлите мне плиз функцию сортировки строк, работающую на ардуино. 
Дальше посмотрим...

kostyamat, 2021-12-11:
> 
в случае потери пакетов (а такое в UDP бывает) - последовательность нарушается

А вы websocket не пробовали? Попробуйте, вам понравится.

vvip-68, 2021-12-11:
Подскажите компонент вебсокета для AppInventor?
А так же пример работы с ним для ESP8266...