vvip-68/GyverPanelWiFi #175: Внешний файл настроек

Des-75, 2021-12-05:
Здравствуйте. Можно поподробнее вот про этот момент:
```
#define A_DEF_DEVC 1          // 1 - Настройки профилей ваших устройств в отдельном файле a_def_devc.h
                              // 0 - Настройки профиля вашего устройства размещены ниже в этом файле
                              // Файл a_def_devc.h в комплект не входит, нужно создать, скопировать туда настройки профилей ваших устройств
```
Я не понимаю где начинаются/заканчиваются границы профиля, который я должен скопировать во внешний файл (у меня ESP8266).

vvip-68, 2021-12-05:
Оно вам действительно нужно? У вас несколько изделий, содержащих разные размеры матриц / состав подключенных модулей / пины подключения?

Профиль устройства - от начала описания профиля до следующего такого же куска в скетче ниже.
Вот пример описания профиля одного устройства:
```
#define WIDTH 16                // ширина одного сегмента матрицы
#define HEIGHT 16               // высота одного сегмента матрицы
#define MATRIX_TYPE 0           // тип соединения диодов в сегменте матрицы: 0 - зигзаг, 1 - параллельная
#define CONNECTION_ANGLE 1      // угол подключения диодов в сегменте: 0 - левый нижний, 1 - левый верхний, 2 - правый верхний, 3 - правый нижний
#define STRIP_DIRECTION 3       // направление ленты из угла сегмента: 0 - вправо, 1 - вверх, 2 - влево, 3 - вниз

#define META_MATRIX_WIDTH 1     // количество сегментов в ширину сборной матрицы
#define META_MATRIX_HEIGHT 1    // количество сегментов в высоту сборной матрицы
#define META_MATRIX_TYPE 0      // тип сборной матрицы: 0 - зигзаг, 1 - параллельная
#define META_MATRIX_ANGLE 0     // угол 1-го сегмента сборной матрицы: 0 - левый нижний, 1 - левый верхний, 2 - правый верхний, 3 - правый нижний
#define META_MATRIX_DIRECTION 0 // направление следующих сегментов сборной матрицы из угла: 0 - вправо, 1 - вверх, 2 - влево, 3 - вниз

#define DEVICE_TYPE 1           // Использование матрицы: 0 - свернута в трубу для лампы; 1 - плоская матрица в рамке   
#define BUTTON_TYPE 1           // Тип кнопки: 0 - сенсорная (touch button); 1 - обычная кнопка на замыкание при нажатии (или если кнопки нет вообще)  
#define USE_SD 0                // поставьте 0, если у вас нет SD-card модуля
#define USE_TM1637 0            // поставьте 1, если используется дополнительный индикатор TM1637, 0 если индикатора нет
#define USE_POWER 1             // 1 - использовать отключение питания матрицы через MOSFET / реле; 0 - не использовать 
#define USE_MQTT 0              // 1 - использовать управление по MQTT-каналу; 0 - не использовать 
#define USE_WEATHER 1           // 1 - использовать получение информации о текущей погоде; 0 - не использовать 
#define USE_E131 0              // 1 - использовать протокол E1.31 для синхронизации устройств или приема внешнего потока на матрицу с Jinx! или другого аналогичного ПО
#define BIG_FONT 0              // 0 - шрифт 5x8, 1 - шрифт  10x16; 2 - шрифт 8x13

// DFPlayer выпускается разными производителями на разных чипах. 
// Оригинальный плеер построен на микросхеме с обозначением на корпусе AS20HH5883-74
// Существует множество клонов DFPlayer от других производителей, построенных на других чипах, например MH2024K-24SS, MH2024K-16SS и др.
// Эти чипы отличаются от оригинальных несколько измененной системой команд. Библиотеке нужно знать команды в каком формате отправлять плееру.
// Укажите ниже какой чип использован в в вашем DFPlayer

#define USE_MP3 0               // поставьте 0, если у вас нет звуковой карты MP3 плеера
#define DFPLAYER_TYPE 0         // 0 - AS20HH5883-74 (оригинальный); 1 - MH2024K-24SS, MH2024K-16SS (клон)

#define LED_PIN  2              // D2 пин ленты
#define PIN_BTN D4              // D4 кнопка подключена сюда (D4 --- КНОПКА --- GND) 
                                // D6 для схем с DFPlayer    (D6 --- КНОПКА --- GND)
#define POWER_PIN D1            // D1 управляющий пин вкл/выкл матрицы через MOSFET / реле; POWER_ON - HIGH, POWER_OFF - LOW
#define SD_CS_PIN D8            // пин выбора SD карты - ChipSelect; Также SD карта использует D5 (CLK), D6 (MISO), D7 (MOSI)
#define STX D3                  // подключен в RX пин (2) модуля DFPlayer
#define SRX D4                  // подключен в TX пин (3) модуля DFPlayer
#define DIO D5                  // D5 TM1637 display DIO pin - требуется для компиляции скетча (USE_TM1637 == 1)
#define CLK D7                  // D7 TM1637 display CLK pin - требуется для компиляции скетча (USE_TM1637 == 1)

                                // Для USE_SD == 1 - поведение при проигрывании файла эффекта с SD-карты 
#define WAIT_PLAY_FINISHED 1    // 1 - переключаться на следующий эффект только когда весь файл полностью показан; 
                                // 0 - прерывать показ файла с SD-карты по истечении времени эффекта

#define DEBUG_SERIAL 1          // включить/выключить отладочный вывод в Serial: 0 - выключено 1 - включено
```
Ниже повторяется такой же набор полей для другого профиля

ЗЫ: Раньше этот файл, содержащий профили (настройки) МОИХ устройств входил в состав проекта, но пользовали почему-то бездумно выбирали МОЕ устройство и удивлялись что у них ничего не работает. И оно вполне естественно - у них то другие размеры, другие пины подключения, другой набор модулей. Поэтому МОИ настройки были вынесены в отдельный файл, который теперь не входит проект, а пользователю предлагается ИЛИ самостоятельно создать файл и настроить в нем профили множества своих устройств под себя ИЛИ просто изменить настройки в этом (a_hard_def.h) файле в профиле, описанном ниже

closed 2021-12-05 by vvip-68

Des-75, 2021-12-05:
Спасибо.
ЗЫ: Оно мне **действительно** нужно. Ваш проект на месте не стоит, и менять настройки каждое обновление несколько муторно, согласитесь?

vvip-68, 2021-12-05:
1. Тогда могу куда-нибудь прислать мой файл. Сделаете на его основе свой.
2. Потому что народ жаловался, что им мало 16 (или 24?) символов в пароли - у них пароли по 30 символов. Место на длинные строки в EEPROM уже не было (а все настройки там хранились) - пришлось переместить длинные пароли на "диск" - в файловую систему микроконтроллера.  Это было с версии то ли 1.10 то ли 1.11
Если прошивка без выделения места под файловую систему - сохранять настройки подклбчения к сети негде - она каждый раз будет требовать их вводить.

Des-75, 2021-12-05:
1. Спасибо, не надо, уже сделал.
2. Было бы неплохо указать на эту особенность. :)