vvip-68/GyverPanelWiFi #176: Макрос {S}

LisovR, 2021-12-05:
Я не отрицаю что я..., но нифига не понимаю что не работает.
Прошивка 1.12, плата wemos d1 mini.
Прописываю: Тест!!!{S05.12.2021 19:00#05.12.2021 19:05} работает всегда!

vvip-68, 2021-12-05:
Боюсь, что ничем помочь не смогу. Проверил - работает в строго указанный промежуток времени.
То что вы описываете - воспроизвести не удалось, следовательно исправит не могу или исправлять нечего.

closed 2021-12-05 by vvip-68

LisovR, 2021-12-05:
Может что то в прошивке я мог поставить. Редактировал только 2 файла хард и софт. Плата голая, даже кондюка нет(на аэродинамику это не влияет) параметры выставлял с этим учетом.

LisovR, 2021-12-05:
Еще есть фигня, когда с телефона выставляю макрос R и нажимаю тест, то все что левее включая : где написано время стирается. На андроид ставил из файлов с прошивкой.

vvip-68, 2021-12-05:
Ничего не могу сказать по этому поводу.
У меня описанное поведение не воспроизводится.
Если я не могу воспроизвести ошибку - я не могу ее исправить...