vvip-68/GyverPanelWiFi #177: SD

kost111233, 2021-12-07:
Добрый день, подскажите. Решил в схему добавить sd карту. Прошить не удается, сначала были одинаковые библиотеки в разных местах, в итоге все удалил, заново установил библиотеку sd и вот такую ошибку выдает
[image]

vvip-68, 2021-12-07:
Вообще удалите эту библиотеку. Она для ардуино (AVR)б а не для ESP.
У ESP поддержка SD карты на уровне ядра

closed 2021-12-07 by vvip-68

kost111233, 2021-12-07:
[image]
Удалил, теперь ошибка что ее нет.

kost111233, 2021-12-07:
[image]

vvip-68, 2021-12-07:
А версия ядра какая стоит?

kost111233, 2021-12-07:
2.7.4

vvip-68, 2021-12-07:
Вот тут папка с библиотекой SD - она у вас есть?
[image]

kost111233, 2021-12-07:
Я от туда ее тоже удалил)

vvip-68, 2021-12-07:
Потому и не находит

kost111233, 2021-12-07:
Спасибо. Скетч залился