vvip-68/GyverPanelWiFi #178: возможность однократного воспроизведения эффекта

limittouch, 2021-12-07:
мне не нужно зацикливать короткие эффекты с SD-карты
например, приведение вылетело за 5 секунд и пропало, еще раза 4 ему вылетать ни к чему
повторы можно обеспечить длиной записи эффекта

одного прохождения лабиринта или змейки тоже достаточно

предлагаю сделать как-то так

`#define REPEAT_PLAY 0         // 1 - повторное проигрывание текущего эффекта, если время не вышло; 0 - следующий эффект
`

```
    if (((ms - autoplayTimer > autoplayTime) // таймер смены режима
         #if (REPEAT_PLAY == 0)
         // при окончании игры не начинать ее снова
         || gameOverFlag
         #if (USE_SD == 1)
         // если файл с SD-карты проигрался до конца - сменить эффект
         || (thisMode == MC_SDCARD && play_file_finished)
         #endif
         #endif
         )
    && !manualMode) {    
```

нормальное решение или я чего-то не учитываю?

проверка, что воспроизведение началось `(ms > autoplayTimer)` не требуется?

в `sdcardRoutine()` на `play_file_finished `не нужно заморачиваться, автоматом по `loadingFlag `отработает или кусок эффекта может по новой проиграть?

vvip-68, 2021-12-08:
Вы мне предлагаете отлаживать ваши доработки скетча?
Экспериментируйте, отлаживайте, смотрите что получается и меняйте если что-то не так.

closed 2021-12-08 by vvip-68

limittouch, 2021-12-08:
достаточно было комментария программиста, который знает общий алгоритм и его подводные камни
возможно, уже проводились подобные попытки

limittouch, 2021-12-10:
проверил, работает
опишу еще раз, вдруг кому пригодится

делал на основе v1.10 (977da4eaad2c0f514f37ebaf164e3e4eae58c142), т.к. мне не нужно управление мультиматрицей

**_a_def_hard.h_**

в разделе настройки матрицы добавить
`#define REPEAT_PLAY 0         // 1 - повторное проигрывание текущего эффекта, если время не вышло; 0 - следующий эффект`

(по желанию) включить "ограничивать время проигрывания эффектов с карты, так же как и из программы"
`#define WAIT_PLAY_FINISHED 0  // 1 - переключаться на следующий эффект только когда весь файл полностью показан; 
`

**_custom.ino_**

добавить "сброс флага окончания игры" при переходе на следующий или предыдущий режим
```
static void nextMode() {
  gameOverFlag = false;

...

static void prevMode() {
  gameOverFlag = false;
```

заменить условие смены режима в `void checkIdleState() `

с
```
    if ((ms - autoplayTimer > autoplayTime) && !manualMode) {    // таймер смены режима
```
на
```
    if (((ms - autoplayTimer > autoplayTime) // таймер смены режима
         #if (REPEAT_PLAY == 0)
         // при окончании игры не начинать ее снова
         || gameOverFlag
         #if (USE_SD == 1)
         // если файл с SD-карты проигрался до конца - сменить эффект
         || (thisMode == MC_SDCARD && play_file_finished)
         #endif
         #endif
         )
    && !manualMode) {    
```