vvip-68/GyverPanelWiFi #179: вероятность эффекта с SD-карты

limittouch, 2021-12-07:
в `setRandomMode2()` пугает фраза
> // Когда эффект следующий выбирается случайным образом, вероятность что выпадет SD-карта достаточно мала.
> // Искусственным образом увеличиваем вероятность эффекта с SD-карты

почему не получается организовать обычный рандом (количество эффектов в памяти + на карте)?
чтобы они стали равновероятностными :)

vvip-68, 2021-12-08:
Потому что эффект "SD карта" это один из эффектов в диапазоне от 0 до MAX_EFFECT-1 в диапазоне которого случайным образом выбирается какой случайный эффект будет следующим. Все количество эффектов на карт - это "подрежимы" эффекта. Данный вариант реализации - вероятно один из способов. Вы вполне можете реализовать свой, на ваш взгляд более эффективный.  

Вряд ли предложенный вами способ даст более равномерное распределение показа эффектов. К примеру на карте 100 эффектов, макс кол-во эффектов вообще - 40.  То есть будем пытаться выбирать числа от 0 до 139. Числа от 0..39 - естественные эффекты. От 40 до 139 - эффекты с SD-карты.  Если выпадает число больше 29 - включаем эффект SD-карта. Исходя из простой логики распределения - нт никакой равномерности, потому что числа от 0..29 будут выпадать значительно реже - то есть обычные эффекты будут показываться гораздо реже и в основном будут "крутиться" ролики с карты.

Пробуйте, экспериментируйте, отлаживайте свой вариант.

closed 2021-12-08 by vvip-68

limittouch, 2021-12-08:
> Если выпадает число больше 39 - включаем эффект SD-карта.

так не просто включаем эффект, а передаем в него [140 - выпавшее число]

потому и спрашиваю, сложно ли в реализации, что для меня текущий алгоритм псевдослучайных чисел - темный лес
даже зачем `setRandomMode()` и `setRandomMode2()` не разобрался

vvip-68, 2021-12-08:
> даже зачем setRandomMode() и setRandomMode2() не разобрался

Не помню. Скорее всего когда setRandomMode() перестал удовлетворять тем, что эффекты с карты выпадали крайне редко - была написана улучшенная setRandomMode2(), а setRandomMode() осталась как рудимент.
Думаю ее можно безболезненно выкинуть и заменить тем где она использовалась (нешел 2 места) на эту улучшенную.

limittouch, 2021-12-10:
реализовал, работает
опишу, вдруг кому пригодится

делал на основе v1.10 (977da4eaad2c0f514f37ebaf164e3e4eae58c142), т.к. мне не нужно управление мультиматрицей

**_a_main.ino_**

заменить функцию `setRandomMode2()` на эту

```
void setRandomMode2() {
  
  byte newMode;
  uint16_t countEffects = MAX_EFFECT;
  #if (USE_SD == 1)
  if (isSdCardReady && getEffectUsage(MC_SDCARD)) // карта на месте и настройка "использовать эффект в демо-режиме" включена
    countEffects += countFiles - 1; // сама SD-карта является эффектом, поэтому "-1"
  #endif

  for (byte i = 0; i < 10; i++) { // 10 попыток найти случайный эффект, используемый в демо-режиме
    newMode = random16(countEffects); // граница НЕ входит в диапазон, эффекты начинаются с 0
    
    if (newMode >= MC_SDCARD || getEffectUsage(newMode)) { // эффект с карты или настройка "использовать эффект в демо-режиме" включена (дополнительная проверка для эффектов не с карты)

      if (newMode >= MC_SDCARD) {
        // effectScaleParam2[MC_SDCARD]: 0 - случайный файл; 1 - последовательный перебор; 2 - привести к индексу в массиве файлов SD-карты
        if (effectScaleParam2[MC_SDCARD] == 0) {
          sf_file_idx = newMode - MC_SDCARD;
        } else if (effectScaleParam2[MC_SDCARD] == 1) {
          sf_file_idx++;
          if (sf_file_idx >= countFiles) sf_file_idx = 0;
        } else {
          sf_file_idx = effectScaleParam2[MC_SDCARD] - 2;
        }
        newMode = MC_SDCARD; // SD-карта - один эффект, отличающийся номером файла (sf_file_idx)
      }
     
      setEffect(newMode);
      return;
    }
  }
  
  setEffect(0);
}
```