vvip-68/GyverPanelWiFi #180: NodeMCU v.3.4 ws2812b не стартует, либо загораются рандомно

bulnv, 2021-12-08:
### Проблема: 
- Гирлянда на NodeMCU v.3.4 прошивке GyverPanelWiFi и адреске на ws2812b не стартует, либо загораются рандомные диоды, либо ничего.  
  
### Железо:  

-   [ws2812b](https://www.amazon.de/gp/product/B09HX5JYCW/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&th=1)  (по заявлению продавца) на проводе залитие соплеклеем, 10m 100диодов
-   блок питания 5V 20A в дырявом корпусе, питание опущено до 4.6 ниже не идет
-   контроллер  [Node MCU v3.4 ESP8266](https://www.makershop.de/plattformen/esp8266/nodemcu-esp8266-dev-kit/)  (на черной плате).

  
### Что сделано:  

- [x]   Собрано по минимальной схеме пин д2 + резистор уходят на средний сигнальный провод адрески. провод 10см
- [x]   От БП запитаны и МК и адреска
- [x]  В ардуино IDE выбрана плата NodeMCU v 1.0 (ESP-12E Module)
- [x] Опрос платки в IDE возвращает номер и еще что-то
- [x]  Скомпилилось и залилось все удачно, вайфай поднимается, могу подключиться прилагой к МК и выставлять настройки (но на адреске глухо)
- [x]  Сначала скопмилил на версиии ядра платы ESP8266 > 3.X.X потом почитал и перекомпилил на 2.7.4. или около того, чтобы было совместимо с FastLED

  
Помогите задебажить в чем проблема, может продавцы обманули с адреской или МК не подходит, куда двигаться дальше?

AndreyLyamkin, 2021-12-09:
Попробуй подключить управляющий провод не на D2, а на D4

bulnv, 2021-12-09:
Пробовал, результат тот же

vvip-68, 2021-12-09:
@bulnv 
Адреска тем концом припаяна? Вход ленты с выходом не перепутан?
На версии ядра 2.7.4 и неизмененных настройках `#define LED_PIN 2` вывод идет на пин микроконтроллера, подписанный **D2**.

Что пишется в мониторе порта в момент старта прошивки и дальше при работе?
На стандартных примерах из библиотеки FastLED что-то светится?

vvip-68, 2021-12-09:
@AndreyLyamkin 
На каком контроллере собрано устройство? Вы практически единственный из тысячи повторивших, который жалуется на неработающий WiFi. Может у вас микроконтроллер дефектный? На другом пробовали?

Мне встречался только один микроконтроллер, на котором были постоянные разрывы связи - вот такой
[image]

У вас не такой?
Остальные - синие Wemos d1 mini, Pro и V3, а так же ESP32 и NodeMCU -все работали нормально

AndreyLyamkin, 2021-12-09:
[image]
[image]

[image]
[image]

AndreyLyamkin, 2021-12-09:
может поделиться моей версией чтобы проверить гипотезу? 
если на старом скетче все запустится, значит что-то не так с нашими платами, так как похоже у нас с @bulnv  они одинаковые. Но у него хоть по Wi-fi можно подключиться...

vvip-68, 2021-12-09:
Вроде нормальный Node MCU...

> Вот что я делаю не так?

Вопрос в другом тогда - что Я и другие делаем не так?
1. И у меня и у людей работает.
2. Код подключения к сети не менялся примерно с версии 1.02... Он каким был - таким и остается - никто ничего там не трогает. С чего бы он перестал работать.

Единственное, что можно предположить - вы указали контроллеру неверный адрес сети.
Например роутер работает в сети 192.168.**1**.xxx, а контроллеру присвоили адрес из сети 192.168.**0**.xxx

Про описанные переделки. Ну, используйте так, раз у вас работает.
Такой вариант подключения использовался с версии 1.00 до 1.02 примерно.
И тогда были массовые жалобы на отваливание WiFi и нестабильную работу сети. Нормализовалось всё только после того, как эти строчки были убраны из скетча.

По поводу правки в FastLED - тоже все с точностью до наоборот. При значении в 50 - стабильно работало. При уменьшении уже до 10 - начинался постоянный отвал WiFi...

vvip-68, 2021-12-09:
@AndreyLyamkin 

>может поделиться моей версией чтобы проверить гипотезу?
если на старом скетче все запустится, значит что-то не так с нашими платами, так как похоже у нас с @bulnv они одинаковые. Но у него хоть по Wi-fi можно подключиться...

Давайте свою версию. Есть у меня вроде такая плата. Попробую в нее залить ваш вариант.
Присылайте тот, который не правили вот с этими включениями перед FastLED и правое в самом фастлед
Вылодите куда-нибудь и дайте ссылку на скачивание

AndreyLyamkin, 2021-12-09:
у меня отвал Wi-Fi наблюдался пару раз за три дня. Я это решил почасовой перезагрузкой по питанию, благо умный дом везде уже зашел )

Настройки сети не мог напутать, я даже копировать стал, дабы не ошибиться. Да и по профилю профессии я эти айпишники в день по несколько десятков раз прописываю... 
Сейчас подготовлю архивы. Т.е. мы сейчас попытаемся воспроизвести хаотичные вспышки? Они, кстати, появляются при динамических эффектах. Если эффект медленно что-то рисует, то редкие артефакты.

AndreyLyamkin, 2021-12-09:
https://yadi.sk/d/oMlD5cIBU7deAA

simonovich, 2021-12-09:
После включения матрицы наблюдаю рандомные вспышки (флэши). По версиям все согласно рекомендациям.
@AndreyLyamkin, у вас были те же симптомы?
Вот как выглядит:
https://disk.yandex.ru/i/ViXoL6k26OgnLw

AndreyLyamkin, 2021-12-09:
@simonovich да. именно такие.

vvip-68, 2021-12-09:
@AndreyLyamkin 
Залил. Погонял. Никаких проблем ни с мерцанием/вспышками ни с WiFi

В присланном вами варианте скетча только поменял DeviceID с 0 на 2 - там настройки подходили под мои часы, собранные на NodeMCU таком же как у вас. Единственное - там нет MQTT в этом варианте.

Вывод - проблема со вспышками - скорее всего в вашем источнике питания - или он шумит или напряжение выше 4.8 вольта при отсутствии подпирающего диода на "минус" в схеме. Еще вариант - сигнальный пин от микроконтроллера до ленты слишком длинный и ловит наводки. Его нужно делать как можно короче и делать витой парой, скручивая его с минусовым проводом. Можно еще экранировать с одного конца.

Проблем с WiFi все время которое я тыкался - минут 15 также не выявлено. Связь держит стабильно.
На другой моей гирлянде, там есть MQTT - там идет постоянный обмен с брокером на каждую смену эффекта. Там связь периодически пропадает - но думаю дело не в WiFi/ "Периодически" - это примерно раз в две недели гирлянда перестает отвечать и ее приходится жестко по питанию перегружать. Но там на ESP32 собрано...

Вероятно источник проблем с WiFi может быть все тот же шумящий блок питания. Есть у вас осциллограф посмотреть что творится по питанию?

AndreyLyamkin, 2021-12-09:
@vvip-68 
Я уже писал, повторюсь:

- Источник питания непричем. Я проверял осцилографом качество. Вообще ничего, кроме прямой линии не выдает. Железный 5Вольт 30 Ампер.
- Напряжение пробовал опускать с шагом 0,1 В, с последующей перезагрузкой платы.
- Подпирающий диод ставил
- Длинна сигнального провода не превышает 10 см

Вопрос: Почему все это-же железо, но с другой версией прошивки работает? И вспышки получилось убрать внесением правок в код?

vvip-68, 2021-12-09:
У меня нет ответа. Можешь взять утилиту сравнения текстов и сравнить исходники в области того же соединения с сетью. Там нет изменений.

Убирание вспышек правкой кода библиотеки - известный факт. Работает только на дефектных лентах. У меня вот та гирлянда, про которую писал выше, когда она была на ESP8266 тоже были вспышки и "подергивания" текста, пока в фастлед константу не уменьшил с 50 до 10 примерно. На другой ленте - тот же самый контроллер, та же прошивка - просто подключил другую ленту - работал без вспышек. При уменьшении значения до 10 на дефектной ленте вспышки пропали, но стал отваливаться WiFi - и минуты не держался стабильно. При заливке все того же скетча на ESP32 и возврата константы в Fastled в изначальное состояние - вспышки пропали.

AndreyLyamkin, 2021-12-09:
@vvip-68 да я разве сказал, что не верю? Просто не могу понять, в чем тут дело. Прошивал раз 50 с разными вариациями. Попробую разобрать твой Device ID 2

Мне хочется обновиться тоже, может потом и рисовать эффекты в Jinx )))

Вот у меня есть ещё такие экземпляры. 
[image]
[image]

Но я так и не смог подобрать настройки чтобы получилось залить прошивку. Постоянно куча ошибок =(

vvip-68, 2021-12-09:
DeviceID = 2 - просто назначение пинов, наличие модулей, угол подключения матрицы и ничего более.
Вот - попробуй на ESP32 все собрать. Если не получается что - стучись лучше [сюда](https://vk.com/iivanovii) в личку/ Оперативнее будет.

simonovich, 2021-12-09:
У меня выход БП - 5V
Вход на МК - 4.8V
Входящее напряжение стабильно до сотых, т.е. колебания не более 0.01V
На выходе с МК на 3V наблюдаю скачки напряжения и параллельно вспышки на матрице.
Возможно на выходе 5V тоже нестабильно, позже проверю.
Если что, на 3V у меня запитан только сигнальный D2 и кнопка.
Выше писали про "подпирающий" диод. Можно подробнее, что за он и куда ставить?

vvip-68, 2021-12-09:
>Выше писали про "подпирающий" диод. Можно подробнее, что за он и куда ставить?

[image]

simonovich, 2021-12-09:
"кремнивый диод" - понятно
"+06в относительно массы" - не понятно
может другими словами?..

vvip-68, 2021-12-09:
Вы действительно хотите разобраться в физике процесса или будет достаточно попробовать поставить этот диод в надежде, что он поможет?

simonovich, 2021-12-09:
Разобраться, в надежде, что поставлю и поможет! ))
Ну, в гугле не забанили, написано бла-бла-бла кремниевый диод пропустит ток при разности напряжений в 0.7V
Речь про эту разность? Тогда почему шесть...
Читаем дальше. "Если указанная разность не достигнута, то проводник обесточен".
Хм...

simonovich, 2021-12-09:
Обычный выпрямительный что-ли?
Есть такие:
https://amperka.ru/product/rectifier-diode

AndreyLyamkin, 2021-12-09:
@simonovich пойдет. ставь, как указано на схеме....

simonovich, 2021-12-09:
Проблема со вспышками решена. И я ничего не паял.
Виноват сигнальный провод к матрице. Даже не так. Провод нормальный, короткий, хорошо припаян. Но припаян он к коннектору в который втыкается кабель матрицы. И этого кабеля я воткнул где-то метра на 2... Вот картинка и сыпалась! Доп. питание, которое я подаю на противоположный край панели, убирает артефакты, но остаются те самые флэши-вспышки.
Я заменил кабель на более короткий и картинка стабилизировалась. Теперь норм даже без доп. питания.

Собственно вопрос по питанию. Панель сборная - из 3-х матриц 16х16. И доп питание я подаю только на самую последнюю. Расчет на то, что матрице по середине и так перепадет с двух сторон, не просядет.
Матрицы самые стандартные, гибкие. У них сзади распаяно на три провода: вход, подпитка, выход на след. матрицу.
Соответственно соединяю вход-выход соседних + последнюю сажу на доп питание по "среднему" проводу.
Всё верно?

AndreyLyamkin, 2021-12-10:
@simonovich не совсем понятно... )))
т.е. сигнальный провод был более 2 метров длинной? Об этом же сразу говорили - укоротить до минимума )))
когда не хватает питания, самые дальние светодиоды горят не так ярко, как первые. Если подать питание с двух сторон, то напряжение не суммируется, оно просто с двух сторон. Если тут три матрицы - самое оптимальное - подавать на каждую матрицу питание сразу от БП, а не от предыдущей матрицы.

vvip-68, 2021-12-10:
Не зря же даже на ОДНУ стандартную матрицу питание предусмотрено в трех точках. Я бы на каждую матрицу подавал питание. И желательно в трех точках...

closed 2021-12-10 by vvip-68

simonovich, 2021-12-10:
@AndreyLyamkin, это выглядело не так. Артефакты сыпались на всех трёх матрицах случайным образом, и это не затухание.
Если по докам, то питать матрицу в OUT нельзя. Ну или это тонкости перевода, и просто имеется ввиду, чтобы сигнальный провод с двух сторон не тыкали, а для подпитки использовали специально выброшенные 2 провода посередине каждой матрицы. 
В варианте "гирлянда" (светодиодная лента) я таким же образом запитывался с двух сторон без проблем.
За совет спасибо, при случае подсажу на питание и две первые матрицы.

simonovich, 2021-12-10:
Теперь картинка рассыпается в такт музыке. Если выключить проигрывание - всё ок.
Полагаю, что провода к матрице и плееру надо развести подальше друг от друга. Желательно экранировать.
Кабель еще короче. И всё будет зашибись.
А глюк даже прикольный...
https://disk.yandex.ru/i/vIs-1MIY9J1YXg

vvip-68, 2021-12-10:
Вопрос: А нафига вообще контроллер так далеко от ленты относить, что 2 метра кабеля понадобилось? Почему его нельзя на заднюю стенку этой матрицы прикрепить?

simonovich, 2021-12-10:
Коробку видишь? Там и контроллер и БП-монстр, и прочий фарш.
Размеры коробчонки 75х200. Короче, так исторически сложилось и я уже привык.
В этом свои плюсы.
А с кабелем еще проще. У меня на подпитке длинный, а на сигнал - короткий. Но когда распаковал всё после лета, то всё перепутал.

vvip-68, 2021-12-10:
Да, серьезное изделие...
Но если изначально планировался достаточно длинный кабель - наверное он должен как-то специально рассчитываться. Там 800кГц данные идут на WS2812. Возможно какой-нибудь коаксиальный/антенный/высокочастотный нужен. Только он же как-то тоже рассчитывается. Волновое сопротивление и все такое там...
Даже длинные USB кабеля - и то - экранируют, или как на кабели монитора - ферритовые кольца ставят...

simonovich, 2021-12-10:
Я новичок в электронике, умных вещей не знаю. Год назад эта коробка стояла на балконе и умела только моргать огоньками. А сейчас в ней половина новых деталей и полный фарш... Год назад у меня через день замыкало светодиоды от конденсата, а макетные дюпоны отваливать от любой вибрации. Я приходил сюда, и жаловался... А теперь коробкой можно в футбол играть, ничего там не отвалится. 
Допускаю, что кабель тоже надо менять. Если хватит понимания, то с крутым расчетом. Если не хватит - то как-то иначе.
Спасибо всем, кто помогает мне разобраться!

simonovich, 2021-12-11:
Картинка больше не сыплется, даже если играет dfplayer.
Просто добавил конденсатор по питанию от блока питания на микроконтроллер.
Точно так, как на схеме, и как советовали.
Когда схема с минимальным обвесом, это не критично. Например, не добавь я плеер - то и не узнал бы о такой проблеме. Ну, или узнал гораздо позже. Ведь все работало!
Ну а если лень паять, то случаются всякие такие эксперименты и приключения как у меня.
Вот он красавчик конденсатор:
https://disk.yandex.ru/i/Lnua8ZopT92U2g