vvip-68/GyverPanelWiFi #181: Редкие фризы на 5 секунд перед сменой эффектов

boomega, 2021-12-09:
Здравствуйте!
В прошлом году была гирлянда 25х12, версия прошивки 1.08, всё работало без нареканий. В этом году расширил до 56х12 и залил прошивку 1.11, периодически на каких-то эффектах чаще, на каких-то реже (мне так кажется), стала останавливаться анимация секунд на 5 перед переключением на следующий режим (установлено автоматическое переключение на следующий эффект через 1 минуту), например, эффект зебра переливается минуту, и в конце стоп-кадр, всё останавливается, через 5 секунд включается следующий эффект как ни в чём не бывало. Очень очень редко бывает стоп-кадр посреди эффекта, так же замерзает на 5 секунд, а потом дальше работает.
Попробовал снова прошиться на 1.08 с подстановкой библиотек "из комплекта", вроде сначала фризов не было, но потом та же история. Плата всё это время используется одна и та же (NodeMCU).
Такое ощущение что либо память переполняется, либо процессор перегревается, что-то такое, не очень понимаю что под "капотом" происходит, поэтому прошу помощи в решении проблемы.
Спасибо.

vvip-68, 2021-12-09:
Чтобы понять что "плд капотом" происходит - полезно посмотреть что выводится при этом в монитор порта.
Внешне эти симптомы похожи на ожидание ответа от сервера в интернете.  Когда запрос отправляется, а сервер не отвечает или нет соединения с интернетом вообще. В интернет лезут часы для синхронизации и погода за актуальной температурой. Если один из этих серверов не отвечает - будут фризы.

Смотрите что в мониторе порта. Или отключите получение погоды. Затем - синхронизацию времени с интернета. Если после отключения фризы прекратятся - это оно.

simonovich, 2021-12-09:
У меня похожие штуки были при подключенном MQTT.
Когда уходит запрос в сеть (неважно MQTT или погода там), то некоторое время ожидаем ответ по таймауту.
Пробуйте отрубать методом исключения.

boomega, 2021-12-09:
Да, заметил ещё что при включении устройства, если выбрать эффект часов, они не выводятся, нужно перейти на вкладку с настройками времени, запросить его и всё начинает работать.
Спасибо за подсказки, погода отключена, а вот с синхронизацией времени очень похоже.
Поэкспериментирую ещё и по результатам отпишусь.

boomega, 2021-12-09:
Версия с подвисающим запросом времени подтвердилась. Действительно, само устройство, прошивка и настройки никак не влияют на появление фризов. Попробовал поднять локальный NTP-сервер (у меня это можно сделать в пару кликов) и проблемы исчезла, снова вернул настройки с NTP-сервером в интернете - опять глюки.
Так что в моём случае самый простой вариант решения проблемы это локальный NTP-сервер, не хочу искать проблему  дальше, т.к. тут может быть и AdGuardHome на докере, и передоз торрентами, и роскомнадзор, и может что ещё.
Спасибо за помощь!

closed 2021-12-09 by vvip-68