vvip-68/GyverPanelWiFi #182: Не стартует клиент mqtt

VahrushevSA, 2021-12-10:
Здравствуйте!
в настройках ESP8266 выставил USE_MQTT 1
в MQTT parameters сервер указал
при работе платы не стартует mqtt клиент

AndreyLyamkin, 2021-12-10:
что пишет в мониторе порта?

vvip-68, 2021-12-10:
f rfr dsа оняzfkчто он не стартует?

VahrushevSA, 2021-12-10:
в мониторе порта по идее должно отображаться подключение к серверу, его нет

vvip-68, 2021-12-10:
Давайте лог с монитора порта сразу после запуска контроллера.
А в настройках подключений (например в приложении) вообще галочка стоит, "Использовать MQTT"? 
Может у вас он вообще отключен?

VahrushevSA, 2021-12-10:
WiFiPanel v.1.12.2021.1206

Инициализация файловой системы... 
FS: Использовано 24576 из 2072576 байт
Загрузка строк выполнена.
Использованы адреса EEPROM 1000 - 1037
Свободно ячеек 3058

Host: 'WiFiPanel' >> 30x17

Найдены сохраненные настройки: FS://eeprom.bin

Подключение к m-k -> 192.168.66.45
.............
WiFi подключен. IP адрес: 192.168.66.45

UDP-сервер на порту 2390
Режим: Морской прибой
NTP-сервер ru.pool.ntp.org -> 193.192.36.3
Отправка NTP пакета на сервер ru.pool.ntp.org

Запрос текущей погоды
Погода получена: Yandex
Город: Kamensk-Uralskiy
Сейчас: пасмурно, -7ºC
Код иконки: 'ovc'
Цвет неба: '#9bccff'
Сейчас: 
Рассвет: 09:14
Закат: 16:15

UDP << ip='193.192.36.3:1

vvip-68, 2021-12-10:
В приложении PanelSetup в настройке соединения галка "Использовать MQTT" включена? Настройки соединения настроены? Какой брокер используется? Скриншот экрана приложения пожалуйста.

VahrushevSA, 2021-12-10:
DEFAULT_MQTT_SERVER "192.168.66.252"
в логе ни слова о сервере

vvip-68, 2021-12-10:
Значит в настройках отключено использование

VahrushevSA, 2021-12-10:
Извините! не глянул PanelSetupWiFi
всё заработало.
Спасибо, что оперативно отвечаете!

closed 2021-12-10 by vvip-68