vvip-68/GyverPanelWiFi #183: Эффекты sd

kost111233, 2021-12-10:
Добрый вечер. У меня установлена смена эффектов через 60 сек., но эффекты сделанные в jinx и записанные на флешку проигрываются примерно 3 минут, подскажите как исправить. 
И могу ошибаться, но когда ставил в приложении в эффекте SD последовательная смена, вроде один и тот же эффект был, я это понимал что если на sd попадает, по очереди записанные эффекты будут без повторения

vvip-68, 2021-12-11:
Думаю, что за это отвечает настройка 
```
#define WAIT_PLAY_FINISHED 1    // 1 - переключаться на следующий эффект только когда весь файл полностью показан; 
```

Она была введена, когда пользователи жаловались, что ролик еще не проигран до конца, а уже переключается на следующий эффект. Зачастую люди просят сделать прямо противоположные вещи...

vvip-68, 2021-12-11:
> И могу ошибаться, но когда ставил в приложении в эффекте SD последовательная смена, вроде один и тот же эффект был, я это понимал что если на sd попадает, по очереди записанные эффекты будут без повторения

Вот, я могу ошибаться, но кажется это означает, что с каждым попаданием в эффект "SD-карта" будет браться следующий ролик, а не случайный или конкретно указанный.

Чтобы эффекты (в т.ч. и ролики с карты) воспроизводились последовательно, нужно в настройках снять галочку "Случайный порядок". Тогда после того, как очередь дойдет до эффекта SD-карта ролики будут прокручиваться последовательно до конца списка...

closed 2021-12-11 by vvip-68

kost111233, 2021-12-11:
Вот я это и имел в виду) думал что при следующем попадании на sd будет следующий воспроизводится записанный на флешку. 

Я вас понял, так как многие записывают в джинкс по 5-10 секунд и он видимо столько же и проигрывался на матрице. Но просто не понятно откуда взялась эта цифра в 3 минуты, почему именно столько:) я точно такие длинные ролики не делал, а дед мороза вообще записывал секунд 5

kost111233, 2021-12-11:
https://youtube.com/shorts/LFIpUIpp9Zo?feature=share

vvip-68, 2021-12-11:
Я тоже не знаю откуда взялась цифра 3 минуты. Ролики с карты играются либо до конца ролика (если длительность больше установленного времени воспроизведения эффекта), либо по установленному времени смены эффекта. Если эффект "кольцевой" - нет смысла записывать фреймы на 3 минуты. Достаточно фрагмент в 10 секунд, который потом будет закольцован... А показывать 10-секундный фрагмент и переходить на следующий эффект не вижу смысла...