vvip-68/GyverPanelWiFi #184: DFPlayer не обнаруживается

Mosrom, 2021-12-10:
У меня установлен оригинальный модуль DFPlayer. Долго мучился, использовал разные варианты задержки. Не помогает.
В мониторе порта сообщение об ошибке "DFplayerError:  Таймаут! MP3 плеер недоступен". С предыдущей библиотекой в принципе работал нормально. Какой можно ещё вариант использовать?
Кроме этого при компиляции возникает ошибка со ссылкой на строку 650 файла a_def_soft.h
Добавил переменную REPEAT_PLAY в файл a_def_hard.h следующего вида "#define REPEAT_PLAY 1" в строку 253, так как у меня модуль ESP32. Ошибка исчезла. Однако точно не уверен, что это верно.

vvip-68, 2021-12-11:
Ну, наверное правильно, если ошибка исчезла...
Хотя, в том варианте файла a_def_hard.h, что есть у меня - это определение  для ESP32 расположено в строке 284 и компиляция проходит без проблем... А в файле - вообще  всего 367 строк...

Про плеер - вообще непонятки. Думаю вообще выбросить его из проекта. От него одни проблемы.
Потому что у всех по разному и никакого единообразия нет. Вот у кого-то только со старой библиотекой работает, у кого-то - наоборот только с новой. Кому-то вообще не удается его заставить работать ни с какой из библиотек...
В общем - одни проблемы и никакой ясной картины как с этим пороться. У меня плеер работает нормально с любой из этих библиотек.

Если откатиться обратно на версию, где используются прежняя библиотека - будет работать?
А то, может у вас физически контакт где-нибудь отвалился?

Mosrom, 2021-12-11:
Вообще плеер с самого начала вел себя загадочным образом. Но я понял, почему он не определялся раннее. Это было связано с напряжением блока питания. Если напряжение более 5 вольт, то он не отображался. Плеер стабильно работал с напряжением 4.85 v. И было это более года назад. А сейчас я просто решил обновить прошивку. У меня уже стояла версия 1.12 и со старой библиотекой работал нормально. Провода и контакты в норме, закреплен он надежно. И не клон, специально посмотрел .
Я бы не выкидывал его из прошивки, функция интересная и полезная. И потом не наш это метод ).

vvip-68, 2021-12-11:
Видите ли... Ведет он себя непредсказуемо и бессистемно.
У кого-то работает, у кого-то нет. У кого-то "частично нет", у кого-то - "совсем нет".
Я не могу выявить закономерность. Но мне известно порядка 5-6 человек на данный момент, которые имеют проблемы с плеером и я им ничего не могу предложить. 

Причем зачастую - "в прошлом году работало, в этом году включил - не работает".
Или - "на прошлой версии прошивки работало, а в этой нет", а вся разница в прошивках на которой "работало" и на которой "нет" - изменено начертание символа '$' в шрифте.

Так что от плеера - одни проблемы.

То же самое с WiFi - "в прошлой версии работало стабильно, в новой - постоянные разрывы связи". А работа с WiFi не менялась, начиная с версии 1.02 полуторагодичной давности. Вообще не менялась. Там все как было. Даже ядро - одно и то же 2.7.4.  И что вот со всем этим делать?...

По плееру - попробуйте поиграться со значением константы задержки GUARD_DELAY.
У меня по умолчанию стоит 75 (миллисекунд), У кого-то (на клонах) меньше 100 не работает. Больше 200 - тоже...

А может быть в прошлой версии у вас были переназначены пины RX/TX плеера, а обновив прошивку - не изменили переназначение? Или RX с TX поменяны местами...

Mosrom, 2021-12-11:
Специально, ради эксперимента установил предыдущую версию прошивки (1.12.2021.1025), которая стояла до обновления с библиотекой "DFrobotDFplayerMini" и плеер работает. То есть определяется прошивкой и звуки опознаются Возвращаюсь к текущей прошивке (1.12.2021.1210) не работает. RX/TX и прочие ошибки с пинами исключены.
Со значениями задержки, как писал выше, уже игрался. В диапазоне от нуля до 200 использовал разные варианты. Вообще никаких изменений. 
С wifi  ни разу не было проблем. Модуль esp32 оригинальный.
Думаю многие используют клоны и собирают с отклонениями от приведенной в примере схемы. 
И почему-то плеер чувствителен к напряжению. Может быть как-то резисторы, через которые он подключен к плате влияют.

vvip-68, 2021-12-11:
Ну вот у меня плеер - оригинальный. В нескольких часах по этому проекту собран. И с резисторами и без. И с прежней библиотекой играл и с этой новой тоже играет. Я не знаю по какой причине возникают проблемы и не знаю что по этому поводу посоветовать...

simonovich, 2021-12-11:
Да, для dfplayer напряжение очень важный момент. По докам он может работать и на 3V и на 5V, но реально он завелся только на 5V (читай 4.8V). Не выше, т.к. на "чистых" пяти он тоже может глючить. Нужно посмотреть по даташиту. Вроде там рабочий вольтаж обозначен как 4.2V
Некоторые мои знакомые даже спецом выкручивали напряжение отдельно на этот модуль. Говорят, после этого всё зашибись.
Потом начинается борьба с помехами по питанию. Плеер серьезно "раскачивает" систему, в итоге и сам трещит, и часы / матрица тоже могут глючить.
Полечилось установкой фильтрующей пары конденсаторов (электролит и керамика) на выходы 5V и GND плеера.
После этого плеер и определяется, и работает. Есть вопрос только по максимальной громкости. Её нельзя подымать выше некоего порога (определяем экспериментально), иначе часть глюков вернется - вплоть до перезагрузки микроконтроллера.
И даже с этим можно еще пободаться, если конечно оно вам надо. Просто я использую внутренний усилок плеера, а он там на 3W. Можно добавить в схему внешний модуль усилителя, и тогда пределы по громкости сильно раздвинутся.
Еще есть вариант на 2 динамика стерео развести, а не на один. Но там скорее всего еще круче проблемы по питанию встанут, т.к. нагрузка возрастет.
Правильно подобрать динамик - отдельная задачка. Технически я в этом не шарю, просто погуглил, с какими динамиками народ собирал схемы в паре с этим плеером. Помогло.
Мой dfplayer похож на оригинальный: цифры на микросхеме как у оригинала в прошивке, но отличается несколько последних позиций. Прошивал с настройками на "оригинал".
Единственное, что может раздражать в плеере - это то, что он включается вместе с МК (слышим щелчок) и работает постоянно, пока получает питание. При этом динамик издает шум. Если собрали нормально, то это тихий пшшшш. Это абсолютно нормально, особенно для такого дешевого плеера.
Имхо, можно подпаять на плеер отдельную кнопку Вкл/Выкл. Это позволит обесточить его в ситуации, когда звук вообще не нужен (работа чисто как гирлянда). И включить его, когда звук нужен (вариант часы).

vvip-68, 2021-12-11:
> Есть вопрос только по максимальной громкости. Её нельзя подымать выше некоего порога (определяем экспериментально), иначе часть глюков вернется - вплоть до перезагрузки микроконтроллера.

Это вполне естественно. При 3W мощности и 43-5 вольт питания - ток составляет от 700 мА до 1А. Если БП не обеспечивает такого тока (например для работоспособности плеер запитан от выходного пина 3.3V контроллера) - падение напряжения просто перезагрузит и плеер и микроконтроллер.

> Единственное, что может раздражать в плеере - это то, что он включается вместе с МК (слышим щелчок) и работает постоянно, пока получает питание. При этом динамик издает шум. Если собрали нормально, то это тихий пшшшш. Это абсолютно нормально, особенно для такого дешевого плеера.

Ну, вроде как в библиотеках существуют команды enableDac() и disableDac(), которые просто включают/отключают АЦП (возможно не некоторых клонах не работают). Тогда можно в прошивку на stop() поставить рядом disableDac(), a в play() - enableDac(). Возможно это уберет шипение. Нужно экспериментировать...

Mosrom, 2021-12-11:
Simonovich, спасибо за комментарий. У меня при проигрывании плеер щелчков не издавал. В режиме простоя шипения не слышал. Про питание так глубоко не копал, понял это  экспериментально. На счёт фильтрующей пары конденсаторов, какие их характеристики. Можете написать?  Буду пытаться оживить плеер.

simonovich, 2021-12-11:
В моем случае это 470nF 10V и керамический 104 (~100 нф).
Керамика - поодойдет любая мелочь, например на 20-33 нф
https://disk.yandex.ru/i/SHDzuXA_313sTQ

closed 2021-12-12 by vvip-68