vvip-68/GyverPanelWiFi #186: Ошибка компиляции версии 1.12

samoylovrv, 2021-12-12:
Надеюсь, автор сможет помочь в решении проблемы: увидел обновление в версии 1.12 с внешним управлением по прот околу Е131. Ранее версия 1.11 отлично компилировалась и стабильно работает на всех имеющихся платах Wemos D1. Сегодня пробую компилировать версию 1.12, все библиотеки установлены, ядро и фаст-лед рекомендованные установлены. При компиляции вылетает ошибка:

'class ESPAsyncE131' has no member named 'createPacket'

Маркером выделяется следующая строка:
e131_packet_t* makeCommandPacket(uint8_t cmd) {
  e131_packet_t *packet = e131->createPacket(host_name.c_str(), e131_cid);

Куда копать??? Очень нужно прикрутить матрицы по Е131 к Jinx/Xlights!!!

vvip-68, 2021-12-12:
Скорее всего, вы не читаете инструкции и библиотеку E131 установили с менеджера библиотек, а не скопировали с проекта. В проекте (в папке Libraries) - библиотека существенно переделанная.

samoylovrv, 2021-12-12:
> Скорее всего, вы не читаете инструкции и библиотеку E131 установили с менеджера библиотек, а не скопировали с проекта. В проекте (в папке Libraries) - библиотека существенно переделанная.

Ок. Попробую почистить библиотеки и подключить из проекта. Спасибо за оперативный ответ!

closed 2021-12-12 by vvip-68