vvip-68/GyverPanelWiFi #188: Светодиоды

playeeerua, 2021-12-12:
Скучали по мне?)
У меня вопрос по питалову. Блок питания 5В 3А. Светодиодов как у гайвера 42 на 8. Отпаяли одну ленту (верхнюю и вуаля, заработало) Возможно ли что ему питания не хватает? Хотя у гайвера ж на видео такой же блок…

[image]

vvip-68, 2021-12-12:
Думаю, что причина не в этом. Скорее в верхней ленте где-то замыкание.

playeeerua, 2021-12-12:
Тоже об этом думал, щас будем верхнюю ленту менять

closed 2021-12-12 by vvip-68

simonovich, 2021-12-12:
На чем оно там. Нанка? Только что зависал на канале у Гайвера, где он рассказывал, что для нанки верхний предел 600 светодиодов. Т.е. все должно работать.

playeeerua, 2021-12-12:
Работает, поменяли верхнюю и нижнюю ленту (внизу половина работала). Но вот еще проблема (как же без нее) Если с резис
[image]
тором, то сверху 2 линии белые (фото 1), подключить напрямую, артефакты убираются
[image]

vvip-68, 2021-12-12:
Может сопротивление резистора великовато?
Совсем без резистора чревато выгоранием пина контроллера. Как повезет.

247760, 2021-12-12:
@playeeerua  Конденсаторы по питанию есть?

playeeerua, 2021-12-12:
Конденсатора нет. Резистор 220 Ом

247760, 2021-12-12:
> Конденсатора нет.
Почему?

playeeerua, 2021-12-12:
Ну написано необязательно 😂
Нам для начало надо было сделать что-б работало, конденсатор есть, допаяем это они как я понимаю?
[image]

playeeerua, 2021-12-12:
Вот резистор
[image]

playeeerua, 2021-12-12:
Вот резисторы
[image]

247760, 2021-12-12:
Запаивайте, он есть на всех схемах, ещё добавляют между ног электролита керамический конденсатор, но Вы хоть этот пока добавьте.

playeeerua, 2021-12-12:
Все, с резистором вижу, продали кОм

playeeerua, 2021-12-12:
Он как бы есть, но пишет необязательно🤔
Но в лампу добавил, вот и остатки есть. Главное что наконец-то спустя 3 месяца мы дошли до финишной прямой, осталось совсем чуть-чуть. Спасибо за терпение, без вас бы я не справился) Но есть вопрос по поводу приложения. А как часы убрать?) Все перетыкал. А то он везде, в бегущей строке, в режимах, а так отдельно включить как режим никак?

vvip-68, 2021-12-12:
С часами не надо все тыкать. Достаточно в настройках часов в приложении снять галку "Часы поверх эффектов"

vvip-68, 2021-12-12:
Отдельно часы включить как режим - есть "режим (эффект) который называется "Часы"

playeeerua, 2021-12-12:
Нашел, спасибо