vvip-68/GyverPanelWiFi #189: Добавление режимов

KirbyZiga, 2021-12-12:
Здравствуйте, душевное спасибо Вам за всё, что вы делаете!

Дано: 
Использую прошивку 1.15 от vvip-68;
Хочу добавить режимы из предыдущих версий прошивок;

Вопрос:
Как правильно добавить режимы из версий 1.11 и 1.12 в версию прошивки 1.15?

Спасибо.

KirbyZiga, 2021-12-12:
Дано: Wemos d1 mini

vvip-68, 2021-12-12:
Мне ничего не известно про прошивку 1.15
Последняя версия прошивки этого проекта - 1.12

Для добавления режимов из любой прошивки в любую нужно освоить основы программирования (если их нет, разобраться как устроена прошивка, что в ней за что отвечает, в какой последовательности выполняется, что от чего зависит. Затем взять кусок кода режима исходной прошивки, вставить его в желаемую, прописать все связи.

closed 2021-12-12 by vvip-68

KirbyZiga, 2021-12-12:
Необходимый код найти не сложно, вопрос в зависимостях.
Может вы разберете 1-2 примера для реализации миграции режимов с проекта 1.12 на 1.15?
Желательно с применением эффектов...
Это бы очень помогло, заранее благодарен.

vvip-68, 2021-12-12:
Нет, тут всё или самостоятельно, или пользоваться тем, что есть.
К тому же, как я уже сказал, мне ничего неизвестно про версию 1.15

frol-aleksan, 2021-12-13:
> Здравствуйте, душевное спасибо Вам за всё, что вы делаете!
> 
> Дано: Использую прошивку 1.15 от vvip-68; Хочу добавить режимы из предыдущих версий прошивок;
> 
> Вопрос: Как правильно добавить режимы из версий 1.11 и 1.12 в версию прошивки 1.15?
> 
> Спасибо.

Добрый день. Что именно интересует? Если я правильно понял, Вы хотите перетащить некоторые эффекты из старых проектов в новый? Можете воспользоваться моей редакцией прошивки WiFiPanel. В ней добавлены Runner, Flappy Bird (впрочем, надо еще разбираться со скоростями и таймингами, т. к. эти игры в данный момент наиболее играбельны только на максимальной скорости) и синусоиды (переделан таймер на Millis, т. к. с штатным таймером наблюдались дикие лаги, также можно в настройках выбирать количество синусоид от 1 до 4, но наиболее хорошо смотрятся 2 или 3) из ранней версии проекта GyverMatrixBT.

KirbyZiga, 2021-12-13:
> Добрый день. Что именно интересует? Если я правильно понял, Вы хотите перетащить некоторые эффекты из старых проектов в новый? Можете воспользоваться моей редакцией прошивки WiFiPanel. В ней [...]

Огромное спасибо, скачал, установил, пока все идеально!!!
Проверяю, тестирую!!!
Низкий поклон Вам!!!