vvip-68/GyverPanelWiFi #190: не воспроизводится презентация Jinx! (*.out)

BoromirKIEV, 2021-12-13:
Уважаемый Автор, такой вопрос: пересобрал заново проект, программно и аппаратно всё заработало (не без проблем, из-за обновлений и несовместимостей версий, но чтение вики помогло). КРОМЕ: не могу запустить презентацию из Jinx. 
Что делал: настроил с чистого листа джинкс (размер матрицы, настройки вывода и т.д, по инструкции. Записал ролик на секунд до 10 длительностью. В JinxFramer созданный файл воспроизводится.  Карту форматирую и в фат32 (пробовал и в фат16) - при старте вижу в мониторе порта нормальный запуск всех служб, в т.ч. ntp и погоды, чтение файлов на карте и далее- цикличное воспроизведение эффекта ( монитор смотрю только подсоединив вемос к ноуту, без подключения матрицы). Когда пытаюсь воспроизвести на матрице - в списке эффектов выбираю Sd card , выбираю свой эффект (1 файл на карте) - и ничего не происходит, проигрывается программный эффект из стандартного списка.
PS Уже настроился попробовать переделать под ЕСП32 всё железо, вдруг там причина, но для начала обращусь сюда (да, пытался спросить в ютубе вначале, т.к не разобрался с наличием функционала Issues тут, но ютуб не отобразил (удалил?) мой вопросю
PPS параметры моей "системы" : 1) Wemos D1 + шилд микросд карты + карта 1 или 2Гб; 2) ESP8266 2.7.4 ; 3) FastLED 3.4.0 ;  4) матрица 46*25 на WS2811.

simonovich, 2021-12-13:
У меня карта в FAT32  вообще не определялась, рекомендую FAT16.
Проверить, что правильно настроен вывод в Jinx (параметры матрицы - тип, направление, кол-во точек и т.д.).

BoromirKIEV, 2021-12-13:
Отвечаю сам себе: после вчераних ночных мытарств с разными картами и мониторингом порта, сейчас подключил к матрице контроллер ...и увидел вопроизведение своей тестой презентации. Однако. Так и не понял, что это было и почему не работало) Пробую дальше.

BoromirKIEV, 2021-12-13:
> У меня карта в FAT32 вообще не определялась, рекомендую FAT16. Проверить, что правильно настроен вывод в Jinx (параметры матрицы - тип, направление, кол-во точек и т.д.). 
- тут подстраховка - это проверка в джинксфреймер, как я понял - если там работает, то и на матрице должно. Один из эффектов, например, в фреймере во время проигрывания периодически (на отдельных кадрах!) - пишет чтото про ошибку в размере вывода или типа того. Почему так его экспортнул джинкс, при настроенной матрице - загадка. Поэтому беру за правило, перед заливкой на карту - проверять в фреймере.
- Название вашего эффекта появляется на экране после выбора в меню SD, или список названий в телефон не попадает?
- и в мониторинге порта, и в списке в приложении - всё было. Но! вспомнил, что тогда на карте был среди прочих и тот "битый" файл презентации с фантомной ошибкой размера. Сейчас попробую перезаписать его, и после проверки в фреймере, залью на карту и в контроллер...

PS как-то много "горящих" пикселей появилось, вместой зеленой простыни - красные точки подергивающиеся.. Это артефакты? лечить корректировкой питания или настройками в скетче? кажись ночью видел в одном из Issues  про уменьшение параметра с 50 на 5.. вспомнить бы где видел :)

simonovich, 2021-12-13:
Скорее всего, вы видели про это [здесь](https://github.com/vvip-68/GyverPanelWiFi/issues/180)

BoromirKIEV, 2021-12-13:
> Скорее всего, вы видели про это [здесь](https://github.com/vvip-68/GyverPanelWiFi/issues/180)

не там (похоже в постах AndreyLyamkin  встречал, помню что автор сисадмин))), но тоже полезно было прочесть. Завтра буду лечить свои артефакты (вспышки тоже есть). Диод и кондер пока что не ставил, но не вопрос, поставлю. Питание достаточно грамотно организовал: толстым проводом от компового 300вт БП подходит 5в к матрице и контроллеру, сигнальный провод от контроллера до первого светодиода - не более 15 см. Сама матрица (из 23 лент по 50шт диодов) - запитана в 12 точках по + 23 точках по минусу (матрица собрана на легкой металл.сетке, эта же сетка служит общим проводом для минуса питания всей матрицы. При такой разводке - нет падения напряжения даже при максимальной яркости. Ранее, в другой реализации, середина матрицы проседала заметно.

kost111233, 2021-12-14:
Мне приходится перевключать микроконтроллер после того как вставил флешку, иначе эффекты sd либо не отображаются, либо показывает только старые, даже если их удалил

simonovich, 2021-12-14:
Я не смотрел по коду, но скорее всего МК не отслеживает момент замены SD-карты.

vvip-68, 2021-12-14:
Да, в прошлой версии библиотеки было, в этой - недовставил.
Нужно в dfPlayer.ino в строку 254 вставить следующее:
    InitializeDfPlayer2();
    if (!isDfPlayerOk) DEBUGLN(F("MP3 плеер недоступен."));

alexder21, 2021-12-14:
Тоже долго мучился с отображением роликов Jinx на ESP32.
В моем случае наличие sd карта определялась, но не находилась папка с размером матрицы.
Форматировал по всякому - результата ноль. Попробовал пример SD_test для ESP32 - работает, файлы создает, пишет и читает, но тоже не отображает содержимое карты (записанное на компьютере), только то что создал сам тест - то и видит. Причем на компьютере тестовых файлов не видно. Странная взаимная слепота :)
Помогло только форматирование специальной программой "SD Memory Card Formatter" - после этого файлы появились, но нарисовалась новая проблема.
Теперь при загрузке список эффектов с карты правильно отображается в терминале, но при загрузке выдается "ошибка загрузки". Добавив отладочной информации оказалось, что выполняется попытка загрузки файлов с двойными расширениями, например, на карте есть 0001.out, а пытается загрузиться 0001.out.out !
Убрал добавление расширения здесь:
https://github.com/vvip-68/GyverPanelWiFi/blob/4978cc00b9c3f9c6dc9701c3abe4c9ca379b34a1/firmware/GyverPanelWiFi_v1.12/sdcard.ino#L165
И все заработало !!!
Автору огромное СПАСИБО за труд и замечательный продукт !!!

simonovich, 2021-12-14:
@alexder21, а какой у вас микроконтроллер?

vvip-68, 2021-12-14:
@alexder21 
Это странно... Вообще-то расширение '.out' удаляется при чтении списка файлов с карточки в строках
[image]

Разве что, если по какой-то причине у вас расширение файла - заглавными буквами '.OUT'?
На заглавные буквы это не сработает...

alexder21, 2021-12-15:
да, так и есть, расширение в отладке выводится заглавными, хотя в windows везде маленькими.
увидел эту строку и расслабился:
https://github.com/vvip-68/GyverPanelWiFi/blob/4978cc00b9c3f9c6dc9701c3abe4c9ca379b34a1/firmware/GyverPanelWiFi_v1.12/sdcard.ino#L61

vvip-68, 2021-12-15:
Не доглядел. Тот кусок что ниже с отрезанием был добавлен заметно позже по по просьбе пользователей. Чтобы больше имен влазило в пакет со списком имен файлов, отправляемых в приложение. В приложении буфер ограничен - если список слишком длинный - он обрезается.

Вероятно в этом куске с отрезанием надо ат использовать для поиска позиции '.out'

vvip-68, 2021-12-15:
Или, наверное правильней так - искать по последней точке в имени файла
```
      p = file_name.lastIndexOf(".");
      if (p>=0) file_name = file_name.substring(0, p);

```

closed 2021-12-15 by vvip-68