vvip-68/GyverPanelWiFi #191: Останавливается демо режим

tih66, 2021-12-15:
Имеется два изделия:
1. Гирлянда на окне размером 20Х13
2. Лампа 16Х16
Гирлянда работает без нареканий, а вот с лампой какие-то не понятки. примерно через минуту, автоматом отключается демо режим, яркость сама сбавляется в минимум и светятся тускло часы. Прошивка последняя. Единственное отличие в конструкции обоих изделий, это то что в лампе установлена тач кнопка. Пробовал в настройках отключать ее (думал может из-за неё), не помогает. Заново настраивал прошивку и прошивал с очисткой - демо режим все равно отключается. В чем может быть косяк?
P.S. Кроме кнопки, ни чего больше из доп элементов не подключено (плеер  и т.д.), диод с конденсатором установлены.

vvip-68, 2021-12-15:
Может быть в этом причина: 

>В: Матрица после включения самопроизвольно меняет яркость. Или до максимума или до минимума.
О: Вариант 1: У вас физически не подключена кнопка, а тип кнопки в скетче указан "Сенсорная": #define BUTTON_TYPE 0. В этом случае микроконтроллер свободный пин кнопки, притянутый внутренним резистором микроконтроллера к "земле" воспринимает как состояние кнопки "нажато". Длительное "нажатие" кнопки воспринимается скетчем как запрос на регулировку яркости матрицы.
Решение: либо припаяйте кнопку, либо измените в скетче в настройках параметров оборудования тип кнопки на "Тактовая кнопка" #define BUTTON_TYPE 1

>Вариант 2: Физическая кнопка не подключена, в назначении пинов указано:

>#define LED_PIN 2             // D2 пин ленты
#define PIN_BTN D4            // кнопка подключена сюда (D4 --- КНОПКА --- GND)
Такое определение ядро ESP8266 версии 2.7.4 назначает D2 - выход сигнала на ленту, D4 - на кнопку.
Ядро версии 3.x.x назначает и кнопку и ленту на один и тот же пин - D4.
Вывод сигнала на D4 воспринимается кнопкой как постоянное нажатие - сигнал к изменению яркости лампы.

>Решение: Варант 1:
Изменить назначение кнопки или ленты на другой пин, например

>#define LED_PIN D2             // D2 пин ленты
#define PIN_BTN D4             // кнопка подключена сюда (D4 --- КНОПКА --- GND)
Однако, ядро 3.x.x некорректно выводит сигнал на матрицу - будет постоянно гореть зеленым первый светодиод матрицы.

>Решение: Вариант 2:
Установить ядро версии 2.7.4 - при этом и кнопка и первый диод матрицы будут работать корректно, самопроизвольного изменения яркости не будет происходить.

tih66, 2021-12-15:
Попробую вечером поиграться, отпишусь. Спасибо.

tih66, 2021-12-16:
Отключил сенсорную кнопку и прошил с параметром #define BUTTON_TYPE 1, все заработало. Что интересно, просто отключая кнопку программно, эффекта не дает.

closed 2021-12-16 by tih66