vvip-68/GyverPanelWiFi #193: Не находит вложенные библиотеки

dymmer, 2021-12-15:
Здравствуйте. пытаюсь скомпилить проект под Nodemcu. Но постоянно ругается на все инклуды внутри esp_littlefs.h и esp_littlefs.c
#include "freertos/FreeRTOS.h"
#include "freertos/semphr.h"
#include "esp_err.h"
#include <sys/types.h>
#include <sys/reent.h>
#include <sys/stat.h>
#include <sys/time.h>
#include <sys/poll.h>
#include <dirent.h>
#include <string.h>
#include "sdkconfig.h"
и т.п.
Я думаю что так не должно быть. Я скачивал и подкладывал их один за одним, но чем дальше, тем больше. Не понимаю что не так. Выглядит как стандартные библиотеки. esp8266 2.7.4. Спасибо

vvip-68, 2021-12-15:
Да, это стандартные библиотеки esp8266 2.7.4
И судя по наличию этих сообщений - у вас версия ядра примерно 2.5.2
По крайней мере - не старше 2.6.2 в которой эти функции были введены.

closed 2021-12-15 by vvip-68

dymmer, 2021-12-15:
[image]
ну вообще если я правильно понимаю это 2.7.4 все-таки. Но спасибо за ответ

vvip-68, 2021-12-15:
Попробуйте удалить и поставить заново.

dymmer, 2021-12-15:
Удаление и установка помогли вроде. Ну как минимум компилируется. Спасибо за помощь и огромное спасибо за труд.