vvip-68/GyverPanelWiFi #200: Матрица на окне 6*16 проблемы размером

bulnv, 2021-12-15:
- нулевой вопрос, куда можно задонатить проекту?
- недавно открывал ишью, что лента не горела, загорелась таки, 10 на 10 было все идеально(в ленте 100 диодов).
- повесил на узкое окно 6*16 (6 длина, 16 высота) какие-то эффекты работают нормально
- плата esp8266
- установки размерности матрицы:
```
#define WIDTH 6                // ширина одного сегмента матрицы
#define HEIGHT 16               // высота одного сегмента матрицы
#define MATRIX_TYPE 0           // тип соединения диодов в сегменте матрицы: 0 - зигзаг, 1 - параллельная
#define CONNECTION_ANGLE 1      // угол подключения диодов в сегменте: 0 - левый нижний, 1 - левый верхний, 2 - правый верхний, 3 - правый нижний
#define STRIP_DIRECTION 0
- контроллер установлен в верхней левой части окна (на холодосе), направление движения направо
```
# проблема
- змейка работает криво, в строки, вроде попадает, но бывает не доходит до последнего пикселя, появляется в рандомной строке и продолжает ход, снова "ныряет" не доходя до конца итд
- камин\огонь тоже самое, не понятно где верх где низ, диоды загораются\затухают не объяснимо
- в прилаге для андроида размер матрицы показан 10*10 хотя в коде все установлено точно (6*16)
- несмотря на то, что размер матрицы указа 6*16 = 96, последняя 17я строка из 4х диодов загорается
- все это наводит на мысль что размер матрицы передается неверно. хотя в файле все указано

bulnv, 2021-12-15:
- установил wifi panel setup и проблема решилась, хотя автор может пояснить что я делаю не так
- нашел куда донатить, оплатил за проезд копеечку
- оставлю ишуй до завтра на случай если автор пояснит где я провафлил
- Огромное спасибо!

vvip-68, 2021-12-16:
Поясняю: настройки в скетче - значения "По умолчанию", которые применяются при первом старте только что залитой на чистый контроллер прошивки. Этими значениями по умолчанию инициализируется хранилище. Эти значения модно настраивать в программе - в данном случае по размерам - в PаnelSetup. 

После перезапуска контроллера (в том числе и после загрузки "измененной" прошивки без стирания всей памяти) настройки берутся из сохраненных в хранилище, то есть те, которые вы настроили в предыдущем сеансе.

Это и в FAQ где-то написано, и в комментариях в программе где-то... Но никто ж не читает...

closed 2021-12-16 by vvip-68