vvip-68/GyverPanelWiFi #201: не могу подключиться к сети

sepulka123, 2021-12-16:
Добрый день.
При включении контроллера панели возникает точка доступа.  
Со смартфона к ней подключаюсь и могу управлять панелью.  Проблем нет.
Но вот захотел сделать синхронизацию с Jinx. Но для этого сперва я должен подключится к роутеру и тут начинаются проблемы:
Не могу найти куда вписать название сети роутера.
Картинка в описании "Вики" не такая, как у меня на смартфоне.
[image]
Нет закладки "Настройки подключения". 
Есть "Соединение". 
[image]
Если отключиться от точки доступа ,  то можно задать имя контроллера панели "Имя устройства", адрес, порт, а вот имя сети нигде не вижу.
[image]
Объясните , пожалуйста, в чем я не прав.

vvip-68, 2021-12-16:
В том, что не установили и не используете специальное приложение для настройки матрицы (в т.ч. и параметров подключения к сети). Wiki слегка устарело - нет времени обновить.

Но вот скажите, вы ставили из папки Android приложение (apk) для управления матрицей - GyverPanelWiFi.apk
Неужели ничто не подтолкнуло вас задуматься зачем в этой же папки лежат еще пара apk-шек, и одно из них имеет ничего не говорящее название **PanelSetupWiFi.apk**?

closed 2021-12-16 by vvip-68

sepulka123, 2021-12-16:
[image]
Не вижу, куда вписать название сети.
Ткните, пожалуйста носом.

sepulka123, 2021-12-16:
Что такое "префикс топика"  ?

sepulka123, 2021-12-17:
Может "Логин" - сюда надо вписать имя сети роутера ?  И, соответственно, - пароль на вход в сеть.

AndreyLyamkin, 2021-12-17:
Ребятушки! Вы инструкцию[ читали](https://github.com/vvip-68/GyverPanelWiFi/wiki/Настройка-скетча-для-вашего-устройства)?

Если не вносили изменения в файл a_def_soft.h (строчки с 17 по 25), то адрес вашей платы будет 192.168.4.1
Плата сама создает точку доступа с вышеуказанным адресом. Все что нужно, это подключиться к сети "PanelAP" с паролем "12341234"
Потом уже подключаться из приложения к плате. И вносить изменения в настройки подключения к вашей домашней сети.
Все разжёвано уже по 150 раз. Читайте!

sepulka123, 2021-12-17:
"...подключаться из приложения к плате..."  -  КАК ?

vvip-68, 2021-12-17:
@sepulka123  - все написано [тут](https://github.com/vvip-68/GyverPanelWiFi/wiki/Настройка-параметров-в-приложении)  
С небольшими поправками - первичная настройка поключения выполняется в приложении PanelSetupWiFi.apk

sepulka123, 2021-12-17:
Что такое   " Префикс топика "  ?
Что туда вписать  ?

vvip-68, 2021-12-17:
@sepulka123 - ты издеваешься, да? Вот ссылку выше давал - там же разжевано все.
Если все равно остается непонятным - бросай, это не твоё...

sepulka123, 2021-12-28:
Добрый день. 
Извините, это опять я со своими 
дурацкими вопросами. 
Один контроллер удалось успешно 
прошить: контроллер ESP32, матрица 32х32, 
отображает еффекты,  отображает поток с 
"Джинкса" (правда только 16х16, определил  
всего две вселенные). 
Но, поскольку это получилось не вполне
очевидно (как то вдруг, что то менял, что то 
вставлял) , то захотел повторить результат, 
прошить еще один контроллер. И вот тут начались
сложности. Контроллер, который прошивал 
раньше вроде даже подключается к сети (вот
копия с экрана терминала) :
12:09:32.583 -> 
12:09:32.583 -> WiFiPanel v.1.12.2021.1210
12:09:32.583 -> 
12:09:32.583 -> 
12:09:32.583 -> Инициализация файловой системы... 
12:09:32.583 -> FS: Использовано 8192 из 1507328 байт
12:09:32.583 -> Загрузка строк выполнена.
12:09:32.583 -> Использованы адреса EEPROM 1000 - 1037
12:09:32.583 -> Свободно ячеек 3058
12:09:32.583 -> 
12:09:32.583 -> Host: 'WiFiPanel' >> 32x32
12:09:32.583 -> 
12:09:32.623 -> 
12:09:32.623 -> Инициализация SD-карты...
12:09:33.390 -> Найдены сохраненные настройки: SD://eeprom.bin
12:09:33.390 -> Загрузка списка файлов с эффектами...
12:09:33.390 -> Папка с эффектами /32x32 не обнаружена.
12:09:33.390 -> Эффекты Jinx! на SD-карте не обнаружены
12:09:33.530 -> 
12:09:33.530 -> Подключение к netis_458223 -> 192.168.13.1
12:09:33.530 -> .........
12:09:35.824 -> WiFi подключен. IP адрес: 192.168.13.1
12:09:35.824 -> 
12:09:35.824 -> UDP-сервер на порту 2390
12:09:35.824 -> 
12:09:35.824 -> Режим работы: ПРИЕМНИК; Тип потока - светодиоды в порядке подключения
12:09:35.824 -> Группа: 0 на 239.255.0.1 - 239.255.0.7
12:09:35.917 -> Режим: Снегопад
12:09:49.489 -> Не удалось получить IP aдрес сервера NTP -> ru.pool.ntp.org -> 0.0.0.0
12:09:49.489 -> Используем сервер по умолчанию: 85.21.78.91
12:09:49.489 -> NTP-сервер ru.pool.ntp.org -> 85.21.78.91
12:09:49.489 -> Отправка NTP пакета на сервер ru.pool.ntp.org
12:09:54.497 -> Таймаут NTP запроса!

Но, точку доступа он не создает, подключиться я к нему
не могу,  чтобы изменить адрес, управлять им. 
Этот адрес в нем храниться где то внутри и при
перепрошивке вдруг возникает без моего участия. 
Хорошо бы полностью стереть/обнулить этот 
контроллер. 
В комментариях в скетче у Вас есть фраза:

"Первая загрузка прошивки 1.12 поверх более 
ранней опции должна быть выполнена с настройкой 
в меню "Инструмнты" - "Erase Flash: Erase All"

Я не смог найти ничего похожего в меню "Инструменты".

sepulka123, 2021-12-28:
[image]

sepulka123, 2021-12-28:
Вставьте, пожалуйста скриншот. (пометьте птичкой, стрелочкой, как для тупых), где это искать.
Извините.
Спасибо.

vvip-68, 2021-12-28:
1. Он в логе пишет, что подключился к локальной WiFi сети:
```
WiFi подключен. IP адрес: 192.168.13.1
```
Если подключение к сети выполнено успешно - точка доступа создаваться не будет (ибо зачем?) пока ее принудительно не включишь в настройках соединения.

>> в меню "Инструмнты" - "Erase Flash: Erase All"

>Я не смог найти ничего похожего в меню "Инструменты".

Ну, для ESP32 такого нет, значит. Это только для ESP8266.
Тогда нужно при каждой загрузке скетча, если вы хотите, чтобы предыдущие настройки стирались и создавались по-новой - нужно в файле a_def_soft.h в строке 7 менять каждый раз значение константы, показывающей корректность сохраненных в EEPROM данных
```
#define EEPROM_OK     0x5D       // Флаг, показывающий, что EEPROM инициализирована корректными данными 
```

sepulka123, 2021-12-28:
Спасибо. Удалось оживить контроллер.
Но, почему то картинка на панели замирает, когда идет запрос NTP (это запрос времени, я так думаю).
Индикацию времени я не использую. Где можно запретить обращение к NTP серверу.

18:23:42.790 -> Не удалось получить IP aдрес сервера NTP -> ru.pool.ntp.org -> 0.0.0.0
18:23:42.790 -> Используем сервер по умолчанию: 85.21.78.91
18:23:42.790 -> NTP-сервер ru.pool.ntp.org -> 85.21.78.91
18:23:42.790 -> Отправка NTP пакета на сервер ru.pool.ntp.org
18:23:42.824 -> Ожидание поступления потока данных E1.31...
18:23:42.858 -> 
18:23:42.858 -> FPS: 0.10
18:23:47.831 -> Таймаут NTP запроса!
18:23:52.873 -> FPS: 24.90
18:24:02.882 -> FPS: 25.00
18:24:26.780 -> Не удалось получить IP aдрес сервера NTP -> ru.pool.ntp.org -> 0.0.0.0
18:24:26.780 -> Используем сервер по умолчанию: 85.21.78.91
18:24:26.780 -> NTP-сервер ru.pool.ntp.org -> 85.21.78.91
18:24:26.814 -> Отправка NTP пакета на сервер ru.pool.ntp.org
18:24:26.814 -> FPS: 24.70
18:24:31.821 -> Таймаут NTP запроса!

vvip-68, 2021-12-28:
> 18:23:42.790 -> Не удалось получить IP aдрес сервера NTP -> ru.pool.ntp.org -> 0.0.0.0
18:23:42.790 -> Используем сервер по умолчанию: 85.21.78.91
18:23:42.790 -> NTP-сервер ru.pool.ntp.org -> 85.21.78.91

У вас не настроен или неверно указан сервер точного времени NTP
К "запасному" серверу тоже нет доступа.
Любой запрос в интернет - блокирующий. Пока сервер не ответит - ничего другого контроллер делать не может. Только ждать. На время ожидания всё замирает. Через таймаут неответа - "жизнь" продолжается до следующего запроса.

Запретить обращения к серверу времени можно в приложении сняв галочку"Синхронизация NTP" на странице настройки часов.
Ну все очевидно же. И в инструкциях все разжевано/пережовано...
Но инструкции же - для лохов?

sepulka123, 2021-12-28:
Безусловно Спасибо !
Но причем здесь "лохи" ?
Я часы не использую, и туда не заглядывал.