vvip-68/GyverPanelWiFi #205: DFPlayer

bdu1977, 2021-12-27:
добрый день. 
Сразу хочу поблагодарить автора за такой крутой проект.
Собрал данный проект на прошивке GyverPanelWiFi-12 от 07.12.21. схема собрана по (DEVICE_VARIANT == 0), в папке проекта со схемами данная сборка именуется как scheme. используется сенсорная кнопка (которая живет своей жизнью), модуль ТМ1637, DFPlayer (на плате установлен чип МН2024К). контроллер NodeMCU V3
Питание контроллера выполненно в Vin. 
настройки DFPlayer следующие
#define USE_MP3 1               
#define DFPLAYER_TYPE 1   

Что попробовал.
версии до 1.12 вообще не видят DFPlayer сама гирлянда работает без вопросов
версия 1.12 ведет себя очень странно. при подаче питания на контроллер через microUSB (с компьтера),то несколько секунд происходит перезапуск контроллера, а в мониторе порта отображается следующее
Инициализация MP3-плеера...
DFPlayerError: Таймаут!
MP3 плеер недоступен.
Создание точки доступа PanelAP
Точка доступа создана. Сеть: 'PanelAP'. Пароль: '12341234'.
IP адрес: 192.168.4.1

UDP-сервер на порту 2390

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4021f5ed epc2=0x00000000 epc3=0x00000000 excvaddr=0x0aff0dc8 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffdf0 end: 3fffffc0 offset: 0190
3fffff80:  00000100 00000000 3ffef450 feefeffe  
3fffff90:  feefeffe 00000000 3fff1440 4022c162  
3fffffa0:  3fffdad0 00000000 3fff1440 4023714c  
3fffffb0:  feefeffe feefeffe 3ffe85e4 40101595  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

WiFiPanel v.1.12.2021.1210

Инициализация файловой системы... 
FS: Использовано 24576 из 2072576 байт
Загрузка строк выполнена.
Использованы адреса EEPROM 1000 - 1037
Свободно ячеек 3058

Host: 'WiFiPanel' >> 16x16

Найдены сохраненные настройки: FS://eeprom.bin

Инициализация MP3-плеера...
DFPlayerError: Таймаут!
MP3 плеер недоступен.
Создание точки доступа PanelAP
Точка доступа создана. Сеть: 'PanelAP'. Пароль: '12341234'.
IP адрес: 192.168.4.1

UDP-сервер на порту 2390

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4021f5ed epc2=0x00000000 epc3=0x00000000 excvaddr=0x0aff

Если еще подать дополнительное питание на отдельный разъем, а питание  через разъем microUSB оставить, то в динамике слышно начало проигрывания мелодии, которая записана на карту, после 2 сек, громкость воспроизведения резко падает и после 3-4 сек, происходит перезагрузка контроллера. в мониторе порта отображается следующее
DFPlayerError: Файл не найден
DFPlayerError: Файл не найден
DFPlayerError: Файл не найден
DFPlayerError: Файл не найден
DFPlayerError: Файл не найден
DFPlayerError: Файл не найден

Звуков будильника найдено: 8
Звуков рассвета найдено: 7
Звуков сообщений найдено: 0
Создание точки доступа PanelAP
Точка доступа создана. Сеть: 'PanelAP'. Пароль: '12341234'.
IP адрес: 192.168.4.1

UDP-сервер на порту 2390

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4021f5ed epc2=0x00000000 epc3=0x00000000 excvaddr=0x0aff0dc8 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffdf0 end: 3fffffc0 offset: 0190
3fffff80:  00000100 00000000 3ffef450 feefeffe  
3fffff90:  feefeffe 00000000 3fff1440 4022c162  
3fffffa0:  3fffdad0 00000000 3fff1440 4023714c  
3fffffb0:  feefeffe feefeffe 3ffe85e4 40101595  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

WiFiPanel v.1.12.2021.1210

Инициализация файловой системы... 
FS: Использовано 24576 из 2072576 байт
Загрузка строк выполнена.
Использованы адреса EEPROM 1000 - 1037
Свободно ячеек 3058

Host: 'WiFiPanel' >> 16x16

Найдены сохраненные настройки: FS://eeprom.bin

Инициализация MP3-плеера...
Поиск файлов на карте плеера...
DFPlayerError: Файл не найден
........

где-то вычитал, что резисторы на 1кОм на входа RX и TX можно убрать, что сам DFPlayer нужно запитывать от 3.3v (ставил отдельную плату на 3.3v). ЗЕМЛЯ везде была соединена между собой. это не дало никакого результата. 

так же была замечена еще одна странность, что пока у меня подключено какое-то оборудование (плеер, часы, кнопка) то залить прошивку с изменениями я не могу. пришлось все это сделать на разъемах. как только отключаю все навесное оборудование, то прошивка заливается. но эффекта нет никакого.

ВОПРОС.
в чем может быть причина?

P.S.
кто-нибудь победил сенсорную кнопку? как ее приручить? кстати, по ней была замечена следующая анамалия, я пока так с ней борюсь. на данный момент я в гирлянде оставил только LED и сенсорную кнопку, остальное отпаял. при включении питания, гирлянда зажигается на полную яркость и тут же начинает гаснуть. если во время угасания просто быстро убрать питание  и тут же его подключить обратно, то кнопка работает просто отлично. но дня через 2-3 опять приходится повторять данную процедуру (когда вижу что яркость упала).

vvip-68, 2021-12-27:
Ну, что можно сказать. Собирайте по частям. Сначала просто матрица, убедитесь что она работает правильно. Затем добавляйте кнопку, затем следующие модули. Локализуйте проблему. Устраняйте.

Резисторы на RX/TX думаю можно убрать.
Только LED и сенсорная кнопка и падающая яркость - у меня такое наблюдалось, когда ядро ESP8266 стояло версии 3.x.x - оно несовместимо с данной прошивкой и FаstLED. При неизменных настройках (от оригинала) это ядро назначает вывод сигнала на D4, на этот же пин назначена кнопка. Управляющий сигнал на этом пине воспринимается библиотекой кнопки как постоянное нажатие. Постоянное нажатие в прошивки отрабатывается как сигнал к изменению яркости.

Невозможность прошить при подключенных модулях говорит о том, что на какой-то из важных пинов - GPIO0, GPIO2, GPIO15 при подключенных модулях подаются не те уровни сигналов, которые блокируют старт микроконтроллера в рабочем режиме или режиме программирования (загрузки скетча).

При правильном подключении (как на схеме) и соответствующей распиновке подключений (как в оригинале) - проблем не наблюдается. Если что-то меняли в подключении - оно может вот так вот проявляться. Другой причиной - неисправные модули, которые подают на эти пины напряжение которого там быть не должно при нормальной работе.

DFPlayer - модуль довольно капризный. Его работоспособность зависит от множества факторов и от конкретного экземпляра плеера в частности. Вот по даташиту - его рабочий диапазон напряжений - от 3.3 вольта до 4.2 максимум. Некоторые экземпляры работают при питании системы в целом до 4.8 вольта, а если подавать больше - плеер начинает дурить.

Циклическая перезагрузка - когда добавляете в систему новые модули, включаете новые функции - отследите момент после чего начинает возникать данная проблема. Сейчас трудно сказать что происходит.

vvip-68, 2021-12-27:
Еще питание сенсорной кнопки (по многочисленным отзывам) лучше брать с пина 3.3V микроконтроллера, параллельно пинам питания кнопки ставить керамический конденсатор порядка 100 нф

bdu1977, 2021-12-27:
собрано как на схеме, пины назначены как и в описании. ядро стоит 2,7,4, все библиотеки из архива. 

на кнопку попробую поставить кондер (буду заказывать, у нас негде купить) и запитать от 3,3В. 

DFPlayer запитывал от 3,3В, но он работает так же как и от 5В. наверное просто вырежу его из схемы и попрощаюсь, до лучших времен.

Цикличная перезагрузка начинается при подключении DFPlayer, как только его отключаю в скетче ( но физически он подключен) то перезагрузки не наблюдается.

vvip-68, 2021-12-27:
Во, вспомнил. В a_def_hard есть константа - GUARD_DELAY
У меня плеер нормально работает с ее значением от 25 до 75.
Некоторые писали мне, что у них плеер начинает нормально работать только начиная со значения 100-150.
А если значение мало - происходит циклическая перезагрузка.
Попробуйте увеличить ее значение. Ну или да - выкинуть плеер совсем.

bdu1977, 2021-12-29:
константу - GUARD_DELAY нашел не в  a_def_hard, а в DFPLAYER но это не важно. с малым значение вообще не запустился, значения от 150 до 200 с шагом 10 делал, эффект один и тот же.  в мониторе отображается ошибка плейера - таймаут, потом писк на секунду начинает проигрываться мелодия, при этом каждую секунду в монитор выходит сообщение что файл с мелодией не найден, на секунду воспроизводится начало мелодии и опять ошибка "файл не найден". так происходит 7 раз, потом показывается что идет подключение к сети. при этом мелодия звучит в динамике. по окончанию звучания - выходит сообщение, что мелодия 1 воспроизведена. световые эффекты работают. 
а вот дальше......... устанавливал будильник, не срабатывает. пробовал в настройках (через телефон) во вкладке будилььника выбирать мелодию и нажимать на кнопку воспроизведения, В динамике тишина. сообщений об ошибке нет в мониторе. 
может кто поможет?????
можно конечно и убрать этот капризный плеер, но я всю эту конструкцию 16*16 LED смог запихнуть в тонкий 1см толщиной 600*600мм светильник для потолка Армстронг, из Леруа. смотрится круто, но плеер.......

vvip-68, 2021-12-29:
> константу - GUARD_DELAY нашел не в a_def_hard, а в DFPLAYER но это не важно

Это важно. Критически важно. Столько возни было с этим плеером, чтобы заставить его работать с учетом разных подходов и того, что плееры собираются на разных микросхемах, которые ведут себя по-разному. И версия, в которой GUARD_DELAY - это какая-то древняя версия, еще до всей этой возни, так что вполне может быть и неработоспособной с вашей версией плеера. Хотя и с новой версией в которой константа перекочевала в A_DEF_HARD, как показывает практика не у всех работает. Правда иногда потом выяснялось, что то пин у контроллера сгорел, то на плеере сопля, то резистор не того номинала... Так что...

vvip-68, 2021-12-29:
А что файл с мелодией не найден - может показывать еще и на неработоспособность SD-карты. Попробуйте ее поменять на другую. Вот была система, в которой карты до 2GB читает прекрасно и все воспроизводит, а карты большей емкости на той же системе - невкакую... И только, кажется, понизив напряжение питания плеера до 3.3 вольт он начал читать и большие карты тожею Так что - DFPlayer - дело тонкое. И непредсказуемое.

bdu1977, 2021-12-29:
> > константу - GUARD_DELAY нашел не в a_def_hard, а в DFPLAYER но это не важно
> 
> Это важно. Критически важно.

еще раз просмотрел код глазами, сделал поиск но константу  GUARD_DELAY не нашел на вкладке a_def_hard, она есть нашлась только dfplayer 
версию прошивки смотрел последнюю 1.12
скачал только что с сайта еще раз архив, и да данная переменная нашлась. вечером попробую эту версию. почему у меня переменная на другой вкладке , не понял.

bdu1977, 2021-12-29:
> А что файл с мелодией не найден - может показывать еще и на неработоспособность SD-карты. Попробуйте ее поменять на другую. Вот эта система, в которой карты до 2GB читает прекрасно и все воспроизводит, а карты большей емкости на той же системе - нивкакую ... Так что - DFPlayer - дело тонкое. И непредсказуемое.

карточка 2 GB, отформатирона в FAT низкоуровневым методом. пробовал разные форматы вплоть до NTFS везде одинаково 
буду пробовать снова и снова

Anatolyj1970, 2021-12-29:
Собрал на ESP WROOM 32. Все работает кроме плеера. Чип оригинальный, плеер рабочий, проверял. GUARD_DELAY увеличил до 100. Ну никак не хочет видеть плеер
11:59:30.180 -> DFPlayerError: Таймаут!
11:59:30.321 -> MP3 плеер недоступен.
11:59:30.415 -> Создание точки доступа PanelAP
11:59:30.461 -> Точка доступа создана. Сеть: 'PanelAP'. Пароль: '12341234'.
11:59:30.461 -> IP адрес: 192.168.4.1
11:59:30.461 -> 
11:59:30.461 -> UDP-сервер на порту 2390
11:59:30.461 -> 
11:59:30.461 -> Режим работы: АВТОНОМНЫЙ, синхронизация E1.31 отключена
11:59:30.461 -> 
11:59:30.508 -> Режим: Морской прибой

bdu1977, 2021-12-30:
> > > константу - GUARD_DELAY нашел не в a_def_hard, а в DFPLAYER но это не важно
> > Это важно. Критически важно.

скачал последнюю версию. менял константы от 30 до 200, постоянная перезагрузка процессора идет. буду выдирать DFPlayer и ждать лучших времен.

спасибо за проект.

Anatolyj1970, 2021-12-30:
> > > > константу - GUARD_DELAY нашел не в a_def_hard, а в DFPLAYER но это не важно 
> > > > Это важно. Критически важно.
> 
> скачал последнюю версию. менял константы от 30 до 200, постоянная перезагрузка процессора идет. буду выдирать DFPlayer и ждать лучших времен.
> 
> спасибо за проект.

Тоже самое

vvip-68, 2021-12-30:
Ну что я могу сказать... Присылайте мне почтой ваше железо на макетке - могу поковыряться. У меня ни с одной из имеющихся у меня комбинаций плееров и контроллеров таких проблем не возникает. И, поскольку не возникает - я ни понять в чем дело, ни исправить ничего не могу.

Ставьте утилиту ESP Exception Decoder - скармливайте ей стек ошибки - оно хоть покажет примерно в каком модуле (в какой процедуре) вылетает и вероятную причину вылета.

Anatolyj1970, 2021-12-30:
Загрузил пример и в мониторе порта Com Error 129

vvip-68, 2021-12-30:
Это где? Это что за ошибка? Скриншот хоть дайте. Ничего не понятно

Anatolyj1970, 2021-12-30:
загрузил скетч из папки примеров Playmp3. После загрузки этого примера  в ком порт выводит:
11:19:00.305 -> 
11:19:00.305 -> Com Error 129
11:19:00.305 -> volume 0
11:19:10.336 -> 
11:19:10.336 -> Com Error 129
11:19:10.336 -> files 0
11:19:20.367 -> 
11:19:20.367 -> Com Error 129
11:19:20.367 -> playback mode 0
11:19:20.367 -> starting...

vvip-68, 2021-12-30:
А, ну тут ничего сказать не могу. Смотрите что скетч из примеров делает и в каком месте эта ошибка выводится. Скорее всего это уведомление о невозможности коммуникации с плеером - не отвечает на запросы.
Значит - либо не туда припаян, либо если есть резисторы - не того номинала, либо при пайке какая сопля повешена, либо плеер сгоревший в части RX/TX пинов, либо соответствующие пины микроконтроллера сгоревшие.

Есть вероятность, что когда разберетесь в чем причина и пример из библиотеки заработает - он заработает и в прошивке.

vvip-68, 2021-12-30:
По даташиту на плеер - возвращаемая ошибка 129 - 
"0x40 Module returns an error data with this command"

Anatolyj1970, 2021-12-30:
Я пробовал два разных плеера, и две разные платы. Единственное попробовать другие пины указать 1 и 3, т.е. аппаратный сериал. Резисторы 1 ком. По поводу соплей проверил еще раз, их нет. А не может быть косяк в SoftwareSerial

vvip-68, 2021-12-30:
> А не может быть косяк в SoftwareSerial

Нет, не может. Косяк где-то или в вашей сборке (физически) или в настройках пинов что-куда подключено.
Потому что у пары сотен человек всё это работает (с этими библиотеками) и никаких проблем не возникает.
И только (как мне известно) у вас двоих есть с этим проблемы. Вероятность, что проблема именно у вас гораздо выше вероятности, что проблема в библиотеках.

Anatolyj1970, 2021-12-30:
11:19:00.305 -> Com Error 129 . Пишут, что эту ошибку выдает библиотека SoftwareSerial (нет связи с плеером). Может скинешь библиотек свою.

vvip-68, 2021-12-30:
Эту ошибку пишет библиотека MP3 плеера (точнее - пример к библиотеке), в проекте - папка libraries/DFPlayer_Mini_Mp3_by_Makuna/examples файл PlayMp3.ino, строки 54-60, функция OnError.
Число 129 - в той же папке библиотеки /ыкс/DFMiniMp3.h - строка 603. 
И да - как я писал выше - это "ошибка связи с  плеером.

Библиотека "SoftwareSerial" является составной частью ядра ESP8266/ESP32 файлы лежат соответственно в папке с ядром, куда оно там установилось.

7918514, 2021-12-30:
Парни, попробуйте этот вариант. У меня тоже не работает этот плеер на последних прошивках. Но вот на одной из промежуточных неожиданно заработал. Библиотеки  ставил из проекта.
[GyverPanelWiFi_v1.12_df_work.zip](https://github.com/vvip-68/GyverPanelWiFi/files/7794555/GyverPanelWiFi_v1.12_df_work.zip)

vvip-68, 2021-12-30:
Точно - по пробуйте и сообщите о результатах. Дюже интересно.
Я проверил - по большому счету эта версия отличается от текущей тем, что в текущей исправлено множество грамматических ошибок в комментариях. Других существенных отличий нет.

Anatolyj1970, 2022-01-06:
Видимо такая схема работает только на микроконтроллере ESP8266. У меня ESP32
На ESP8266 и ESP32 треск из динамика. Причем на прошивке от Шайтан работает корректно.  Буду пробовать!

Anatolyj1970, 2022-01-14:
Удалось прошить GyverPanelWiFi_v1.12_df_work.zip. Ничего не изменилось. Плеер не видит. Попробовал загрузить пример из библиотеки тоже плеер не видит. Пока отключил плеер. Лампа заработала. Заказал пару штук новых. как придут буду пробовать ставить новый.

Mosrom, 2022-01-14:
Может быть попробовать заставить их работать на тестовом примере бибилиотеки DFRobotDFPlayerMini.Если заработают, то точно дело в библиотеке. Резисторы при подключении DfPleer и модулю ESP использовали? Может и в них дело. 
Я тоже купил несколько модулей, но они в новой библиотеке как и у Вас не определяются.

Anatolyj1970, 2022-01-14:
Резисторы 1 ком. Мультиметром проверил резисторы, они исправны. Пересобрал заново всю схему. Спиртом удалил остатки флюса. Поставил ради эксперимента другие резисторы. И пробовал на тестовом примере библиотеки DFRobotDFPlayerMini. Не определяется плеер

Mosrom, 2022-01-14:
Я смотрю Вы и ESP разные пробовали и DFplayer разные подключали.
Можно, чтобы наверняка купить DFplayer от DFRobot. Он стоит рублей 600.
Однако у меня такое ощущение, что что-то не так в схеме.
Некоторые ESP имеют разные назначения и номера PIN. Проверьте их. Где покупали, должна быть схема распиновки.
Сообщите о результате.
Я тоже с этим мучаюсь.

Anatolyj1970, 2022-01-14:
Все подключения я делаю по даташит "железа" и соответственно меняю пины, если надо в прошивке. Вчера специально поменял пины дисплея, при пайке. После переназначения их местами в прошивке дисплей заработал.

Anatolyj1970, 2022-01-14:
"Можно, чтобы наверняка купить DFplayer от DFRobot. Он стоит рублей 600."
Я, не уверен на 100%, что в место него не пришлют клон.

Anatolyj1970, 2022-01-14:
Надо посмотреть завтра. 
Контроллер wemos lolin32 esp-wroom-32. 
У него GPIO 1- TX0, GPIO 3 - RX0, а также есть и GPIO17 TX2 и GPIO16 - RX2. Может переключить их местами.

vvip-68, 2022-01-14:
Так, если вы на аппаратные RX/TX пины садите плейер - может потому у вас и не работает? Может быть назначение SoftwareSerial на аппаратные пины вызывают конфликты?

Anatolyj1970, 2022-01-14:
Так и сажаю не на аппаратные, а на GPIO17 TX2 и GPIO16 - RX2 . А завтра и хочу на аппаратные повесить.

Simafor47, 2022-01-22:
> > > > константу - GUARD_DELAY не нашел в a_def_hard, а в DFPLAYER но это не важно 
> > > > Это важно. Критическая важность.
> 
> скачал последнюю версию. менял константы от 30 до 200, постоянная перезагрузка процессора идет. буду выдирать DFPlayer и ждать лучших времен.
> 
> спасибо за проект.
Та же проблема. На более старой версии прошивки всё работает.
[image]

Anatolyj1970, 2022-01-24:
На версии 1.10 и библиотекой DFplayer от DFRobot, плеер обнаружен. Осталось разобраться с звуковыми файлами. В монитор порта выводится сообщение, что звуковых файлов не найдено.

Mosrom, 2022-01-24:
Попробуйте записать файлы на другую CD карту. Может быть важен размер карты (более 8 гб возможно DFPlayer не читает), а также в какой файловой системе отформатировна карта. Обычно это FAT32.

Anatolyj1970, 2022-01-24:
Все находит. Карта была не защелкнута до конца. Но на этой прошивке  сенсорная кнопка глючит сама по себе включает и выключает яркостьяркость

vvip-68, 2022-01-24:
Это у вас скорее всего вывод на ленту и кнопка - на один и тот же пин посажены или неправильно сенсорная кнопка инициализируется с подтяжкой не к тому уровню.

Anatolyj1970, 2022-01-24:
кнопка на d6, лента на d2. С прошивкой 1.12 кнопка работает без нареканий. Может ее надо от 5 вольт запитывать.

7918514, 2022-01-24:
> кнопка на d6, лента на d2. С прошивкой 1.12 кнопка работает без нареканий. Может ее надо от 5 вольт запитывать.

Если у вас включена в коде sd карта, то на d6 нельзя ничего вешать. SD карта использует d6, d7 и пин чипсплект (по умолчанию d8)

Anatolyj1970, 2022-01-24:
Нет у меня подключен только DFplayer. // D6 для схем с DFPlayer    (D6 --- КНОПКА --- GND). Перезагрузил вроде пока все нормально. Подскажи, а где настраивается звук часов.

Anatolyj1970, 2022-01-24:
Сигнальный провод от матрицы был рядом с сигнальным проводом кнопки. Пинцетом отвел в сторону и кнопка перестала глючить. Видимо сигнальный провод от матрицы сильно шумит.

7918514, 2022-01-24:
Если нет sd, то на d6 всё должно работать.
Звук часов? Типа отбивка каждого часа? Если да, то этого нет в проекте. Это задел на будущее. Будет реадизовано или нет, нужно у vvip-68 спрашивать.

vvip-68, 2022-01-24:
Звук часов настраивается... нигде.
Это все из нереализованных планов...

vvip-68, 2022-01-24:
> Сигнальный провод от матрицы был рядом с сигнальным проводом кнопки. Пинцетом отвел в сторону и кнопка перестала глючить. Видимо сигнальный провод от матрицы сильно шумит.

И опыт, сын ошибок трудных...

Mosrom, 2022-01-24:
> На версии 1.10 и библиотекой DFplayer от DFRobot, плеер обнаружен. Осталось разобраться с звуковыми файлами. В монитор порта выводится сообщение, что звуковых файлов не найдено.

Зачем билиотеку DFRobot заменили не понятно. Теперь ищем старые версии скетчей и не можем использовать новые.

vvip-68, 2022-01-25:
Заменили затем, что с прежней версией библиотеки DFPlayer были сотни жалоб, что с ней плеер не работает, особенно если это не "оригинальный" плеер, а клон на чипах МН2024К

Anatolyj1970, 2022-01-27:
Сегодня попробовал залить прошивку  FireLamp_EmbUI. Плеер определился с первого раза.

7918514, 2022-01-27:
> Сегодня попробовал залить прошивку FireLamp_EmbUI. Плеер определился с первого раза.

А файлы проигрываются? Вообще с этим плеером очень много было возни. Разные чипы ведут себя по разному, библиотеки все перепробовали, сравнивали разные проекты. В общем для себя сделал вывод, что эта китайская поделка хлам полнейший и выбросил её.

Anatolyj1970, 2022-01-27:
Да и озвучка эффектов и времени и будильников. Все работает.

Anatolyj1970, 2022-01-27:
А вообще я собрал 4 наливатора с плеером и 2 часов. Все плееры запускались без проблем

Simafor47, 2022-01-28:
Вот аналогично, 2 наливатора готово и даже намёка небыло на сбои. Перекидывал местами плееры, думал дело в нём, но нет,увы

vvip-68, 2022-01-28:
И самое печальное, что код работы с DFPlayer в этот проект был проверен и взят из кода наливатора...

voshodvoshod, 2022-02-04:
Добрый день. У меня уже собралась немалая коллекция этих модулей. С такими чипами декодера: 1-MH2024K-24SS(широкий), 2-JL AC218PQA672 SSOP24(узкий), 3-GD3200B SO16(узкий), 4-MH2024K-24SS(узкий), 5-MH2024K-16SS(узкий). Провёл небольшое исследование с подключением всех модулей к плате с LGT328P. Загружал в неё примеры из библиотек DFRobot и  makuna. Всегда использовал Softwareserial  и резисторы 1к на линиях связи при питании от 5 Вольт. Результаты такие: первые три модуля работают в примерах из двух библиотек без проблем. Остальные - только с библиотекой makuna. Далее подключал все модули к макету лампы в следующей конфигурации: ESP8266 LoLin, 16х16, управление матрицей-2, МР3плеер-3 и 4, дисплей- 6 и 7, сенсорная кнопка - 8. В прошивке отключено всё кроме визуальных эффектов на матрице, рассвета и будильника. Лампа всегда в режиме точки доступа.Плеер питается от 5 В, дисплей и кнопка от 3,3 В с платы контроллера. Естественно выполнены все рекомендации автора по версиям библиотек и ядра. Так вот: в прошивке 1.11 работает только модуль с чипом JL AC218PQA672 SSOP24(узкий). Все остальные не определяются с различными задержками. В целом лампа работоспособна, с небольшими глюками, но это не беда. В таком виде собранное изделие используется у знакомых несколько месяцев без замечаний. Пока не пробовал в этой версии заменить библиотеку на makuna. Сразу перешёл на версию 1.12 с теми же параметрами. Все нововведения не используются. Сделал это только для проверки совместимости модулей плеера. Прошивка собралась и загрузилась. Загрузка с полным стиранием, как указано автором. Ситуация 1 в 1, что и в первом сообщении- циклический рестарт контроллера. Ни одна платка плеера не определилась. Пробовал менять параметр для выбора типа модуля (т.е библиотеки) и значение задержки- результат тот же. Таким образом получается, что на одинаковом железе (JL AC218PQA672 SSOP24(узкий)) с одинаковыми параметрами одна прошивка работает, а другая нет. Я считаю, что проблема в версии 1.12, а не в модулях плеера. Для себя вижу выход в замене библиотеки на makuna в версии прошивки 1.11. Остаётся ещё один вопрос: почему прошивка 1.11 не видит модули с MH2024K-24SS(широкий) и GD3200B SO16(узкий), а примеры из библиотек с ними работают?

vvip-68, 2022-02-04:
1. Версия 1.11, насколько я помню - работает с библиотекой DFPlayerRopotics, которая вроде как работает только с оригинальными DFPlayer.
2.  Циклическая перезагрузка - вероятно недостаточная величина задержки между командами - GUARD_DELAY - попробуйте ее увеличить.
3. Попробуйте найти разницу между 1.11 и 1.12 в плане использования и инициализации подключения к плееру.
На мой взгляд (насколько я помню) - разницы почти нет. Во всех случаях при проверке наличия плеера в системе с него пытаются получить что-нибудь (например уровень текущей громкости), и если плеер не отвечает - считается, что его нет в системе.

Возможно вам удастся найти отличие и понять что идет не так.
У меня не такой богатый набор плееров по разнообразию, но пока что - на всех имеющихся у меня разновидностях - плееры определяются и работают.

Позже, когда у меня появится свободное время - я вернусь к изучению поведения плееров.
Возможно что-то найду. Но если вы всё "почините" раньше меня - сообщите что было не так.
Я исправлю в прошивке все так, как работает у вас со всеми плеерами.

voshodvoshod, 2022-02-04:
1.В Arduino IDE библиотека называется "DFRobotDFPlayerMini". Указанные выше первые три декодера в ней работают, но в Вашей прошивке только чип от JL.
2. При значении GUARD_DELAY 100 изменений нет. Проверю дальше.
3. 1.12 прошил только для проверки, переходить на неё для лампы не имеет смысла.
Однако я уже использовал её для гирлянды на окно. Но там отсутствовали  плеер, дисплей и кнопка. Не удалялись игры, погода и бегущая строка. Остальные параметры такие же. По словам владельца всё работало.Попробую изучить глубже.
Если что найду-сообщу.

voshodvoshod, 2022-02-04:
Методом перебора был найден модуль, обнаруживаемый прошивкой 1.12  при значении задержки 100. Это опять же модуль с чипом от JL. Прошивка видит его даже при 75-и. Оставил 100.  Далее все опыты провожу с этим модулем. Количество файлов в папках читает верно. На этом хорошие новости заканчиваются. Далее происходит вывод сообщений, как в начальном посте этой темы, и перезапуск по кругу. И при каждом плеер и карта обнаруживаются правильно. И точно так же при отключении плеера в прошивке #define USE_MP3 0,  запуск контроллера проходит успешно. Сравнение участков кода для плеера в разных прошивках показало их значительное отличие. Анализ "чужого кода" дело долгое и пока отставил. "CUT HERE FOR EXCEPTION DECODER "- это я предполагаю сам плеер выдаёт, а прошивка транслирует в монитор порта?

vvip-68, 2022-02-04:
А в чем значительное отличие? В том, что в разных библиотеках по сути одни и те же функции, выполняющие одно и то же действие называются немного по разному? Или какое-то существенное смысловое отличие есть?

voshodvoshod, 2022-02-04:
Пока не понял, нужно немного больше времени. Хочу просмотреть EspExceptionDecoder

[24 more replies]

closed 2022-02-11 by vvip-68