vvip-68/GyverPanelWiFi #206: Проблемы с подключением.

bosser2407, 2021-12-27:
Никакую не хочет работать через роутер. 
Said пароль и IP ввожу верно, в роутере устройство появляется. Но по этому адресу в приложении подключатся отказывается((

vvip-68, 2021-12-27:
Ставите на телефон программу типа Ping и добиваетесь, чтобы контроллер пинговался с телефона. 
Смотрите настройки сети в контроллере и в роутере.
Как только контроллер начнет отвечать на запросы программы с телефона - вы сможете подключиться.

closed 2021-12-27 by vvip-68

bosser2407, 2021-12-28:
> Ставите на телефон программу типа Ping и добиваетесь, чтобы контроллер пинговался с телефона. Смотрите настройки сети в контроллере и в роутере. Как только контроллер начнет отвечать на запросы программы с телефона - вы сможете подключиться.

[image]
Сразу после ввода SSId, пароля и перезагрузки пингуется. Через секунду уже нет. Какие настройки можно ковырять, я честно не представляю. Если бы так себя вели все устройства, а такое только с этой прошивкой

vvip-68, 2021-12-28:
Отключите для начала получение погоды и синхронизацию времени с интернета. Если контроллер пытается получить погоду или время - операции блокирующие - и сервер не отдает данные - контроллер больше ничего делать не умеет кроме как ждать ответа. В этот момент подключиться приложением к контроллеру не удастся. Если после отключения погоды и времени соединение будет устанавливаться - дальше смотрите почему не получает погоду. Если вы с Украины и стоит синхронизация с Яндекс - погоды - ваше правительство заблокировало сервера яндекса и получить с него погоду не получится.

vvip-68, 2021-12-28:
Эффект "Лампа" - это просто чтобы забить место в менюшках приложения. Настоящий список эффектов принимается от контроллера после установки соединения. Очевидно, что если нормального соединения установить не удается - ничего от контроллера и не получается в приложение. В том числе и список эффектов.

И смотрите еще что в мониторе порта. Может у вас контроллер циклически перегружается по какой-то причине...

bosser2407, 2021-12-28:
> Эффект "Лампа" - это просто чтобы забить место в менюшках приложения. Настоящий список эффектов принимается от контроллера после установки соединения. Очевидно, что если нормального соединения [...]

Чудеса. Отключение погоды кажется помогло. Я в Крыму, и с яндексом у нас вроде проблем нет.  И с отключением погоды, в MQTT пошли топики. До этого MQTT тоже не работал
[image]

vvip-68, 2021-12-28:
Потому что скорость порта надо выставить такую же как настроено в приложении - а там стоит 115200

vvip-68, 2021-12-28:
Что у вас возвращает в браузере, скажем. такой запрос
 https://yandex.ru/time/sync.json?geo=62

bosser2407, 2021-12-28:
> Как то так

[image]

vvip-68, 2021-12-28:
Ну, то есть - так то погоду с яндекса он получает... Почему же в приложении включение получения погоды завешивает?... А если погоду с OpenWeatherMap брать?