vvip-68/GyverPanelWiFi #207: Фризы при эффектах

Chez718, 2021-12-31:
Здравствуйте, сделал панель 30х60, использовал сначала esp8266, все было нормально, но не хватало фпс, заменил на esp32, стало работать заметно быстрее, но иногда проскакивают фризы, каждые 4-5 секунд на пол секунды и бывает каждые 10 секунд на секунду, в чем может быть проблема?

Chez718, 2021-12-31:
Дополню: эффекты абсолютно любые, от простых до сложных, поэтому, думаю вряд ли проблема в нехватке оперативной памяти. И да, с есп8266 работало медленно но без фризов, а тут быстрее, но с фризами.

vvip-68, 2022-01-01:
Не настроено соединение с интернетом.

closed 2022-01-01 by vvip-68

Chez718, 2022-01-01:
Вроде с самого начала настроил и время синхронизировал, сеть есть.

vvip-68, 2022-01-01:
Других причин для фризов кроме как блокирующих запросов в интернет - нет. Время, погода, MQTT - что так у вас включено...