vvip-68/GyverPanelWiFi #208: SD Card

suprime532, 2022-01-03:


closed 2022-01-03 by vvip-68

suprime532, 2022-01-03:
Добрый день! Большое спасибо вам за данный проект!
Все собрал, прошил, работает!
Вопрос по поводу SD Card: Закинул 13 эффектов на карточку, все эффекты работаю но только когда в ручном режиме их переключаешь. В режиме случайный выбор, последовательно, в автоматическом режиме после одного эффекта сразу переходит не на следующий не с карточки а с внутренней памяти контроллера. 
В файле a def_hard.h   на строке WAIT_PLAY_FINISHED стоит 1.
На строке REPEAT_PLAY стоит 0.
Пробовал менял значения в обоих строчках, но все равно проигрывает только один эффект с карточки.
После нажатия на кнопку сброс, проигрывание с карточки не происходит.
карточки 3 штуки менял.
Может подскажите что можно сделать ?
Прилагаю кусок информации из монитора порта:

21:54:24.889 -> WiFiPanel v.1.12.2021.1210
21:54:24.889 -> 
21:54:24.923 -> 
21:54:24.923 -> Инициализация файловой системы... 
21:54:24.923 -> FS: Использовано 253952 из 2072576 байт
21:54:25.025 -> Загрузка строк выполнена.
21:54:25.025 -> Использованы адреса EEPROM 1000 - 1037
21:54:25.025 -> Свободно ячеек 3058
21:54:25.025 -> 
21:54:25.025 -> Host: 'WiFiPanel' >> 18x25
21:54:25.025 -> 
21:54:25.025 -> 
21:54:25.025 -> Инициализация SD-карты...
21:54:25.059 -> Найдены сохраненные настройки: FS://eeprom.bin
21:54:25.059 -> Загрузка списка файлов с эффектами...
21:54:25.059 -> Папка с эффектами /18x25 обнаружена.
21:54:25.059 -> Найдены файлы эффектов:
21:54:25.059 ->   001		575.23 К
21:54:25.059 ->   002		790.28 К
21:54:25.093 ->   003		728.27 К
21:54:25.093 ->   004		906.38 К
21:54:25.093 ->   005		730.91 К
21:54:25.093 ->   006		744.11 К
21:54:25.093 ->   010		744.11 К
21:54:25.093 ->   008		736.19 К
21:54:25.093 ->   009		753.34 К
21:54:25.093 ->   007		572.59 К
21:54:25.127 ->   012		877.36 К
21:54:25.127 ->   011		781.05 К
21:54:25.127 ->   013		860.21 К
21:54:25.501 -> Создание точки доступа PanelAP
21:54:27.473 -> Точка доступа создана. Сеть: 'PanelAP'. Пароль: '12341234'.
21:54:27.473 -> IP адрес: 192.168.4.1
21:54:27.473 -> 
21:54:27.473 -> UDP-сервер на порту 2390
21:54:48.858 -> Настройки сохранены в EEPROM
21:54:49.504 -> Режим: Узоры
21:55:11.515 -> Режим: SD-Карта
21:55:11.562 -> Загрузка файла эффекта: '/18x25/001.out' -> ok
21:55:37.706 -> '/18x25/001.out' - завершено
21:55:37.740 -> Режим: Часы
21:55:59.736 -> Режим: Узоры
21:56:21.748 -> Режим: SD-Карта
21:56:21.748 -> Режим: Часы
21:56:43.766 -> Режим: Узоры
21:57:05.739 -> Режим: SD-Карта
21:57:05.739 -> Режим: Часы
21:57:27.759 -> Режим: Узоры
21:57:49.786 -> Режим: SD-Карта
21:57:49.786 -> Режим: Часы
21:58:11.818 -> Режим: Узоры
21:58:33.779 -> Режим: SD-Карта
21:58:33.813 -> Режим: Часы

vvip-68, 2022-01-03:
Насколько я помню, также должен быть отключен режим "Случайный порядок" вообще у эффектов.  Тогда он последовательно перебирает эффекты, а когда доходит до эффекта SD-карта - начинает перебирать все эффекты с карты (у которой также выбран режим "последовательный") пока не переберет все эффекты с карты. Потом переходит на первый обычный эффект и так далее по кругу.

suprime532, 2022-01-03:
Так у меня везде убрано случайный порядок!