vvip-68/GyverPanelWiFi #209: SD Card

suprime532, 2022-01-03:
Добрый день! Большое спасибо вам за данный проект!
Все собрал, прошил, работает!
Вопрос по поводу SD Card: Закинул 13 эффектов на карточку, все эффекты работаю но только когда в ручном режиме их переключаешь. В режиме случайный выбор, последовательно, в автоматическом режиме после одного эффекта сразу переходит не на следующий не с карточки а с внутренней памяти контроллера.
В файле a def_hard.h на строке WAIT_PLAY_FINISHED стоит 1.
На строке REPEAT_PLAY стоит 0.
Пробовал менял значения в обоих строчках, но все равно проигрывает только один эффект с карточки.
После нажатия на кнопку сброс, проигрывание с карточки не происходит.
карточки 3 штуки менял.
Может подскажите что можно сделать ?
Прилагаю кусок информации из монитора порта:

21:54:24.889 -> WiFiPanel v.1.12.2021.1210
21:54:24.889 ->
21:54:24.923 ->
21:54:24.923 -> Инициализация файловой системы...
21:54:24.923 -> FS: Использовано 253952 из 2072576 байт
21:54:25.025 -> Загрузка строк выполнена.
21:54:25.025 -> Использованы адреса EEPROM 1000 - 1037
21:54:25.025 -> Свободно ячеек 3058
21:54:25.025 ->
21:54:25.025 -> Host: 'WiFiPanel' >> 18x25
21:54:25.025 ->
21:54:25.025 ->
21:54:25.025 -> Инициализация SD-карты...
21:54:25.059 -> Найдены сохраненные настройки: FS://eeprom.bin
21:54:25.059 -> Загрузка списка файлов с эффектами...
21:54:25.059 -> Папка с эффектами /18x25 обнаружена.
21:54:25.059 -> Найдены файлы эффектов:
21:54:25.059 -> 001 575.23 К
21:54:25.059 -> 002 790.28 К
21:54:25.093 -> 003 728.27 К
21:54:25.093 -> 004 906.38 К
21:54:25.093 -> 005 730.91 К
21:54:25.093 -> 006 744.11 К
21:54:25.093 -> 010 744.11 К
21:54:25.093 -> 008 736.19 К
21:54:25.093 -> 009 753.34 К
21:54:25.093 -> 007 572.59 К
21:54:25.127 -> 012 877.36 К
21:54:25.127 -> 011 781.05 К
21:54:25.127 -> 013 860.21 К
21:54:25.501 -> Создание точки доступа PanelAP
21:54:27.473 -> Точка доступа создана. Сеть: 'PanelAP'. Пароль: '12341234'.
21:54:27.473 -> IP адрес: 192.168.4.1
21:54:27.473 ->
21:54:27.473 -> UDP-сервер на порту 2390
21:54:48.858 -> Настройки сохранены в EEPROM
21:54:49.504 -> Режим: Узоры
21:55:11.515 -> Режим: SD-Карта
21:55:11.562 -> Загрузка файла эффекта: '/18x25/001.out' -> ok
21:55:37.706 -> '/18x25/001.out' - завершено
21:55:37.740 -> Режим: Часы
21:55:59.736 -> Режим: Узоры
21:56:21.748 -> Режим: SD-Карта
21:56:21.748 -> Режим: Часы
21:56:43.766 -> Режим: Узоры
21:57:05.739 -> Режим: SD-Карта
21:57:05.739 -> Режим: Часы
21:57:27.759 -> Режим: Узоры
21:57:49.786 -> Режим: SD-Карта
21:57:49.786 -> Режим: Часы
21:58:11.818 -> Режим: Узоры
21:58:33.779 -> Режим: SD-Карта
21:58:33.813 -> Режим: Часы

vvip-68, 2022-01-03:
Duplicate of #208

closed 2022-01-03 by vvip-68

suprime532, 2022-01-03:
Так у меня и поставлено последовательно!

vvip-68, 2022-01-03:
Ну, попробуйте тогда отладить. Может поймете в чем дело. У меня на момент последней проверки работает именно так, как я написал.

sidi111, 2022-01-03:
У меня тоже такая проблема https://github.com/vvip-68/GyverPanelWiFi/discussions/198#discussioncomment-1901139

suprime532, 2022-01-03:
Пока не могу понять в чем проблема

suprime532, 2022-01-03:
sidi111 у вас точно так же только 1 эффект показывает на карточке ?

sidi111, 2022-01-03:
Да. На следующий не переходит.

suprime532, 2022-01-03:
Кстати надо будет проверить питание карточки, там же должно быть 3.3 вольта

suprime532, 2022-01-03:
3.3 вольта показывает.

sidi111, 2022-01-03:
Здесь так должно быть или стоит что-то изменить?

`// ************************* РАБОТА С SD КАРТОЙ *************************

#if (USE_SD == 1)

#define  MAX_FILES 126                      //  больше нельзя - переполнение индекса массива для типа int8_t -128..127; Значение 127 и < 0 - специальные значения для перехода по циклу эффектов

String   nameFiles[MAX_FILES];              // Список имен файлов, найденных в папке на SD-карте
bool     isSdCardReady = false;             // Флаг - SD карта инициализирована корректно
bool     play_file_finished = false;        // Флаг - воспроизведение эффекта завершено
bool     wait_play_finished = (WAIT_PLAY_FINISHED == 1); // Флаг - true - переключаться на следующий эффект только когда весь файл показан; false - прерывать показ по истечении времени эффекта 
uint8_t  countFiles = 0;                    // Количество найденных файлов эффектов, найденных на SD-карте
int8_t   sf_file_idx = -1;                  // Эффект, проигрываемый с SD-карты - индекс в массиве nameFiles

#endif

// ************************* ПРОЧИЕ ПЕРЕМЕННЫЕ *************************

bool     repeat_play = (REPEAT_PLAY == 1);  // Флаг - true - повторное проигрывание текущего эффекта, если время не вышло; false - перейти к следующему эффекту после однократного проигрывания текущего (игры, SD-карта)
uint8_t  fadeMode = 4;                      // Фаза режима плавной смены эффекта
bool     modeDir;                           // Направление перехода к следующему / предыдущему режиму
`

suprime532, 2022-01-03:
Здесь ничего менять не нужно

sidi111, 2022-01-03:
Вот такой замкнутый процесс получается. Куда копать не пойму, мозг уже плавится.

20:06:37.628 -> Ответ на 192.168.50.199:2390 >> ack27;
20:06:38.579 -> '/36x16/rgb-red.out' - завершено
20:06:38.646 -> Загрузка файла эффекта: '/36x16/rgb-red.out' -> ok
20:06:42.996 -> UDP << ip='192.168.50.199:2390'; cmd='$18 0;'
20:06:42.996 -> Получен UDP пакeт размером 6
20:06:42.996 -> Ответ на 192.168.50.199:2390 >> ack28;
20:06:48.298 -> UDP << ip='192.168.50.199:2390'; cmd='$18 0;'
20:06:48.298 -> Получен UDP пакeт размером 6
20:06:48.298 -> Ответ на 192.168.50.199:2390 >> ack29;
20:06:53.634 -> UDP << ip='192.168.50.199:2390'; cmd='$18 0;'
20:06:53.634 -> Получен UDP пакeт размером 6
20:06:53.634 -> Ответ на 192.168.50.199:2390 >> ack30;
20:06:59.002 -> UDP << ip='192.168.50.199:2390'; cmd='$18 0;'
20:06:59.002 -> Получен UDP пакeт размером 6
20:06:59.002 -> Ответ на 192.168.50.199:2390 >> ack31;
20:07:04.338 -> UDP << ip='192.168.50.199:2390'; cmd='$18 0;'
20:07:04.338 -> Получен UDP пакeт размером 6
20:07:04.375 -> Ответ на 192.168.50.199:2390 >> ack32;
20:07:09.712 -> UDP << ip='192.168.50.199:2390'; cmd='$18 0;'
20:07:09.712 -> Получен UDP пакeт размером 6
20:07:09.712 -> Ответ на 192.168.50.199:2390 >> ack33;
20:07:15.077 -> UDP << ip='192.168.50.199:2390'; cmd='$18 0;'
20:07:15.077 -> Получен UDP пакeт размером 6
20:07:15.077 -> Ответ на 192.168.50.199:2390 >> ack34;
20:07:20.413 -> UDP << ip='192.168.50.199:2390'; cmd='$18 0;'
20:07:20.413 -> Получен UDP пакeт размером 6
20:07:20.413 -> Ответ на 192.168.50.199:2390 >> ack35;
20:07:24.795 -> '/36x16/rgb-red.out' - завершено
20:07:24.896 -> Загрузка файла эффекта: '/36x16/rgb-red.out' -> ok
20:07:25.747 -> UDP << ip='192.168.50.199:2390'; cmd='$18 0;'

suprime532, 2022-01-03:
Заработало! )))
В общем решил я переформатировать карточку в fat, до этого была fat32.
Отформатировал, заного загрузил эффекты и все заработало!
До этого форматировал несколько раз в fat32, сразу не допер что надо было попробовать в fat.
sidi111 попробуй, мне помогло!
Монитор порта теперь такой )))
22:56:55.205 -> WiFiPanel v.1.12.2021.1210
22:56:55.205 -> 
22:56:55.205 -> 
22:56:55.205 -> Инициализация файловой системы... 
22:56:55.205 -> FS: Использовано 253952 из 2072576 байт
22:56:55.252 -> Загрузка строк выполнена.
22:56:55.299 -> Использованы адреса EEPROM 1000 - 1037
22:56:55.299 -> Свободно ячеек 3058
22:56:55.299 -> 
22:56:55.299 -> Host: 'WiFiPanel' >> 18x25
22:56:55.299 -> 
22:56:55.299 -> 
22:56:55.299 -> Инициализация SD-карты...
22:56:55.299 -> Найдены сохраненные настройки: FS://eeprom.bin
22:56:55.299 -> Загрузка списка файлов с эффектами...
22:56:55.299 -> Папка с эффектами /18x25 обнаружена.
22:56:55.299 -> Найдены файлы эффектов:
22:56:55.346 ->   001		575.23 К
22:56:55.346 ->   002		790.28 К
22:56:55.346 ->   003		728.27 К
22:56:55.346 ->   004		906.38 К
22:56:55.346 ->   005		730.91 К
22:56:55.346 ->   006		744.11 К
22:56:55.346 ->   007		572.59 К
22:56:55.346 ->   008		736.19 К
22:56:55.346 ->   009		753.34 К
22:56:55.346 ->   011		781.05 К
22:56:55.346 ->   012		877.36 К
22:56:55.346 ->   013		860.21 К
22:56:55.767 -> Создание точки доступа PanelAP
22:56:57.732 -> Точка доступа создана. Сеть: 'PanelAP'. Пароль: '12341234'.
22:56:57.732 -> IP адрес: 192.168.4.1
22:56:57.732 -> 
22:56:57.732 -> UDP-сервер на порту 2390
22:56:57.732 -> Режим: SD-Карта
22:56:57.732 -> Загрузка файла эффекта: '/18x25/001.out' -> ok
22:57:19.494 -> Настройки сохранены в EEPROM
22:57:19.775 -> Загрузка файла эффекта: '/18x25/002.out' -> ok
22:57:41.771 -> Загрузка файла эффекта: '/18x25/003.out' -> ok
22:58:03.767 -> Загрузка файла эффекта: '/18x25/004.out' -> ok
22:58:25.763 -> Загрузка файла эффекта: '/18x25/005.out' -> ok
22:58:47.755 -> Загрузка файла эффекта: '/18x25/006.out' -> ok
22:59:09.752 -> Загрузка файла эффекта: '/18x25/007.out' -> ok
22:59:31.748 -> Загрузка файла эффекта: '/18x25/008.out' -> ok
22:59:53.744 -> Загрузка файла эффекта: '/18x25/009.out' -> ok
23:00:15.757 -> Загрузка файла эффекта: '/18x25/011.out' -> ok
23:00:37.753 -> Загрузка файла эффекта: '/18x25/012.out' -> ok
23:00:59.749 -> Загрузка файла эффекта: '/18x25/013.out' -> ok
23:01:21.745 -> Режим: Часы
23:01:43.741 -> Режим: Узоры
23:02:05.784 -> Режим: SD-Карта
23:02:05.784 -> Загрузка файла эффекта: '/18x25/001.out' -> ok
23:02:27.780 -> Загрузка файла эффекта: '/18x25/002.out' -> ok
23:02:49.777 -> Загрузка файла эффекта: '/18x25/003.out' -> ok
23:03:11.755 -> Загрузка файла эффекта: '/18x25/004.out' -> ok
23:03:33.763 -> Загрузка файла эффекта: '/18x25/005.out' -> ok
23:03:55.769 -> Загрузка файла эффекта: '/18x25/006.out' -> ok
23:04:17.753 -> Загрузка файла эффекта: '/18x25/007.out' -> ok
23:04:39.756 -> Загрузка файла эффекта: '/18x25/008.out' -> ok
23:05:01.845 -> Загрузка файла эффекта: '/18x25/009.out' -> ok
23:05:23.820 -> Загрузка файла эффекта: '/18x25/011.out' -> ok
23:05:45.808 -> Загрузка файла эффекта: '/18x25/012.out' -> ok
23:06:07.828 -> Загрузка файла эффекта: '/18x25/013.out' -> ok
23:06:29.777 -> Режим: Часы

sidi111, 2022-01-03:
У меня тоже в FAT32, сейчас форматну.

suprime532, 2022-01-03:
Надеюсь остальным тоже поможет )

suprime532, 2022-01-03:
sidi111 ну что помогло ?

sidi111, 2022-01-03:
> sidi111 ну что помогло?

Пока не понятно файлов вообще не видит на SD

suprime532, 2022-01-03:
У меня до сих пор работает )
попробуй еще отформатировать в fat16

suprime532, 2022-01-03:
не видит ?
а ты папку создал, потом в нее кинул файлы ?

sidi111, 2022-01-03:
Вроде бы все получилось, переключаются файлы. Только вот я заметил что файлы с SD не подчиняются установленному времени переключения режима, работают только по времени файлу созданному в Jinx.
По сути проблема решена, как вы и предположили, форматированием SD в FAT.
Спасибо за решение.

suprime532, 2022-01-03:
Рад что как то помог, по сути решение получилось случайно )

suprime532, 2022-01-03:
сейчас проверил, у меня четко по установленному времени переключаются файлы

suprime532, 2022-01-03:
кстати еще, вы говорите что у вас файлы меняются по времени созданому в Jinx, так у вас наверно настроено так. В файле def_hard посмотрите, там есть настройки как по времени самого файла, так и по времени заданному в приложении

suprime532, 2022-01-05:
В общем проблема сегодня снова появилась, проверил блок питания осцилографом, оказалось большие пульсации.
Заменил вздутые конденсаторы, теперь вроде работает как надо.
Через несколько дней дополню отчёт )

sidi111, 2022-01-05:
У меня сейчас вроде все нормально переключается.
Вопрос один. 
Чем можно воспроизвести файлы **.out** на компьютере?
Jinx не воспринимает этот файл.

sidi111, 2022-01-05:
> Чем можно воспроизвести файлы **.out** на компьютере?
> Jinx не воспринимает этот файл.

Разобрался, jinx frame.