vvip-68/GyverPanelWiFi #210: E1.31 Jinx

Vasap62, 2022-01-09:
Доброго времени суток! собрал проэкт на последней прошивке на контроллере ESP32. Настроил подключение через роутер и трансляцию с Jinx, но на панели все эффекты с сильной задержкой и низким фпс. Не подскажете в чем может быть проблема? Тоже самое при синхронизации 2 контроллеров во всех режимах кроме режима команды. Заранее спасибо!

vvip-68, 2022-01-09:
Проблема может быть с настройками роутера или данный роутер плохо поддерживает работу с  multicast UDP.
Попробуйте поковыряться с настройками роутера изи заменить его на другой.

Принцип работы прост - на каждый вызов FastLED.show() в "эфир" выплевываются UDP пакеты с изображением того что на матрице. А поймает их кто-нибудь, обработает или нет - мастер-контроллер в принципе не знает.

В свою очередь потребитель - как только поймал UDP пакет с пикселями немедленно вывел их на матрицу.
Если роутер "потерял" половину пакетов - отобразятся только те, что получены.

В режиме "команды" посылается не "картинка" с мастер-матрицы, а команда типа "включить эффект номер N"
Объем данных в таких UDP пакетах маленький и они, очевидно, доходят до потребителей в полном объеме.

Мой DIR615 никак не хотел нормально принимать/передавать UDP multicast пакеты и принимаемое изображение "рвалось". Замена роутера на более современный (TPLink Archer C80) решило все проблемы.

closed 2022-01-09 by vvip-68

Vasap62, 2022-01-09:
Спасибо за консультацию! Я попробовал сделать точку доступа при помощи другого ESP32 и подключиться через него, стало получше, но все равно есть фризы. Наверное придется менять роутер

Sting666th, 2022-06-15:
Здравствуйте. Тоже столкнулся с такой проблемой. Всё решилось включением мультикаста в роутере. Если сложно самим найти эти настройки (они есть не во всех роутерах, но в большинстве), наймите соображающего человека. Это всяко дешевле, чем покупать новый роутер, в котором мультикаст может быть включён из коробки, а может и не быть.