vvip-68/GyverPanelWiFi #212: призрак бегущей строки

bdu1977, 2022-01-27:
долго не мог поймать призрака, но в итоге он пойман, вот осталось только изгнать его.
версия прошивки 1.12, погода берется с Yandex , к другому ресурсу вопросов нет, там нет призрака. одна из текстовых строк такая "Сегодня{D: dd.MM.yyyy HH:mm}, температура {WS} {WT}
При принудительном просмотре или же когда она автоматически выводится на экран, то есть два варианта показа
1. все выводится нормально, вопросов нет, отрицательная температура показывается синим цветом. 
2. вот тут и есть призрак. если в показаниях вывода погоды содержится слово СНЕГ, не зависимо от того в каком он сочетании идет (сильный снег, местами или еще как-то, главное слово СНЕГ), в этом случае вывод на экран следующий, как пример "облачно с прояснениями небольшой снег {C#00FFFF} -12" температура, в данном случае -12 выводится белым цветом

если сделать другую строку в которой будет просто написано "{WS} {WT}" то в этом случае, даже если в прогнозе есть слово снег, то "призрака нету" все выводится нормально.

у кого-нибудь есть такое? как изгнать демона?

CrazyGhostRider, 2022-01-29:
Я замечал, что даже при установленной галочке "случайный цвет", то всегда текст погоды белый. А вот цифры, то цветные, то белые. Чинить не пытался. 

П.с. Чтобы не плодить темы: Обычная бегущая строка иногда прерывается на середине показа и показываются эффекты из демо режима. В последний раз это было на строке "До нового года осталось--//--" и включился демо-арканоид.

frol-aleksan, 2022-01-31:
> Обычная бегущая строка иногда прерывается на середине показа

Решается раскомментированием строки (showTextNow && !fullTextFlag) в файле custom.ino

bdu1977, 2022-02-01:
прерывание бегущей строки тут не причем. вопрос-то заключается в том, что в строке появляется управляющий символ, вот этот {C#00FFFF} и он же и выводится на экран в самой бегущей стреке.

vvip-68, 2022-02-03:
Код периодически появляется. Причине непонятна. Снега у нас давно не было, чтобы при такой погоде посмотреть. Просто добавление слова "снег" в строку эффекта не имеет.

Иногда связь приложения с матрицей плохая, соединение происходит, но список строк в приложение передать не удается. Внешне это выглядит как "все строки похерены". Нужно просто отключиться от матрицы и подключиться снова. 
Обычно "пропавшие строки" появляются.

Прерывание бегущей строки эффектом, как было указано выше товарищем @frol-aleksan указано совершенно верно.

Des-75, 2022-02-03:
нет, я именно видел в строке прописанный код, который и выводился. Т.е. с точки зрения работы строк всё было верно. Не понятно только почему и как он туда прописывается.

vvip-68, 2022-02-03:
Так и я о том же.

Des-75, 2022-02-03:
Ладно, попробую объяснить с моей точки зрения:
1. Я вижу, что вместо обычных строк матрица отображает код типа {C#00FFFF} (я точно не помню, но ОЧЕНЬ похоже, что именно этот).  Ещё раз, вместо строк "бежит" следующая надпись: "{C#00FFFF}".
2. Я загружаю приложение, подключаюсь к матрице и вижу, что в одной из строк этот код и прописан. Остальные строки пустые.
3. Как это можно объяснить плохой связью с приложением?! Я приложением подключаюсь после появления этой строки. Скажу больше, я перезагружал матрицу выключением питания и ничего не изменялось. Лечилось только восстановлением настроек из приложения.

closed 2022-02-07 by bdu1977

Des-75, 2023-02-01:
2022/23, версия прошивки 1.13. За время работы с середины декабря 2022 по январь 2023 один раз был глюк со строками... Лечил переписыванием всех строк заново в приложении.