vvip-68/GyverPanelWiFi #213: не компилируется ESP32

djshmel, 2022-02-02:
In file included from C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_rmt_esp32.h:116,
                 from C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms/esp/32/fastled_esp32.h:8,
                 from C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms.h:36,
                 from C:\Program Files (x86)\Arduino\libraries\FastLED\src/FastLED.h:52,
                 from sketch\a_def_hard.h:318,
                 from D:\ардуино\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.12\GyverPanelWiFi_v1.12.ino:64:
C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/esp_hw_support/include/esp_intr.h:8:2: warning: #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [-Wcpp]
 #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead
  ^~~~~~~
In file included from sketch\a_def_hard.h:318,
                 from D:\ардуино\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.12\GyverPanelWiFi_v1.12.ino:64:
C:\Program Files (x86)\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000
 #    pragma message "FastLED version 3.004.000"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms/esp/32/fastled_esp32.h:3,
                 from C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms.h:36,
                 from C:\Program Files (x86)\Arduino\libraries\FastLED\src/FastLED.h:52,
                 from sketch\a_def_hard.h:318,
                 from D:\ардуино\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.12\GyverPanelWiFi_v1.12.ino:64:
C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h: In static member function 'static void _ESPPIN<PIN, MASK>::hi()':
C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h:15:17: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'out1_w1ts'; did you mean 'out_w1ts'?
       else GPIO.out1_w1ts.val = MASK;
                 ^~~~~~~~~
                 out_w1ts
C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h: In static member function 'static void _ESPPIN<PIN, MASK>::lo()':
C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h:20:17: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'out1_w1tc'; did you mean 'out_w1tc'?
       else GPIO.out1_w1tc.val = MASK;
                 ^~~~~~~~~
                 out_w1tc
C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h: In static member function 'static void _ESPPIN<PIN, MASK>::set(_ESPPIN<PIN, MASK>::port_t)':
C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h:25:17: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'out1'; did you mean 'out'?
       else GPIO.out1.val = val;
                 ^~~~
                 out
C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h: In static member function 'static void _ESPPIN<PIN, MASK>::toggle()':
C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h:31:31: error: no match for 'operator^=' (operand types are 'volatile gpio_dev_s::<unnamed union>' and 'long unsigned int')
       if(PIN < 32) { GPIO.out ^= MASK; }
                      ~~~~~~~~~^~~~~~~
In file included from C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/dport_access.h:19,
                 from C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/hal/esp32c3/include/hal/cpu_ll.h:19,
                 from C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/hal/include/hal/cpu_hal.h:24,
                 from C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/esp_hw_support/include/esp_cpu.h:14,
                 from C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/esp_hw_support/include/soc/cpu.h:14,
                 from C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/esp_hw_support/include/soc/spinlock.h:11,
                 from C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/freertos/port/riscv/include/freertos/portmacro.h:38,
                 from C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/freertos/include/freertos/portable.h:51,
                 from C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/freertos/include/freertos/FreeRTOS.h:63,
                 from C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\cores\esp32/Arduino.h:33,
                 from sketch\GyverPanelWiFi_v1.12.ino.cpp:1:
C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/esp_common/include/esp_attr.h:131:25: note: candidate: 'touch_pad_intr_mask_t& operator^=(touch_pad_intr_mask_t&, touch_pad_intr_mask_t)'
 FORCE_INLINE_ATTR TYPE& operator^=(TYPE& a, TYPE b) { a = a ^ b; return a; } \
                         ^~~~~~~~
C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/esp_common/include/esp_attr.h:135:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/esp_common/include/esp_attr.h:136:19: note: in expansion of macro 'FLAG_ATTR_U32'
 #define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/hal/include/hal/touch_sensor_types.h:168:1: note: in expansion of macro 'FLAG_ATTR'
 FLAG_ATTR(touch_pad_intr_mask_t)
 ^~~~~~~~~
C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/esp_common/include/esp_attr.h:131:25: note:   no known conversion for argument 1 from 'volatile gpio_dev_s::<unnamed union>' to 'touch_pad_intr_mask_t&'
 FORCE_INLINE_ATTR TYPE& operator^=(TYPE& a, TYPE b) { a = a ^ b; return a; } \
                         ^~~~~~~~
C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/esp_common/include/esp_attr.h:135:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
C:\Users\yuki\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32c3/include/esp_common/include/esp_attr.h:136:19: note: in expansion of macro 'FLAG_ATTR_U32'
 #define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
[... 190 lines cut ...]
                 from D:\ардуино\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.12\GyverPanelWiFi_v1.12.ino:64:
C:\Program Files (x86)\Arduino\libraries\FastLED\src/pixeltypes.h:816:13: note: candidate: 'CRGB operator&(const CRGB&, const CRGB&)'
 inline CRGB operator&( const CRGB& p1, const CRGB& p2)
             ^~~~~~~~
C:\Program Files (x86)\Arduino\libraries\FastLED\src/pixeltypes.h:816:13: note:   no known conversion for argument 1 from 'volatile gpio_dev_s::<unnamed union>' to 'const CRGB&'
In file included from C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms/esp/32/fastled_esp32.h:3,
                 from C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms.h:36,
                 from C:\Program Files (x86)\Arduino\libraries\FastLED\src/FastLED.h:52,
                 from sketch\a_def_hard.h:318,
                 from D:\ардуино\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.12\GyverPanelWiFi_v1.12.ino:64:
C:\Program Files (x86)\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h:68:24: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'out1'; did you mean 'out'?
       else return GPIO.out1.val & MASK;
                        ^~~~
                        out
[... cut ...]
C:\Program Files (x86)\Arduino\libraries\FastLED\src/fastspi.h: At global scope:
C:\Program Files (x86)\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
exit status 1
Ошибка компиляции для платы ESP32C3 Dev Module.
 
Помогите в чем проблема

vvip-68, 2022-02-02:
Проблема в вашей невнимательности.
В проекте указано (основной файл, строки 41-43) черным по белому написано какая версия ядра требуется для проекта, и что на версии 2.x.x работать не будет и скорее всего даже не скомпилится.
Вот, это оно.

closed 2022-02-02 by vvip-68

djshmel, 2022-02-02:
In file included from D:\ардуино\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.12\GyverPanelWiFi_v1.12.ino:64:0:
a_def_hard.h:314:10: fatal error: ArduinoOTA.h: No such file or directory
 #include <ArduinoOTA.h>          // Библиотека обновления "по воздуху"
          ^~~~~~~~~~~~~~
compilation terminated.
exit status 1
ArduinoOTA.h: No such file or directory

А с этим что не так, подскажите ?

vvip-68, 2022-02-02:
С этим не подскажу. ArduinoOTA - часть ядра ESP32.
Если у вас нету этого файла в составе ядра - значит оно криво установлено.
Попробуйте удалить его и установить заново.
 Этот файл лежит тут (с учетом вашего имени пользователя) - проверьте его наличие

[image]

djshmel, 2022-02-02:
спасибо все прошло , осталось какая-то ошибка но скетч компилируется и залит

In file included from C:\Users\yuki\Documents\Arduino\libraries\FastLED\src/FastLED.h:67:0,
                 from sketch\a_def_hard.h:318,
                 from D:\ардуино\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.12\GyverPanelWiFi_v1.12.ino:64:
C:\Users\yuki\Documents\Arduino\libraries\FastLED\src/fastspi.h:145:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

vvip-68, 2022-02-02:
Если бы было ошибки - скетч бы не скомпилился и не был бы залит на платку.

IgorBratsk, 2023-12-20:
Здравствуйте, не могу скомпилировать...

Arduino: 1.8.13 (Windows 7), Плата:"ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\Igor\AppData\Local\Arduino15\packages -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\Igor\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Igor\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Temp\arduino_build_588777 -warnings=none -build-cache C:\Temp\arduino_cache_882896 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.dfu-util.path=C:\Users\Igor\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=C:\Users\Igor\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1 -prefs=runtime.tools.esptool_py-4.5.1.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.riscv32-esp-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp-elf-gdb.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.xtensa-esp-elf-gdb-11.2_20220823.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.riscv32-esp-elf-gdb.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.riscv32-esp-elf-gdb-11.2_20220823.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.mklittlefs.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.openocd-esp32.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20230419 -prefs=runtime.tools.openocd-esp32-v0.12.0-esp32-20230419.path=C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20230419 -verbose C:\Users\Igor\Downloads\GyverPanelWiFi-master\GyverPanelWiFi-master\firmware\GyverPanelWiFi_v1.13\GyverPanelWiFi_v1.13.ino

C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\Igor\AppData\Local\Arduino15\packages -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\Igor\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Igor\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Temp\arduino_build_588777 -warnings=none -build-cache C:\Temp\arduino_cache_882896 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.dfu-util.path=C:\Users\Igor\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=C:\Users\Igor\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\Igo [... cut ...] 3.ino.bootloader.bin" "C:\\Users\\Igor\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.14\\tools\\sdk\\esp32\\bin\\bootloader_qio_80m.elf" ) )

����� "C:\Users\Igor\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1\esptool.exe" �� ᮢ���⨬� � ���ᨥ� Windows, ࠡ���饩 �� �⮬ ��������. �஢���� ᢥ����� � ��⥬�, �⮡� 㧭���, ����� ����� �ணࠬ�� x86 (32-ࠧ�來��) ��� x64 (64-ࠧ�來��) ��� �㦭�, � ������ �� � ���⠢騪� �ணࠬ����� ���ᯥ祭��.

exit status 1

Ошибка компиляции для платы ESP32 Dev Module.

vvip-68, 2023-12-20:
А кто вам сказал, что версия 1.13 вообще будет компилиться под ядро 2.0.14?