vvip-68/GyverPanelWiFi #214: синхронизация матриц

RichardKeonig, 2022-02-03:
спасибо большое за синхронизации в прошивке и за сам проект в целом. но я бы не написал если бы не возникла проблема( 
Дело в том что после создания учётной записи устройства в приложении-компаньоне "WiFi Panel Setup", указания IP-адреса: 192.168.0.100 и попытки подключиться ничего не происходит... подключиться возможно только если IP-адрес будет: 192.168.4.1
прошу помогите советом :-/

vvip-68, 2022-02-04:
Вот на этот 192.168.4.1 и подключаетесь, переходите в раздел настройки сетевого подключения, указываете там имя сети и желаемый постоянный IP адрес в этой сети (192.168.0.100 в вашем примере) и нажимаете кнопку "Сохранить". Если все прошло успешно - на матрице бегущей строкой отобразится этот адрес. Далее уже подключаете телефон к указанной сети и в приложениях используете этот 192.168.0.100 для подключения.

Если что-то идет не так - смотрите в мониторе порта что выводится - анализируя лог можно сказать что идет так или не так.

RichardKeonig, 2022-02-04:
добрый вечер, подключение приёмников прошло успешно, на матрице бегущей строкой высветился ip-адрес. но при подключении к сети источника не удаётся подключиться в WiFi Panel Setup к устройствам приёмникам

RichardKeonig, 2022-02-05:
Если не затруднит можете помочь разобраться с тем как заставить их работать не от роутера, пожалуйста. Вообще матрицы используются как детали пиксельного костюма. Ваша прошивка очень помогла. Ещё раз огромное спасибо!!!

vvip-68, 2022-02-05:
>  разобраться с тем как заставить их работать не от роутера,

Если не подключать к локальной сети - матрица будет работать в режиме точки доступа по адресу 192.168.4.1
Но это "начальный" режим, чтобы настроить доступ в локальную сеть - подключение к роутеру. В этом режиме функционал ограничен (только одна матрица, нет получения времени и погоды и т.д - в общем - начальный настроечный режим). Полный функционал - только в локальной сети роутера.

> возможно ли синхронизировать все матрицы подключив к сети одной из них? Мои попытки не увенчались успехом

Нет. Только при включении всех матриц в одну локальную сеть через роутер. Для того что вы хотите - нужно брать другую прошивку или писать свою.

closed 2022-02-05 by vvip-68