vvip-68/GyverPanelWiFi #216: Заполнение памяти 100%

Mosrom, 2022-02-20:
Обратил внимание, что после добавления немецкого алфавита память модуля полностью заполнена. Даже когда память заполнялась на 98% наблюдались задержки при воспроизведении эффектов. Скетч сильно разросся. Возникает ошибка компиляции. Раньше можно было удалить игры, но сейчас это сделать не так просто, так они глубоко интегрированы в прошивку. Те, кто использует модули с памятью 4 мб обязательно с этим столкнутся. Здорово, если была бы возможность отключать в скетче игровую составляющую, думаю, что не многие этим пользуются. Я даже в демо режиме их отключаю.

vvip-68, 2022-02-21:
Это у вас какие-то проблемы, не связанные с добавлением немецких букв.
Для немецких букв добавлено начертание самой буквы - это порядка 100 байт и пару case в условиях определения какой образ буквы из массива шрифтов выбирать для отображения.

Вот результат компиляции СО ВСЕМИ ВКЛЮЧЕННЫМИ фичами - звуки, индикатор, MQTT, SD-карта, погода и так далее. "Полный фарш"

[image]

Как видите - и до 60% заполнение не дотягивает...

closed 2022-02-21 by vvip-68

Mosrom, 2022-02-22:
Всё зависит от того, какую плату Вы выбрали в менеджере плат и какой объём памяти установили. У меня это ESP32 Dev Module.

vvip-68, 2022-02-22:
Гирлянда на 1000 лампочек. 92% постоянной (памяти кода) и 17% оперативной под переменные.
Ничего не виснет, всё фунциклирует.

[image]

Ну, берите ESP8266 - - там памяти больше остается. Или модуль с бОльшим количеством памяти.
Сам в недоумении почему ядро ESP32 столько памяти занимает. Наверное последствия индусских программистов....

vvip-68, 2022-02-22:
Хотя вот интересно - модуль выбран 4МБ в настройках - а объем доступной пишет 1МБ...
А... Там выделяется 1МБ на приложение, 1.2MБ на ОТА и 1.5 на SPIFFS...

Попробуйте отключить ОТА (будете по проводу прошивку заливать). Тогда 3МБ пот приложение и 1МБ под SPIFF.
Если выбрать Huge APP 3MB No ATA / 1 MB SPIFFS...