vvip-68/GyverPanelWiFi #218: 12 hour mode

Red-77, 2022-03-03:
is it possible to change the clock mode to 12 hour mode, cuz by default it is in 24 hour mode, thank you

vvip-68, 2022-03-03:
Here in the country of developer nobody uses 12H format of time display, only 24H
By this reason nobody wants this and nobody use it.
Source code is open - you may change script by yourself to use time format you want.

Red-77, 2022-03-03:
in what part of the code the time is formatted? alas where should i look and make changes to the format, sorry but im not good with coding, but i am willing to do it myself so can you please guide me a bit

vvip-68, 2022-03-03:
More obvious part of code is in clock.ino file, function DrawClock(), starting line 237
'hrs' is parameter of function containing hours of time in range 0..23.
You must doing simplest arithmetic operation to convert it from 0..23 to 1..12 range

Red-77, 2022-03-03:
ok i kind of figured it out the library has a function to return 12 hour time  "hourFormat12(); ". so that was simple to do, one last thing what is the purpose of line 128 : String clockCurrentText, do i need to change anything in that?

vvip-68, 2022-03-03:
I think yes. This function is used fro display current time as running text in some cases.
line 130 returns hours in range 0..23; Obviously you also need convert it to 12H format

Red-77, 2022-03-03:
ok i did it and it worked so if someone whats to change the time format to 12 hours

change the  hour();  function to  hourFormat12();
in line 74, 130, 801 and leave the rest as it is

Red-77, 2022-03-05:
ok so the clock is working fine but ,i am facing a problem, the time is displayed in 12 hour format if i am connected to the internet, but if i disconnect internet after 12 it turn to 13 cuz it cant sync to ntp, also if i use "set device time" it sets the time in 24 hours, so how can i set the timer to be 1-12 so that after 12 it turns back to 1

vvip-68, 2022-03-05:
Why you can`t sync time via internet? What is the reason? Internally time stored (and received from NTP server) in system as long integer (seconds since 1970 year) and it is no difference between 12H or 24H of **display** format. The same with "set device time" - phone internally sends time to controller in 24H format (as I understand) and sketch of controller converts them into long integer internally. I do not understand the issue...

If you means setting time of alerts and time dependent modes page of application - fields containing time shows it as returns clock component of application. All clocks an timers uses time in natural 24H format. If you want to see time in special 12H format - it is necessary somehow rework phone application (in aia-file in folder Android of project) in AppInventor web constructor. Because of size of application it works very slow - it is paine to change something in application. But you can try if you want.

Red-77, 2022-03-05:
i think you misunderstood my issue my issue is that, if internet connection is lost, after 12:59 it goes to 13:00, is there someway i can make it go from 12:59 to 01:00, otherwise i have to sync to ntp every minute

Red-77, 2022-03-05:
see let me explain my issue
i modified the code so that when it syncs to ntp it shows 12 hour format, which forces it to go to 1:00 after 12:59
but if it doesn't sync to ntp after 12:59 it turns to 13:00, cuz the internal counter still counts (0-23)
so if i loose my internet connection at 12:59, it will go to 13:00, is there some way i can fix it

vvip-68, 2022-03-05:
I still do not understand the issue. Whole day has 24 hours. All internal counters work in 24-hour cycle. It is system things. More exactly it counts every one second tick to increment large int value stored seconds. The method how you doing **print** int seconds counter (more common converted to hours with hour() or hourFormat12() function) on matrix screen - is fully your design and implementation. You even can display time in 8-hour intervals 0..7 and next again 0..7 if you need it for some exotic reason. I do not understand a subject of issue

Red-77, 2022-03-05:
ok whole day has 24 hours, in 24 hour system (0-23), in 12 hour system (1-12am 1-12pm), so all i want to know is how do i make the internal conter count from (1-12am 1-12pm) because if i loose internet connection and the clock is not able to sync to ntp, it shows 13:00 instead of showing 1pm. the line of code that i changed, requested 12 hour time from library, but it doesn't change the internal counter, so what i want to know is in which part of the code is the counter function located

vvip-68, 2022-03-05:
No. It is impossible. Because of nature. Because Earth doing one turn around in 24 hours. It not depend on ntp sync. Ntp Sync time returns time as amount of milliseconds since 1970.01.01 00:00 and does not depend on **DISPLAY** format you use - 24H or 12H. If in drawClock() function call of hourFormat12() returns wrong (unexpected) value as 13 - it is error in library function and you need to override it writing code manually. Like - if hour value is greater or equal to 13 - decrease value by 12. If returned value of hours is 0 - assign it to 12.

Red-77, 2022-03-05:
actually i fixed the issue, there were some function in other file that used hour(); changing them to hourFormat12 (); fixed the issue, i didn't even have to modify the app, now it all works

closed 2022-03-05 by Red-77