vvip-68/GyverPanelWiFi #219: Ошибка аутентификации в созданной матрицей точке доступа.

BrickerAsked, 2022-03-08:
Здравствуйте, хочется описать такую ситуацию: нужно было уехать из дома. Решил взять матрицу с собой, дома всё сделал и настроил все правильно. Поставил на матрице режим точки доступа, дома всё работало, матрица подключалась.

Затем после уезда я ее подключил к электричеству и она долго висела не включаясь, но на плате огонёк мигал (мол, может быть, сеть wifi ищет). Спустя минут 5 матрица включилась, точка появилась. Начал к ней подключаться, но не получилось "ошибка аутентификации" матрица сама работает в демо режиме. 

Попытался создать точку доступа с таким же SSID и паролем что и wifi дома, и о чудо! Матрица включилась мгновенно, но работает сильно тормозя.

Посмотреть IP/логи, просимулировав ситуацию смогу позже. Т.к. в матрице нет кнопки, а компьютера нет под рукой.

BrickerAsked, 2022-03-08:
[image]

BrickerAsked, 2022-03-08:
Если интересует: пароль менял, SSID точки тоже.

vvip-68, 2022-03-12:
Полагаю, что вы что-то сделали неправильно и НЕ настроили матрицу на создание точки доступа. Либо не дали времени сохраниться настройкам в энергонезависимую память. Настройки сохраняются после 30 секунд бездействия и если выключить матрицу сразу же после изменения настроек - они не успеют сохраниться в EEPROM.

Описанное поведение указывает как раз на это. После включения матрица в течении 5 минут пытается подключиться к сконфигурированной сети. Такой интервал задан специально, так как в случае отключения электричества, после включения матрица "загружается" гораздо быстрее роутера, не находит сети и дальше уже не пытается подключиться. Поэтому установлена пауза на ожидание пока роутер загрузится и установит соединение с интернетом.

Если за 5 минут подключиться к сконфигурированной сети не удается - матрица создает собственную точку доступа с IP 192.168.4.1. Прервать ожидание можно нажатием на кнопку (но у вас ее нет).

Естественно, если вы создаете сеть в роутере с таким же SSID и паролем - матрица при включении охотно к ней подключится. Никакого чуда тут нет. Если работает сильно тормозя - значит у данной сети нет выхода в интернет и попытки получить время из интернета сильно тормозят работу скетча из за таймаутов ожидания ответа на отправляемые запросы. Так работают использованные библиотеки доступа к интернету.

closed 2022-03-12 by vvip-68