vvip-68/GyverPanelWiFi #220: Удалённый доступ к матрице

Dem9ih4ik, 2022-03-08:
Подскажите где найти, если уже есть, гайд по подключению к матрице извне? Имею ввиду подключение из глобальной сети. АйПи белый и статичный, матрица подключена к вайфай, все нужные порты проброшены, хочу сделать интерактив на трансляции, чтоб не устанавливая приложения человек мог что-то вывести на матрицу вне зависимости от того где он и в какой сети. Правильность проброса портов протестил - родным приложением подключается из мобильной сети с телефона, а с браузера никак. Заранее спасибо)

vvip-68, 2022-03-08:
Чтоб не устанавливая приложения человек мог что-то вывести на матрицу вне зависимости от того где он и в какой сети из браузера, вам нужно переписать скетч, чтобы он принимал команды с этого самого браузера (очевидно по протоколу HTTP/HTTPS). Исходный вариант скетча принимает только команды по UDP, отправляемые из приложения. Еще может принимать команды, отправленные по каналу MQTT, чаще всего - тоже из приложения. Если вам нужно обеспечить возможность отправлять по каналу MQTT команды не из приложения, а из чего-то самописанного - вам нужно ЭТО самописанное, соответственно, написать. API команд управления по каналу MQTT - описано в wiki.

closed 2022-03-08 by vvip-68

vvip-68, 2022-03-08:
Нет, так как не нужно. И было и есть.
Могу посоветовать либо взять другую прошивку, в которой есть управление через браузер (прошивок масса), либо найти на форуме у Гайвера разработчика, который за какую-то денюжку перепишет прошивку под ваши требования. Код прошивки - открытый - берите что нужно и изменяйте на здоровье.